esphome/issues #2145: Wrong getting started instructions

Misiu, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** HAOS

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
I'm trying to install/add ESP Home to Home Assistant 2021.6.5. I follow the instruction available at https://esphome.io/guides/getting_started_hassio.html
The last version that was available to install was 1.17.1.
I wasn't able to install the new version.
I had to add a new repository to the addon store to be able to install 1.19.
The default repository I had in my HA instance was https://addons.community
With this repository, the version of ESPHome that is available to install is 1.17.1.
After adding https://github.com/esphome/hassio as the second repository additional versions showed up:
[image]

The instructions (https://esphome.io/guides/getting_started_hassio.html) should be more clear and the ESPHome should be available in one place and the old version available in the community add-ons repository should be removed.

**Problem-relevant YAML-configuration entries:**
```yaml
NOT RELATED

```

**Logs (if applicable):**

```
NOT RELATED

```

**Additional information and things you've tried:**

I also noticed that the ADD-ON button navigates to the wrong addon.
On page it is:
[image]
but the new version (1.19) installed from https://github.com/esphome/hassio is available at http://192.168.0.138:8123/15ef4d2f_esphome/dashboard

So in docs the id is `a0d7b954` but the new one actually is `15ef4d2f`

OttoWinter, 2021-06-18:
This looks like a bug in ha supervisor/community addons.

If you see [here](https://github.com/hassio-addons/repository/tree/master/esphome) (the community addons index), the 1.19.1 version is already deployed.

Please check you have the latest supervisor version installed and if there are any errors in the supervisor logs.

jesserockz, 2021-06-22:
There is also a `reload` option in the supervisor addon store menu:

[image]

If that does not work, please check the supervisor logs for errors reloading the addon repo.

Misiu, 2021-06-24:
@jesserockz I just tried that.
I'm now able to update to the latest version using the official addon (from https://addons.community).
Not sure what was the cause, but now everything works perfectly.
@jesserockz, @OttoWinter  thank you for your reply.

closed 2021-06-24 by jesserockz