esphome/issues #2146: Update 1.19 crash connect to wi-fi with power_mode=none

Aliskin-Papa, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker Supervised

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32S

**ESPHome version (latest production, beta, dev branch)**
1.19

**Affected component:**

**Description of problem:**
After update to 1.19 next section make esp32 not connected to wi-fi:
wifi:
  power_save_mode: none
    

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Malachi80, 2021-06-17:
Can confirm the issue (M5Stick-C and M5Atom-lite).

baskapteijn, 2021-06-18:
This seems to be the same issue as I reported yesterday morning: https://github.com/esphome/issues/issues/2141

`power_save_mode` and `ble_tracker` seems to cause problems when used together.

OttoWinter, 2021-06-18:
Merging issue into https://github.com/esphome/issues/issues/2141 as that has more discussions

closed 2021-06-18 by OttoWinter