esphome/issues #2147: ESPHome 1.19 esp8266 fails to connect to wifi if restore from flash is true

slipx06, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Sonoff Basic

**ESPHome version (latest production, beta, dev branch)**

1.19

**Affected component:**

**Description of problem:**
After upgrading to 1.19 my device wont connect to the wifi network if `esp8266_restore_from_flash: true` is set.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: $devicename
  platform: ESP8266
  board: esp8285
  arduino_version: 2.4.2
  esp8266_restore_from_flash: true

```

**Logs (if applicable):**

```
INFO Starting log output from 192.168.1.107 using esphome API
INFO Connecting to 192.168.1.107:6053 (192.168.1.107)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.107: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to 192.168.1.107:6053 (192.168.1.107)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.107: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to 192.168.1.107:6053 (192.168.1.107)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.107: [Errno 111] Connection refused). Re-Trying in 2 seconds
INFO Connecting to 192.168.1.107:6053 (192.168.1.107)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.107: [Errno 111] Connection refused). Re-Trying in 3 seconds
INFO Connecting to 192.168.1.107:6053 (192.168.1.107)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.1.107: [Errno 111] Connection refused). Re-Trying in 5 seconds
INFO Connecting to 192.168.1.107:6053 (192.168.1.107)

```

**Additional information and things you've tried:**

dattas, 2021-06-18:
Having a the same problem (same configuration as above but I have `esp8266_restore_from_flash: false`).
Here are the logs when connected to serial:
```[21:06:30]
[21:06:30] ets Jan  8 2013,rst cause:4, boot mode:(3,7)
[21:06:30]
[21:06:30]wdt reset
[21:06:30]load 0x4010f000, len 1384, room 16 
[21:06:30]tail 8
[21:06:30]chksum 0x2d
[21:06:30]csum 0x2d
[21:06:30]vbb28d4a3
[21:06:30]~ld
[21:06:30]
[21:06:30]SDK:2.2.1(cfd48f3)/Core:2.4.2/lwIP:2.0.3(STABLE-2_0_3_RELEASE/glue:arduino-2.4.1-13-g163bb82)/BearSSL:6d1cefc
[21:06:30][I][logger:170]: Log initialized
[21:06:30][C][status_led:014]: Setting up Status LED...
[21:06:30][C][ota:366]: There have been 9 suspected unsuccessful boot attempts.
[21:06:30][I][app:029]: Running through setup()...
[21:06:30][V][app:030]: Sorting components by setup priority...
[21:06:30][C][switch.gpio:011]: Setting up GPIO Switch 'Test Sonoff'...
[21:06:30][D][switch:025]: 'Test Sonoff' Turning OFF.
[21:06:30][D][switch:045]: 'Test Sonoff': Sending state OFF
[21:06:30][D][light:275]: 'led' Setting:
[21:06:30][D][light:314]:   Transition Length: 1.0s
[21:06:30][D][switch:025]: 'Test Sonoff' Turning OFF.
[21:06:30][C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[21:06:30][C][light:097]: Setting up light 'led'...
[21:06:30][D][light:275]: 'led' Setting:
[21:06:30][D][light:288]:   Brightness: 100%
[21:06:30][C][wifi:037]: Setting up WiFi...
[21:06:30][V][wifi_esp8266:054]: Disabling AP.
[21:06:30]bcn 0
[21:06:30]del if1
[21:06:30]usl
[21:06:30]mode : null
[21:06:30][V][wifi_esp8266:044]: Enabling STA.
[21:06:30]mode : sta(cc:50:[V][wifi_esp8266:492]: Event: Changed Mode old=AP new=OFF
[21:06:30]wifi evt: 8
[21:06:30]e3:02:b5:9f)
[21:06:30]add if0
[21:06:30][V][wifi_esp8266:492]: Event: Changed Mode old=OFF new=STA
[21:06:30]wifi evt: 8
[21:06:30]sleep disable
[21:06:30][D][wifi:365]: Starting scan...
[21:06:30][D][binary_sensor:034]: 'btn_relay': Sending initial state OFF
[21:06:36]scandone
[21:06:36][D][wifi:380]: Found networks:
<redacted>
[21:06:36][I][wifi:235]: WiFi Connecting to '<>'...
[21:06:36][V][wifi:237]: Connection Params:
[21:06:36][V][wifi:238]:   SSID: '<redacted>'
[21:06:36][V][wifi:241]:   BSSID: <redacted>
[21:06:36][V][wifi:261]:   Password: '<redacted>'
[21:06:36][V][wifi:266]:   Channel: 11
[21:06:36][V][wifi:276]:   Using DHCP IP
[21:06:36][V][wifi:278]:   Hidden: NO
[21:06:36][V][wifi_esp8266:451]: Event: Changed AuthMode old=OPEN new=WPA2 PSK
[21:06:36]wifi evt: 2
[21:06:37]scandone
[21:06:38]state: 0 -> 2 (b0)
[21:06:38]state: 2 -> 3 (0)
[21:06:38]state: 3 -> 5 (10)
[21:06:38]add 0
[21:06:38]aid 42
[21:06:38]cnt 
[21:06:38]
[21:06:38]connected with <redacted>, channel 11
[21:06:38]dhcp client start...
[21:06:38][V][wifi_esp8266:432]: Event: Connected ssid='<redacted>' bssid=<redacted> channel=11
[21:06:38]wifi evt: 0
[... 97 lines cut ...]
[21:06:53][C][status_led:019]: Status LED:
[21:06:53][C][status_led:020]:   Pin: GPIO13 (Mode: OUTPUT, INVERTED)
[21:06:53][C][wifi:484]: WiFi:
[21:06:53][C][wifi:344]:   SSID: '<redacted>'
[21:06:53][C][wifi:345]:   IP Address: 10.1.61.244
[21:06:53][C][wifi:347]:   BSSID: <redacted>
[21:06:53][C][wifi:348]:   Hostname: 'test_sonoff'
[21:06:53][C][wifi:352]:   Signal strength: -63 dB ▂▄▆█
[21:06:53][V][wifi:354]:   Priority: 0.0
[21:06:53][C][wifi:356]:   Channel: 11
[21:06:53][C][wifi:357]:   Subnet: 255.255.192.0
[21:06:53][C][wifi:358]:   Gateway: 10.1.0.1
[21:06:53][C][wifi:359]:   DNS1: 10.1.0.8
[21:06:53][C][wifi:360]:   DNS2: 10.1.0.9
[21:06:53][C][logger:189]: Logger:
[21:06:53][C][logger:190]:   Level: VERBOSE
[21:06:53][C][logger:191]:   Log Baud Rate: 115200
[21:06:53][C][logger:192]:   Hardware UART: UART0
[21:06:53][C][captive_portal:148]: Captive Portal:
[21:06:53][C][ota:029]: Over-The-Air Updates:
[21:06:53][C][ota:030]:   Address: test_sonoff.local:8266
[21:06:53][C][ota:032]:   Using Password.
[21:06:53][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[21:07:02]pm open,type:0 0
```

0x3333, 2021-06-29:
Reverting to 1.18 fixed my issues. Looks like a problem with 1.19. Mine didn't connect to wifi.

oxan, 2021-08-10:
See #2309, and try the instructions there. Especially try removing `arduino_version`, as everyone reporting this issue seems to be using ancient versions.

slipx06, 2021-08-10:
> See #2309, and try the instructions there. Especially try removing `arduino_version`, as everyone reporting this issue seems to be using ancient versions.

That fixed the issue for me

closed 2021-12-27 by github-actions[bot]