esphome/issues #2148: Update to v1.19.0 - Install via browser doesn't appear to rebuild (or doesn't complete upload)

logiclabs, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** HomeAssistant / ESPHome Web UI

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32

**ESPHome version (latest production, beta, dev branch):** v1.19.0

**Affected component:**

**Description of problem:**
Just updated to v1.19.0. After making changes to YAML in the web UI, saving the changes and then clicking the new "Install" > To Local browser" browser option - The firmware update appears to go through the motions for 2 minutes, but when my device restarts, it has retained the existing firmware.  Performing an update OTA/wirelessly works OK.

**Problem-relevant YAML-configuration entries:** Not applicable
```yaml
```

**Logs (if applicable):** Can't see logs for Install

```
```

**Additional information and things you've tried:**

balloob, 2021-06-17:
Can you open the browser console for information?

logiclabs, 2021-06-17:
Yes, I can see it's doing a hard reset and uploading a flash binary OK.  After restart, it's still running the an old version of the config.  There aren't any errors being logged. 

Below is a log of 3 separate, different edit + uploads. The last one says the image size is different, but none of the changes made are present on the device after upload and hard reset via the WebSerial installer.

[image]

I then tried an OTA/wireless install .. at this point, the firmware appears to already be compiled (with the new changes), so it looks like that is working OK.  It is also reporting the same size as reported in the browser based install console log.  But when the device now uploads and hard resets, the changes are available on the device.  

[image]

logiclabs, 2021-06-17:
I think I may have found the issue, and doesn't appear to be strictly related to WebSerial.  

My ESP32 device has an external UART sensor configured and has RX/TX pins mapped in the ESPHome config.  Either removing the power to the UART sensor or disconnecting the RX/TX pins appears to solve the issue.     

Plugging both power/UART back in and the WebSerial upload doesn't work again.    

I will see if I get same issue with a standard serial upload.

logiclabs, 2021-06-19:
After more investigation, it appears this was an issue specific to my config/wiring. Having a device connected to GPIO12 on the ESP32 can cause flashing/boot issues. I guess these don't affect OTA updates.

closed 2021-06-19 by logiclabs