esphome/issues #2149: Tuya datapoint is not expected size / unknown datapoint error

CurrentThread, 2021-06-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

```yaml
  platform: ESP8266
  board: esp01_1m
```

**ESPHome version (latest production, beta, dev branch)**

1.19.0

**Affected component:**

https://esphome.io/components/tuya.html
https://esphome.io/components/fan/tuya.html

**Description of problem:**

3rd party Tuya based fan, flashed with ESPHome.

Following available datapoints in log:
```
[19:29:37][C][tuya:026]: Tuya:
[19:29:37][C][tuya:037]:   Datapoint 1: switch (value: ON)
[19:29:37][C][tuya:043]:   Datapoint 2: enum (value: 0)
[19:29:37][C][tuya:043]:   Datapoint 4: enum (value: 0)
[19:29:37][C][tuya:043]:   Datapoint 5: enum (value: 0)
```
Prior to version 1.19.0 I was able to control the fan speed on (the not listed) Datapoint 3. 

Now with 1.19.0 changing fan speed is not possible anymore on Datapoint 3. Changing speed from within HA results in the ESPHome error:
```
[19:33:28][E][tuya:405]: Attempt to set unknown datapoint 3
```

Still can control the fan speed with the remote that comes with the device, but that also gives an ESPHome warning:
```
[19:34:56][W][tuya:235]: Datapoint 3 is not expected size
```

**Problem-relevant YAML-configuration entries:**
```yaml

uart:
  rx_pin: GPIO13
  tx_pin: GPIO15
  baud_rate: 9600

tuya:

fan:
  - platform: tuya
    name: fan01
    switch_datapoint: 1
    speed_datapoint: 3
    oscillation_datapoint: 4
    direction_datapoint: 5
    speed_count: 26
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

aidbish, 2021-06-17:
Just tested with latest dev version and seeing the same thing

[image]

jesserockz, 2021-06-18:
The incoming data for datapoint 3 looks broken:
```
[20:39:41][V][tuya:105]: Received Tuya: CMD=0x07 VERSION=3 DATA=[03.04.00.01.09.00.00.00 (8)] INIT_STATE=5
```

aidbish, 2021-06-18:
This is only recenlty appeared on the latest builds for me, 1.18 and dev 1.19 was ok, just updating to 1.20 dev build this now appears

CurrentThread, 2021-06-18:
Although maybe not according to specifications, there are implementation differences between manufacturers. With 1.19 being stricter in datapoint validation it is causing issues now.

jesserockz, 2021-06-18:
@aidbish Can you send a log with VERY_VERBOSE so we can see the incoming data?

I also have a tuya heater that uses a whole different protocol that is not found on the internet yet.

aidbish, 2021-06-18:
hopefully i have collected enough. i tried turning the kettle on and off via ha to see if they got additional info

[22:28:08][VV][scheduler:152]: Running interval 'wifi' with interval=1000 last_execution=23098 (now=24098)
[22:28:09][VV][scheduler:152]: Running interval 'wifi' with interval=1000 last_execution=24098 (now=25098)
[22:28:10][VV][scheduler:152]: Running interval 'wifi' with interval=1000 last_execution=25098 (now=26098)
[22:28:10][VV][uart_esp8266:132]:     Read 0b01010101 (0x55)
[22:28:10][VV][uart_esp8266:132]:     Read 0b10101010 (0xAA)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000111 (0x07)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00011100 (0x1C)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000101 (0x05)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000010 (0x02)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000100 (0x04)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00101001 (0x29)
[22:28:10][VV][uart_esp8266:132]:     Read 0b01100101 (0x65)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b01100110 (0x66)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000100 (0x04)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b01100111 (0x67)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000100 (0x04)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:10][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:10][VV][uart_esp8266:132]:     Read 0b10010111 (0x97)
[22:28:10][V][tuya:105]: Received Tuya: CMD=0x07 VERSION=0 DATA=[01.01.00.01.00.05.02.00.04.00.00.00.29.65.01.00.01.00.66.04.00.01.00.67.04.00.01.00 (28)] INIT_STATE=5
[22:28:10][W][tuya:235]: Datapoint 1 is not expected size
[22:28:11][VV][scheduler:152]: Running interval 'wifi' with interval=1000 last_execution=26098 (now=27098)
[22:28:11][VV][scheduler:152]: Running interval 'heartbeat' with interval=10000 last_execution=17149 (now=27149)
[22:28:11][V][tuya:321]: Sending Tuya: CMD=0x00 VERSION=0 DATA=[04 (1)] INIT_STATE=5
[22:28:11][VV][uart_esp8266:111]:     Wrote 0b01010101 (0x55)
[22:28:11][VV][uart_esp8266:111]:     Wrote 0b10101010 (0xAA)
[22:28:11][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[22:28:11][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[22:28:11][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[22:28:11][VV][uart_esp8266:111]:     Wrote 0b00000001 (0x01)
[22:28:11][VV][uart_esp8266:111]:     Wrote 0b00000100 (0x04)
[22:28:11][VV][uart_esp8266:101]:     Wrote 0b00000100 (0x04)
[22:28:11][VV][uart_esp8266:132]:     Read 0b01010101 (0x55)
[22:28:11][VV][uart_esp8266:132]:     Read 0b10101010 (0xAA)
[22:28:11][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:11][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:11][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:28:11][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:11][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:11][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:28:11][V][tuya:105]: Received Tuya: CMD=0x00 VERSION=0 DATA=[01 (1)] INIT_STATE=5
[22:28:11][V][tuya:122]: MCU Heartbeat (0x01)
[22:28:12][VV][scheduler:152]: Running interval 'wifi' with interval=1000 last_execution=27098 (now=28100)
[22:28:12][VV][api.service:220]: on_ping_request: PingRequest {}
[22:28:12][VV][api.service:032]: send_ping_response: PingResponse {}
[22:28:13][VV][scheduler:152]: Running interval 'wifi' with interval=1000 last_execution=28098 (now=29104)
[22:28:14][VV][scheduler:152]: Running interval 'wifi' with interval=1000 last_execution=29098 (now=30098)
[22:28:14][VV][api.service:220]: on_ping_request: PingRequest {}
[22:28:14][VV][api.service:032]: send_ping_response: PingResponse {}
[... 566 lines cut ...]
[22:29:00][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:29:00][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:29:00][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:29:00][VV][uart_esp8266:132]:     Read 0b01100111 (0x67)
[22:29:00][VV][uart_esp8266:132]:     Read 0b00000100 (0x04)
[22:29:00][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:29:00][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[22:29:00][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[22:29:00][VV][uart_esp8266:132]:     Read 0b10010111 (0x97)
[... cut ...]
[22:29:01][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[22:29:01][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[22:29:01][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[22:29:01][VV][uart_esp8266:111]:     Wrote 0b00000001 (0x01)
[22:29:01][VV][uart_esp8266:111]:     Wrote 0b00000100 (0x04)
[22:29:01][VV][uart_esp8266:101]:     Wrote 0b00000100 (0x04)
[22:29:01][VV][uart_esp8266:132]:     Read 0b01010101 (0x55)
[22:29:01][VV][uart_esp8266:132]:     Read 0b10101010 (0xAA)
[22:29:01][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)

nuttytree, 2021-06-25:
I have seen a similar issue with Feit dimmers.  On startup a datapoint report command is sent but the response to that only includes the dimmer datapoint (2) not the switch datapoint (1).  As a result the switch cannot be controlled from HA until it has been turned on/off at the switch triggering the send of a switch datapoint value.  I can handle doing the work to resolve the unknown datapoint error but a decision needs to be made on how it should be resolved:

1. When sending data for an "unknown" datapoint create an entry for it in datapoints_ and send, potentially with a warning
2. Add an optional configuration option to enable the behavior in 1.
3. Add an optional configuration option to enable the behavior in 1. for specific datapoints
4. Some other behavior I have not considered?

nuttytree, 2021-06-25:
I did a little deeper digging into the code and found that for string datapoints the code actually tries to send the data even if it is an "unknown" datapoint, however it doesn't actually work because it is getting the datapoint Id and Type from the datapoint retrieved from this->datapoints_ which has no value.  For numeric datapoints (including boolean, enum, integer, etc.) it explicitly does not send the datapoint because it doesn't know what size the data should be.  For string datapoints I think we can probably update the code to use the datapoint Id that was passed into set_datapoint_value and hardcode the type to TuyaDatapointType::STRING.  To fix my issue with the Feit dimmers I updated my local version of the Tuya component with a new override for set_datapoint_value(uint8_t datapoint_id, bool value) that does send the datapoint when it is "unknown" and uses the passed in datapoint_id and a hard code type of TuyaDatapointType::BOOLEAN.  However for the other numeric data I am not sure there is anything we can do as I am assuming sending the wrong size data causes issues also.

aidbish, 2021-06-25:
Can't we just rollback the change that broke this in the first place. It was working and in recent updates it's not?

nuttytree, 2021-06-25:
I think the problem is some of this is needed for some of the newer supported Tuya device types.  However I did not write the code so I can't answer that for sure.

trvrnrth, 2021-07-04:
@nuttytree I quite like the idea of being able to define specific datapoints in configuration as it avoids any guessing games later about what type the device is expecting on send whilst also maintaining the ease of use the simple setters provide when wiring up more complex functionality in lambdas (here's my [thermostat config](https://pastebin.com/qatJjG97) for example).

It wouldn't be particularly difficult to add back the ability to set a datapoint by way of providing a `TuyaDatapoint` (which is what the components used to do) but it does tie them to specific types / data lengths. Perhaps that's fine, but it seemed much nicer to me to make use of the known reported type/length to avoid potential issues. Of course I didn't know at the time that some devices failed to report them all properly!

With respect to parsing incoming datapoints I guess it would probably be safe enough to respect the reported length and ignore any trailing bytes in the message with a warning. This isn't a new problem so will probably have been affecting some devices for a while.

trvrnrth, 2021-07-04:
On that last point I've PRed handling of datapoint messages that contain trailing bytes in https://github.com/esphome/esphome/pull/1982

nuttytree, 2021-07-21:
I have been thinking about all of this and the core problem is trying to determine the type and size of the datapoint to send on the type and size of the first datapoint received.  This presents issues (as we have seen) when we either have not received a datapoint yet or when the data we are sending doesn't match.  I believe we are also seeing issues when receiving datapoints that are a different size than the first datapoint (feel free to correct me if I am wrong here).  I am also wondering if the "size" of all sent data that can be cast as a uint32_t (uint16_t, uint8_t, bool, etc.) ends up being seen as 4 bytes because the parameter type is uint32_t (I don't have a lot of C++ experience so I could definitely be wrong here).  The long and short of this is that I think we need to determine the answers to the following questions before we can come up with a comprehensive fix:
* Can and does the Tuya MCU send data of varying sizes for datapoints with the same id
* Can the Tuya MCU accept data that is a different size than it was "expecting" (does it even have any size expectations)
Assuming the answer to both of these questions is yes I propose the following changes:
* Don't do validation on the size of incoming datapoints
* Update the method `set_datapoint_value(uint8_t datapoint_id, uint32_t value)` to be `set_datapoint_value(uint8_t datapoint_id, uint32_t value, TuyaDatapointType datapoint_type)` and update it to not validate size but validate that datapoint_type is valid for an int type (not string, etc.), and validate the datapoint_type against the type of a received datapoint of the same id IF one has been received.
* Add the method `set_datapoint_value(uint8_t datapoint_id, bool value)` that hard codes the datapoint type to TuyaDatapointType::BOOL and validates that type against the type of a received datapoint of the same id IF one has been received.
* Update the method set_datapoint_value(uint8_t datapoint_id, std::string value) to hard code the datapoint type to TuyaDatapointType::STRING and validates that type against the type of a received datapoint of the same id IF one has been received.
While I have been able to work around this issue I am hoping to get some discussion going on this so that we can fix everyone that is not able to work around it.

trvrnrth, 2021-07-21:
If you haven't already it's worth checking out the [protocol docs](https://developer.tuya.com/en/docs/iot/tuya-cloud-universal-serial-port-access-protocol?id=K9hhi0xxtn9cb#title-2-Status%20data%20unit). You'll see that for the "numeric" types ints are always 4 bytes, enums and bools 1, and bitmaps 1, 2 or 4. Hence we can simply check against the length recorded when we received a datapoint. If it weren't for the bitmaps we could infer it from just the recorded type but you'd end up in exactly the same place.

The annoying thing of course is that seemingly not all devices correctly report all datapoints up-front as they should, in which case when we come to send we don't always know what type we're dealing with.

What you're proposing to allow providing a fallback/default type when setting datapoints that will be used if the device hasn't reported a different type already doesn't sound like a bad approach in general to me.

I had been thinking that it would be neat to allow defining the "missing" datapoints via yaml config which would always allow full flexibility but there'd be nothing stopping that from being a later addition if needed as doing so wouldn't conflict with the concept of a default type when setting datapoints.

jesserockz, 2021-07-21:
Well with the previous code, before the changes that use the incoming data as a map to encode outgoing data, all of the datapoint types were hardcoded into each platform, and this worked fine as we always know a switch is going to be a `bool`, or a temperature will be an `int` etc. 

I know being dynamic can be better, but is there a specific reason not to go back to the hardcoded types? as the code should already know the type of the value it is trying to set.

trvrnrth, 2021-07-22:
The primary motivation for the new setters was to simplify usage from lambdas so that config for other devices or extra functionality was less long-winded. However, in light of the fact that not all devices report their datapoints as they should on startup the components do need a way of specifying a type to set as you say.

This could be achieved either by adding back the ability to set datapoints by way of providing a full `TuyaDatapoint` as used to be the case or by adding the ability to specify an optional fallback/default type in the "new setter style" as described above.

nuttytree, 2021-07-23:
@jesserockz If we can get these 2 PR's merged and included in a 1.20.x version we should have all of the issues resolved.

closed 2021-08-10 by jesserockz