esphome/issues #2152: Midea AC screen blinking after 1.19.1 update

wouterbouvy, 2021-06-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

System: Home Assistant OS 6.0  (amd64 / qemux86-64)
Home Assistant Core: 2021.6.5
Home Assistant Supervisor: 2021.06.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

 platform: ESP8266
 board: esp01_1m

**ESPHome version (latest production, beta, dev branch)**
1.19.1

**Affected component:**
Midea_ac

**Description of problem:**

After updating to 1.19.1 which introduced "fixed" query frame in https://github.com/esphome/esphome/pull/1922
my screen on the AC blinks with every query while on

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: midea_airco_slaapkamer
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "SD-IoT"
  password: "0625225590"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Midea Airco Slaapkamer"
#    password: "OqGu44NyS6j9"

#captive_portal:

# Disable logging over UART
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  password: "ub3JfYhGAW2K!aG@LyaP"

ota:
  password: "ub3JfYhGAW2K!aG@LyaP"

uart:
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600

sensor:
  - platform: uptime
    name: "Airco slaapkamer uptime"
    id: uptime_sensor
#    internal: true
    update_interval: 3600s

climate:
  - platform: midea_ac
    name: "Midea AC slaapkamer"
    beeper: false
    outdoor_temperature:
      name: "Temperatuur airco slaapkamer buitenunit"
      filters:
[... 226 lines cut ...]
[08:42:43][D][midea_ac:034]: RX: query frame
[08:42:44][D][midea_ac:072]: TX: query
[08:42:44][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[08:42:44][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 01 27 65 7F 7F 00 00 00 00 00 63 5A 0B 70 46 00 00 00 00 00 00 00 00 FF 63 
[08:42:44][D][midea_ac:034]: RX: query frame
[08:42:45][D][midea_ac:072]: TX: query
[08:42:45][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[08:42:45][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 01 27 65 7F 7F 00 00 00 00 00 63 5A 0B 00 46 00 00 00 00 00 00 00 00 02 D0 
[08:42:45][D][midea_ac:034]: RX: query frame
[08:42:46][D][midea_ac:072]: TX: query
[08:42:46][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[08:42:46][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 01 27 65 7F 7F 00 00 00 00 00 63 5A 0B 70 46 00 00 00 00 00 00 00 00 FF 63 
[08:42:46][D][midea_ac:034]: RX: query frame
[08:42:47][D][midea_ac:072]: TX: query
[08:42:47][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[08:42:47][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 01 27 65 7F 7F 00 00 00 00 00 63 5A 0B 00 46 00 00 00 00 00 00 00 00 02 D0 
[08:42:47][D][midea_ac:034]: RX: query frame
[08:42:48][D][midea_ac:072]: TX: query
[08:42:48][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[08:42:48][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 01 27 65 7F 7F 00 00 00 00 00 63 5A 0B 70 46 00 00 00 00 00 00 00 00 FF 63 
[08:42:48][D][midea_ac:034]: RX: query frame

```

**Additional information and things you've tried:**

dudanov, 2021-06-18:
@wouterbouvy please test [fix](https://github.com/dudanov/esphome/tree/midea-new-query)

wouterbouvy, 2021-06-18:
@dudanov fixes blinking for me. My HA intergration doesn't work anymore though. Gives me:

ValueError: 6 is not a valid ClimateMode

Is https://github.com/esphome/esphome/pull/1933 already in your fork?

dudanov, 2021-06-18:
> @dudanov fixes blinking for me. My HA intergration doesn't work anymore though. Gives me:
> 
> ValueError: 6 is not a valid ClimateMode
> 
> Is [esphome/esphome#1933](https://github.com/esphome/esphome/pull/1933) already in your fork?

And now: https://github.com/dudanov/esphome/tree/midea-fix-query

wouterbouvy, 2021-06-18:
Yep. Now it all works again!
Thanks for the work @dudanov & @jesserockz
I think this one is ready for merging.

dudanov, 2021-06-18:
@jesserockz @wouterbouvy esphome/esphome/pull/1940

javierdc, 2021-06-18:
After this fix, my midea U AC beeps on each quey frame even if I set beeper: false

dudanov, 2021-06-18:
> After this fix, my midea U AC beeps on each quey frame even if I set beeper: false

Are you sure there is no configuration error or you haven't updated the firmware?
Send your YAML config and logs.

javierdc, 2021-06-19:
> > After this fix, my midea U AC beeps on each quey frame even if I set beeper: false
> 
> Are you sure there is no configuration error or you haven't updated the firmware?
> Send your YAML config and logs.

I'm sure there are no config errors since I didn't changed anything and was working fine, below are the config and logs:

climate:
  - platform: midea_ac
    name: ${friendly_name} Midea AC
    id: ${device_id}_midea_ac
    visual:
      min_temperature: 60 °F
      max_temperature: 86 °F
      temperature_step: 1
    beeper: false
    swing_horizontal: true
    swing_both: false
    outdoor_temperature:
      name: ${friendly_name} Outdoor Temperature
      id: ${device_id}_outdoor_temperature
    power_usage:
      name: ${friendly_name} Power Usage
      id: ${device_id}_power_usage
    humidity_setpoint:
      name: ${friendly_name} Humidity
      id: ${device_id}_humidity

[06:48:14][I][app:105]: ESPHome version 1.19.1 compiled on Jun 18 2021, 08:01:51
[06:48:14][C][wifi:484]: WiFi:
[06:48:14][C][wifi:344]:   SSID: [redacted]
[06:48:14][C][wifi:345]:   IP Address: 192.168.0.150
[06:48:14][C][wifi:347]:   BSSID: [redacted]
[06:48:14][C][wifi:348]:   Hostname: 'dinning-room-midea-ac'
[06:48:14][C][wifi:352]:   Signal strength: -70 dB ▂▄▆█
[06:48:14][C][wifi:356]:   Channel: 6
[06:48:14][C][wifi:357]:   Subnet: 255.255.255.0
[06:48:14][C][wifi:358]:   Gateway: 192.168.0.1
[06:48:14][C][wifi:359]:   DNS1: 192.168.0.1
[06:48:14][C][wifi:360]:   DNS2: (IP unset)
[06:48:14][C][uart_esp8266:075]: UART Bus:
[06:48:14][C][uart_esp8266:077]:   TX Pin: GPIO1
[06:48:14][C][uart_esp8266:080]:   RX Pin: GPIO3
[06:48:14][C][uart_esp8266:081]:   RX Buffer Size: 256
[06:48:14][C][uart_esp8266:083]:   Baud Rate: 9600 baud
[06:48:14][C][uart_esp8266:084]:   Data Bits: 8
[06:48:14][C][uart_esp8266:085]:   Parity: NONE
[06:48:14][C][uart_esp8266:086]:   Stop bits: 1
[06:48:14][C][uart_esp8266:088]:   Using hardware serial interface.
[06:48:14][C][uptime.sensor:030]: Uptime Sensor 'Dining Room Midea AC Uptime Sec'
[06:48:14][C][uptime.sensor:030]:   State Class: ''
[06:48:14][C][uptime.sensor:030]:   Unit of Measurement: 's'
[06:48:14][C][uptime.sensor:030]:   Accuracy Decimals: 0
[06:48:14][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[06:48:14][D][midea_ac:072]: TX: query
[06:48:14][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[06:48:14][C][template.text_sensor:020]: Template Sensor 'Dining Room Midea AC Uptime'
[06:48:14][C][template.text_sensor:020]:   Icon: 'mdi:clock-start'
[06:48:14][C][logger:189]: Logger:
[06:48:14][C][logger:190]:   Level: DEBUG
[06:48:14][C][logger:191]:   Log Baud Rate: 0
[06:48:14][C][logger:192]:   Hardware UART: UART0
[06:48:14][C][homeassistant.time:010]: Home Assistant Time:
[06:48:14][C][homeassistant.time:011]:   Timezone: 'PST8PDT7,M3.2.0/2,M11.1.0/2'
[06:48:14][C][status:034]: Status Binary Sensor 'Dining Room Midea AC Status'
[06:48:14][C][status:034]:   Device Class: 'connectivity'
[06:48:14][C][version.text_sensor:021]: Version Text Sensor 'Dining Room Midea AC ESPhome Version'
[06:48:14][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[06:48:14][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 00 44 66 00 00 00 30 00 00 04 53 49 00 00 00 00 00 00 00 00 00 00 00 0C E5 
[06:48:14][D][midea_ac:034]: RX: query frame
[... 46 lines cut ...]
[06:48:20][D][climate:380]:   Mode: OFF
[06:48:20][D][climate:385]:   Fan Mode: AUTO
[06:48:20][D][climate:391]:   Preset: HOME
[06:48:20][D][climate:397]:   Swing Mode: OFF
[06:48:20][D][climate:400]:   Current Temperature: 17.00°C
[06:48:20][D][climate:406]:   Target Temperature: 20.00°C
[06:48:21][D][midea_ac:072]: TX: query
[06:48:21][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[06:48:21][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 00 44 66 00 00 00 30 00 00 04 54 49 00 70 00 00 00 00 00 00 00 00 00 AF D1 
[06:48:21][D][midea_ac:034]: RX: query frame
[06:48:22][D][midea_ac:072]: TX: query
[06:48:22][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[06:48:22][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 00 44 66 00 00 00 30 00 00 04 54 49 00 00 00 00 00 00 00 00 00 00 00 52 9E 
[06:48:22][D][midea_ac:034]: RX: query frame
[06:48:23][D][midea_ac:072]: TX: query
[06:48:23][D][midea_dongle:094]: TX: AA 22 AC 00 00 00 00 00 03 03 41 21 01 44 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 04 00 17 6A 
[06:48:23][D][midea_dongle:029]: RX: AA 22 AC 00 00 00 00 00 03 03 C1 21 01 C4 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 0B 7A 
[06:48:24][D][midea_ac:072]: TX: query
[06:48:24][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[06:48:24][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 00 44 66 00 00 00 30 00 00 04 54 49 00 70 00 00 00 00 00 00 00 00 00 AF D1 
[06:48:24][D][midea_ac:034]: RX: query frame
[06:48:25][D][midea_ac:072]: TX: query
[06:48:25][D][midea_dongle:094]: TX: AA 21 AC 00 00 00 00 00 00 03 41 61 00 FF 02 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 E3 A8 
[06:48:25][D][midea_dongle:029]: RX: AA 23 AC 00 00 00 00 00 03 03 C0 00 44 66 00 00 00 30 00 00 04 54 49 00 00 00 00 00 00 00 00 00 00 00 52 9E 
[06:48:25][D][midea_ac:034]: RX: query frame

wouterbouvy, 2021-06-19:
Looks like you are on 1.19.1 and haven't installed this fix yet: https://github.com/esphome/esphome/pull/1940

wouterbouvy, 2021-06-19:
@javierdc You should try this repo: https://github.com/dudanov/esphome/tree/midea-fix-query

dudanov, 2021-06-21:
@wouterbouvy issue may closed. v1.19.2 is out.

closed 2021-06-21 by jesserockz