esphome/issues #2153: Constantly disconnecting/reconnecting to Home Assistant on 1.19.0

Jopand, 2021-06-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

v1.19.1

**Affected component:**

**Description of problem:**
Since upgrading from 1.18.x to 1.19.0 and later 1.19.1, I get constant disconnects/reconnect to Home Assistant.
The project was never disconnecting in v.1.18

**Problem-relevant YAML-configuration entries:**

common/base.yaml:
```yaml
esphome:
  name: $device_name
  
  on_boot:
    priority: -10
    then:
    - wait_until:
        api.connected:
    - logger.log: API is connected!

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_password
  ap:
    ssid: "${friendly_name} Fallback Hotspot"
    password: !secret wifi_password

# Enable the captive portal for inital WiFi setup
captive_portal:

# Enable OTA upgrade
ota:

# Enable logging
logger:
  level: ${log_level}

# Enable Home Assistant API
api:

# Enable web server
web_server:
  port: 80
  auth:
    username: !secret web_server_user
    password: !secret web_server_password

binary_sensor:
  - platform: status
    name: "${friendly_name} Status"
    id: system_status

switch:
  - platform: restart
    name: $friendly_name restart

sensor:
[... 311 lines cut ...]
[09:25:23][D][climate:400]:   Current Temperature: 24.30°C
[09:25:23][D][climate:406]:   Target Temperature: 21.00°C
[09:25:23][D][sensor:117]: 'Outdoor temperature': Sending state 22.50000 °C with 1 decimals of accuracy
[09:25:23][D][sensor:117]: 'Humidity': Sending state 49.08000 % with 1 decimals of accuracy
[09:25:23][D][api.connection:644]: Client 'Home Assistant 2021.6.5 (192.168.87.225)' connected successfully!
[09:25:24][D][api:067]: Disconnecting Home Assistant 2021.6.5 (192.168.87.225)
[09:25:25][D][api.connection:644]: Client 'Home Assistant 2021.6.5 (192.168.87.225)' connected successfully!
[09:25:26][D][api:067]: Disconnecting Home Assistant 2021.6.5 (192.168.87.225)
[09:25:26][D][api.connection:644]: Client 'Home Assistant 2021.6.5 (192.168.87.225)' connected successfully!
[09:25:27][D][sensor:117]: 'Nilan Control Uptime Sensor': Sending state 3594.17896 s with 0 decimals of accuracy
[09:25:27][D][sensor:117]: 'Heat exchange efficiency': Sending state 53.64000 % with 1 decimals of accuracy
[09:25:27][D][api:067]: Disconnecting Home Assistant 2021.6.5 (192.168.87.225)
[09:25:28][D][api.connection:644]: Client 'Home Assistant 2021.6.5 (192.168.87.225)' connected successfully!
[09:25:29][D][api:067]: Disconnecting Home Assistant 2021.6.5 (192.168.87.225)
[09:25:29][D][nilan:243]: Control state holding data: 00.0D.00.01.00.03.00.02.08.34 (10)
[09:25:29][D][api.connection:644]: Client 'Home Assistant 2021.6.5 (192.168.87.225)' connected successfully!
[09:25:30][D][api:067]: Disconnecting Home Assistant 2021.6.5 (192.168.87.225)
[09:25:30][D][api.connection:644]: Client 'Home Assistant 2021.6.5 (192.168.87.225)' connected successfully!
[09:25:32][D][api:067]: Disconnecting Home Assistant 2021.6.5 (192.168.87.225)
[09:25:32][D][api.connection:644]: Client 'Home Assistant 2021.6.5 (192.168.87.225)' connected successfully!
[09:25:32][D][nilan:304]: Version info data: 00.08.2E.32.31.33.64.2E (8)

```

**Additional information and things you've tried:**

AussiSG, 2021-06-18:
Mine stopped working totally.

It's connected to my router. Can ping it. Can upload a .yaml to it.

But I'm unable to connect to it through ESPhome to read out the logs nor is it available in the Home Assitant

OttoWinter, 2021-06-18:
Might be related to https://github.com/esphome/esphome/pull/1789 - can you test if the following works?

```yaml
esphome:
  arduino_version: 1.0.4
```

Jopand, 2021-06-18:
Just tested with your suggestion @OttoWinter. It didn't solve the issue, unfortunately

Jopand, 2021-06-18:
FYI, the "nilan" component I'm running is a custom component, that I'm actively developing (https://github.com/Jopand/custom_components). However I don't think it is the issue, since it has been working perfectly until the 1.19 version

HoLy1337, 2021-06-18:
I have the same problem on a sonoff power plug...
Before the update it worked fine!

substitutions:
  devicename: lampe_rechts
  friendlyName: Rechte Lampe
esphome:
  name: lampe_rechts
  platform: ESP8266
  board: esp8285
  arduino_version: platformio/espressif8266@1.8.0
  build_path: lampe_rechts
  platformio_options: {}
  esp8266_restore_from_flash: false
  board_flash_mode: dout
  includes: []
  libraries: []
  name_add_mac_suffix: false
light:
- platform: binary
  name: Rechte Lampe
  output: s20_relay
  id: switch1
  restore_mode: RESTORE_DEFAULT_OFF
output:
- platform: gpio
  id: s20_relay
  pin:
    number: 12
    mode: OUTPUT
    inverted: false
binary_sensor:
- platform: gpio
  pin:
    number: 0
    mode: INPUT_PULLUP
    inverted: true
  id: btn
  on_press:
  - then:
    - light.toggle:
        id: switch1
  name: btn
  internal: true
wifi:
  fast_connect: true
  ap:
    ssid: lampe_rechts
    password: !secret 'wifi_password'
    manual_ip:
      static_ip: 10.0.0.1
      gateway: 10.0.0.1
      subnet: 255.255.255.252
      dns1: 0.0.0.0
      dns2: 0.0.0.0
    ap_timeout: 1min
  enable_mdns: true
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  output_power: 20.0
  networks:
  - ssid: !secret 'wifi_ssid'
    password: !secret 'wifi_password'
    priority: 0.0
  use_address: lampe_rechts.local
captive_portal: {}
logger:
  level: DEBUG
  baud_rate: 115200
  tx_buffer_size: 512
  hardware_uart: UART0
  logs: {}
  esp8266_store_log_strings_in_flash: true
api:
  password: !secret 'secred_pw'
  port: 6053
  reboot_timeout: 15min
ota:
  password: !secret 'secred_pw'
  safe_mode: true
  port: 8266
  reboot_timeout: 5min
  num_attempts: 10
web_server:
  port: 80
  auth:
    username: admin
    password: !secret 'secred_pw'
  css_url: https://esphome.io/_static/webserver-v1.min.css
  js_url: https://esphome.io/_static/webserver-v1.min.js

nrsfhass, 2021-06-20:
I have the same problem on all my ESPhome that worked flowlessly upgrading from 1.18.x to 1.19.0 and later 1.19.1.

Jopand, 2021-06-29:
My custom climate component was crashing due to the changes in Climate modes in 0.19.x (AUTO -> HEAT_COOL). That was causing the disconnecting every second. So I would say the issue can be closed

closed 2021-06-29 by Jopand