esphome/issues #2154: Climate Bang_Bang: Mode AUTO is not supported by this device

ptr727, 2021-06-18:
Ran 18.0, all well.
Upgraded to 19.0, HA reported error.
Upgraded to 19.1, HA now again reports state, but thermostat is broken, stays on permanently.
`[07:34:03][W][climate:056]:   Mode AUTO is not supported by this device!`

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff TH10

**ESPHome version (latest production, beta, dev branch)**

19.1

**Affected component:**

https://esphome.io/components/climate/bang_bang.html

**Description of problem:**
Thermostat output never turns off.

**Problem-relevant YAML-configuration entries:**
https://github.com/ptr727/ESPHome-Config/blob/master/sonoff_th10_hot_recirc_pump.yaml

**Logs (if applicable):**

```
[07:34:03][D][main:196]: RecircTimer : Delay End.
[07:34:03][D][main:199]: RecircTimer : Setting Pump to Auto.
[07:34:03][D][climate:009]: 'sonoff_th10_hot_recirc_pump_bangbang' - Setting
[07:34:03][W][climate:056]:   Mode AUTO is not supported by this device!
[07:34:03][D][climate:377]: 'sonoff_th10_hot_recirc_pump_bangbang' - Sending state:
[07:34:03][D][climate:380]:   Mode: HEAT
[07:34:03][D][climate:382]:   Action: HEATING
[07:34:03][D][climate:400]:   Current Temperature: 49.90°C
[07:34:03][D][climate:404]:   Target Temperature: Low: 30.00°C High: 40.00°C
```

**Additional information and things you've tried:**
v18.0 worked fine.

ptr727, 2021-06-18:
Seems to be related to this change: https://github.com/esphome/esphome/pull/1933

jesserockz, 2021-06-20:
@ptr727 Can you please change your YAML to use `HEAT_COOL` instead of `AUTO` as `AUTO` has another meaning.

If that is working please close this issue.

ptr727, 2021-06-21:
I already tried HEAT_COOL, see my comment in https://github.com/esphome/esphome/pull/1933, when I use HEAT_COOL the code does not compile.

And I am being picky, but adding HEAT_COOL after the existence of AUTO makes HEAT_COOL different, not AUTO, changing the definition of AUTO, and breaking code, and not documenting the breaking change, is not cool.

```
/config/sonoff_th10_hot_recirc_pump.yaml: In function 'void setup()':
/config/sonoff_th10_hot_recirc_pump.yaml:202:37: error: 'CLIMATE_HEAT_COOL' is not a member of 'esphome::climate'
Archiving .pioenvs/sonoff_th10_hot_recirc_pump/lib189/libESPAsyncTCP-esphome.a
Compiling .pioenvs/sonoff_th10_hot_recirc_pump/lib768/ESPAsyncWebServer-esphome/WebRequest.cpp.o
Archiving .pioenvs/sonoff_th10_hot_recirc_pump/lib6f5/libHash.a
Compiling .pioenvs/sonoff_th10_hot_recirc_pump/lib768/ESPAsyncWebServer-esphome/WebResponses.cpp.o
*** [.pioenvs/sonoff_th10_hot_recirc_pump/src/main.cpp.o] Error 1
```

jesserockz, 2021-06-21:
Sorry about that @ptr727 

I have just released [1.19.2](https://github.com/esphome/esphome/releases/tag/v1.19.2) which fixes the `mode: HEAT_COOL` in the climate call action.

And yes a breaking change going from `AUTO` to `HEAT_COOL` is not cool. But in order to unify the ESPHome and Home Assistant implementations, it was a necessary change. I apologise for any inconveniences it has caused.

closed 2021-06-21 by ptr727