esphome/issues #2156: HTU21D - I2C: Received NACK on transmit of address 0x40 on ESP32

j4m35, 2021-06-19:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10

**ESP (ESP32/ESP8266, Board/Sonoff):**

 ESP32 nodemcu-32s

**ESPHome version v1.19.0**

**Affected component: HTU21D**

https://esphome.io/components/sensor/htu21d.html

**Description of problem:**
ESP32 is unable to communicate with HTU21D via I2C.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: tester
  platform: ESP32
  board: nodemcu-32s

i2c:

sensor:
  - platform: htu21d
    temperature:
      name: "Temperature"
    humidity:
      name: "Humidity"
    update_interval: 60s

# Enable logging
logger:
```

**Logs (if applicable):**

```
Wrote 306032 bytes (136711 compressed) at 0x00010000 in 12.1 seconds (effective 201.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
=============================================================== [SUCCESS] Took 18.48 seconds ===============================================================
←[32mINFO Successfully uploaded program.←[0m
←[32mINFO Starting log output from COM3 with baud rate 115200←[0m
[17:20:32][I][logger:170]: Log initialized
[17:20:32][I][app:029]: Running through setup()...
[17:20:32][C][htu21d:016]: Setting up HTU21D...
[17:20:32][W][i2c:070]: Received NACK on transmit of address 0x40
[17:20:32][E][component:094]: Component was marked as failed.
[17:20:32][I][app:059]: setup() finished successfully!
[17:20:32][I][app:105]: ESPHome version 1.19.0 compiled on Jun 19 2021, 17:20:09
[17:20:32][C][i2c:028]: I2C Bus:
[17:20:32][C][i2c:029]:   SDA Pin: GPIO21
[17:20:32][C][i2c:030]:   SCL Pin: GPIO22
[17:20:32][C][i2c:031]:   Frequency: 50000 Hz
[17:20:32][I][i2c:033]: Scanning i2c bus for active devices...
[17:20:32][I][i2c:049]: Found no i2c devices!
[17:20:32][C][logger:189]: Logger:
[17:20:32][C][logger:190]:   Level: DEBUG
[17:20:32][C][logger:191]:   Log Baud Rate: 115200
[17:20:32][C][logger:192]:   Hardware UART: UART0
[17:20:32][C][htu21d:027]: HTU21D:
[17:20:32][C][htu21d:028]:   Address: 0x40
[17:20:32][E][htu21d:030]: Communication with HTU21D failed!
[17:20:32][C][htu21d:032]:   Update Interval: 60.0s
[17:20:32][C][htu21d:033]:   Temperature 'Temperature'
[17:20:32][C][htu21d:033]:     Device Class: 'temperature'
[17:20:32][C][htu21d:033]:     State Class: 'measurement'
[17:20:32][C][htu21d:033]:     Unit of Measurement: '°C'
[17:20:32][C][htu21d:033]:     Accuracy Decimals: 1
[17:20:32][C][htu21d:034]:   Humidity 'Humidity'
[17:20:32][C][htu21d:034]:     Device Class: 'humidity'
[17:20:32][C][htu21d:034]:     State Class: 'measurement'
[17:20:32][C][htu21d:034]:     Unit of Measurement: '%'
[17:20:32][C][htu21d:034]:     Accuracy Decimals: 1
```

**Additional information and things you've tried:**
1. Tried 2 different HTU21D modules.
2. on 3 different ESP32 nodemcu-32s boards.
3. Tried enabling / disabling 4k7 pull-up resistors on SDA and SCL.
4. Both HTU21D modules are verified to be working by testing on a Raspberry Pi. I.e. able to communicate via I2C and pull temperature and humidity on a Raspberry Pi but not with the ESP32 boards running ESPHome.

agners, 2021-06-22:
The code sends a soft reset on setup. A NACK is not expected there, there seems to be something wrong with the I2C bus. Do you have a different I2C device to try?

j4m35, 2021-06-22:
@agners you were spot on that there was something wrong with the I2C bus! Based on your comment, I changed the wires that I used to thicker ones and it's working with the ESP32 now. I did not think of changing the wires (as they were working on a Raspberry Pi) till I saw your comment; thanks for that.

closed 2021-06-22 by j4m35

agners, 2021-06-22:
Yeah I2C is sometimes a bit brittle. After all, it is meant for on-board communication (Inter-Integrated Circuit)...