esphome/issues #2159: ESPHome constant reboot / disconnects on DHCP

ilium007, 2021-06-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 - Adafruit Huzzah ESP8266 breakout (issue occurs with all ESPxx devices)
PLATFORMIO_ESP01_1M

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

Uptime

**Description of problem:**
Device loses API connection at approx 23-26mins consistently. I do not believe it is wifi / DHCP related as no other device on subnet is affected. Router is a Mikrotik Chateau (routeros 7). DHCP lease time is 5min. If I change the config to use a static IP the devices stay up indefinitely. 

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: "office-test-0"
  name: "Office Test 0"
  
esphome:
  name: office_test_0
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
    - ssid: LIBERTY
      password: !secret liberty_wifi  
  fast_connect: on
  
  #manual_ip:
  #  static_ip: 192.168.10.200
  #  gateway: 192.168.10.1
  #  subnet: 255.255.255.0
  #  dns1: 192.168.10.1
  power_save_mode: none
  domain: .home
  
  #use_address: 192.168.10.21

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Office Test 0 Fallback Hotspot"
    password: "BGwbVrAWKMw4"

captive_portal:

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: "Australia/Sydney"

sensor:
  - platform: uptime
[... 537 lines cut ...]
[0;35m[C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'[0m
[0;35m[C][wifi_signal.sensor:009]:   Accuracy Decimals: 0[0m
[0;35m[C][wifi_info:009]: WifiInfo IPAddress 'office-test-0 IP'[0m
[0;36m[D][api:067]: Disconnecting ESPHome v1.18.0 (192.168.10.12)[0m
[0;36m[D][api.connection:617]: Client 'ESPHome v1.18.0 (192.168.10.12)' connected successfully![0m
[0;36m[D][sensor:099]: 'office-test-0 Wifi Signal': Sending state -53.00000 dBm with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Uptime': Sending state 56.45600 s with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Wifi Signal': Sending state -53.00000 dBm with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Uptime': Sending state 116.45200 s with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Wifi Signal': Sending state -53.00000 dBm with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Uptime': Sending state 176.45700 s with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Wifi Signal': Sending state -53.00000 dBm with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Uptime': Sending state 236.45000 s with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Wifi Signal': Sending state -53.00000 dBm with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Uptime': Sending state 296.45099 s with 0 decimals of accuracy[0m
[0;32m[I][ota:046]: Boot seems successful, resetting boot loop counter.[0m
[0;36m[D][sensor:099]: 'office-test-0 Wifi Signal': Sending state -47.00000 dBm with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Uptime': Sending state 356.45001 s with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Wifi Signal': Sending state -49.00000 dBm with 0 decimals of accuracy[0m
[0;36m[D][sensor:099]: 'office-test-0 Uptime': Sending state 416.45001 s with 0 decimals of accuracy[0m
```

**Additional information and things you've tried:**

[image]

Urs-O, 2021-06-20:
I can confirm the issue, i'm also affected by the Bug. I use a sensor which needs to be powercycled after a reset, so after 20-40 minutes it doesn't deliver anything anymore. I have confirmed it with the uptime sensor, same problem here.

ilium007, 2021-06-20:
[image]

ilium007, 2021-06-20:
I don't know if it is an ESPHome issue or a Home Assistant issue. Not sure what other logs I can get from Home Assistant or whether I can enable some sort of debug logging. 

```
2021-06-20 11:40:44 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 12:03:52 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 12:26:36 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 13:10:27 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 13:34:18 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 13:56:26 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 14:18:57 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 14:42:09 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 15:03:19 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 15:24:51 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 15:46:43 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 16:10:56 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 16:33:05 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 16:55:22 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 17:19:33 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 17:42:16 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 18:06:25 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 18:30:16 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 19:18:18 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 19:40:38 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!
2021-06-20 20:03:25 WARNING (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for None (office_test_0.home): Timeout while waiting for API response!

```

ilium007, 2021-07-09:
Tried v1.20.0-dev with `arduino_version: espressif8266@3.0.0` with no success. Same behavior configured for DHCP vs static IP.

[image]

chemelli74, 2021-07-10:
Investigation will be not so easy, as in some networks (mine for example) connection between device and HA is rock stable for weeks.
Maybe your router says something about disconnections ? 

Simone

ilium007, 2021-07-10:
It’s nothing to do with my router (Mikrotik routerOS). I’ve posted the exception dumps from the device. It’s certainly code / memory related.

ilium007, 2021-07-11:
I've tested with a Ubiquiti EdgeRouter with same results. Both Adafruit Huzzah breakouts, one the ESP8266 and the other is the ESP32.

chemelli74, 2021-07-11:
> I've tested with a Ubiquiti EdgeRouter with same results. Both Adafruit Huzzah breakouts, one the ESP8266 and the other is the ESP32. And this is nothing about HA at this point, the boards are crashing and rebooting.

Sorry then, I cannot be of any help.
I'm on a different board:

```
esphome:
  name: esp32-salotto-ble-gateway
  platform: ESP32
  board: nodemcu-32s
```

Simone

ilium007, 2021-07-11:
I'll try with a different access point to see if I can narrow it down to the wifi.

closed 2021-11-24 by github-actions[bot]

mmakaay, 2021-11-24:
> I’ve posted the exception dumps from the device.

Did you decide the stack trace at some point? It's impossible to decode that without access to the firmware.elf that was used at the time of the stack trace dump.
Connecting to a serial console and using command line `esphome logs` would do the decoding for you.
That should give an idea about where in the code the exception is being triggered