esphome/issues #2160: SSD1306 - OLED display doesn't show anything anymore since 1.17.2

7h30n3, 2021-06-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Python 3.9.5 (tags/v3.9.5:0a7dcbd, May  3 2021, 17:27:52) [MSC v.1928 64 bit (AMD64)] on win32
pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU from Amica (ESP8266)

**ESPHome version (latest production, beta, dev branch)**

1.19.1

**Affected component:**

https://esphome.io/components/display/ssd1306.html#ssd1306-spi

**Description of problem:**
After updating ESPHome from 1.16.2 to 1.19.1 my OLED display (SH1106) doesn't show anything anymore.
Even downgrading to 1.18.0 and 1.17.2 didn't bring any improvement.
After downgrading to 1.16.2 everything works again as expected.
Not matter the version the log doesn't show any warnings or errors.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  platform: ESP8266
  name: sh1106_si7021
  board: nodemcuv2

api:

wifi:
  ssid: ' '
  password: ' '
  manual_ip:
    static_ip: 192.168.1.41
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  fast_connect: true
  power_save_mode: high

# Enable logging
logger:
  level: debug
  baud_rate: 0
  
i2c:

ota:

substitutions:
  device_name: SH1106 Si7021
  device_id: sh1106_si7021

spi:
  clk_pin: 14
  mosi_pin: 13

sensor:
  - platform: wifi_signal
    name: "${device_name} RSSI"
    filters:
        - delta: 2.0
    
  - platform: htu21d
    temperature:
      filters:
        - delta: 0.4
        - offset: -3.4
      name: "${device_name} Temperature"
      unit_of_measurement: "°C"
      id: temperature
    humidity:
      filters:
        - delta: 1.2
      name: "${device_name} Humidity"
      accuracy_decimals: 1
      unit_of_measurement: "%"
      id: humidity
    update_interval: 60s
    
time:
  - platform: homeassistant
    id: time_hass

font:
  - file: "sh1106_si7021_dependencies/arial.ttf"
    id: arial_14
    size: 14
    glyphs: "HeloWrd! "    

display:
  - platform: ssd1306_spi
    model: "SH1106 128x64"
    brightness: 60%
    cs_pin: 15
    dc_pin: 0
    reset_pin: 2
    id: my_display
    lambda: |-
      it.print(0, 0, id(arial_14), "Hello World!");
```

**Logs (if applicable):**

```
[spi:081]: SPI bus:
[spi:082]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[spi:084]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[spi:085]:   Using HW SPI: YES
[ssd1306_spi:019]: SPI SSD1306
[ssd1306_spi:019]:   Rotations: 0 °
[ssd1306_spi:019]:   Dimensions: 128px x 64px
[ssd1306_spi:020]:   Model: SH1106 128x64
[ssd1306_spi:021]:   CS Pin: GPIO15 (Mode: OUTPUT)
[ssd1306_spi:022]:   DC Pin: GPIO0 (Mode: OUTPUT)
[ssd1306_spi:023]:   Reset Pin: GPIO2 (Mode: OUTPUT)
[ssd1306_spi:024]:   External VCC: NO
[ssd1306_spi:025]:   Update Interval: 1.0s

```

**Additional information and things you've tried:**

jesserockz, 2021-06-20:
Can you add a serial log from boot (so that it includes the setup logs)?

7h30n3, 2021-06-21:
```
Here we go:

[09:37:18][C][ota:366]: There have been 2 suspected unsuccessful boot attempts.
[09:37:18][I][app:029]: Running through setup()...
[09:37:18][C][spi:022]: Setting up SPI bus...
[09:37:18][C][htu21d:016]: Setting up HTU21D...
[09:37:18][C][ssd1306_spi:011]: Setting up SPI SSD1306...
[09:37:18][C][wifi:037]: Setting up WiFi...
[09:37:18][I][wifi:235]: WiFi Connecting to 'skynet'...
[09:37:18][D][htu21d:052]: Got Temperature=29.4°C Humidity=63.6%
[09:37:18][D][sensor:117]: 'SH1106 Si7021 Temperature': Sending state 25.98400 °C with 1 decimals of accuracy
[09:37:18][D][sensor:117]: 'SH1106 Si7021 Humidity': Sending state 63.57245 % with 1 decimals of accuracy
[09:37:22][I][wifi:498]: WiFi Connected!
[09:37:22][C][wifi:344]:   SSID: 'skynet'
[09:37:22][C][wifi:345]:   IP Address: 192.168.1.41
[09:37:22][C][wifi:347]:   BSSID: XXXXXXXXXXXXXXX
[09:37:22][C][wifi:348]:   Hostname: 'sh1106_si7021'
[09:37:22][C][wifi:352]:   Signal strength: -64 dB ▂▄▆█
[09:37:22][C][wifi:356]:   Channel: 6
[09:37:22][C][wifi:357]:   Subnet: 255.255.255.0
[09:37:22][C][wifi:358]:   Gateway: 192.168.1.1
[09:37:22][C][wifi:359]:   DNS1: (IP unset)
[09:37:22][C][wifi:360]:   DNS2: (IP unset)
[09:37:22][C][ota:029]: Over-The-Air Updates:
[09:37:22][C][ota:030]:   Address: 192.168.1.41:8266
[09:37:22][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[09:37:22][C][api:022]: Setting up Home Assistant API server...
[09:37:22][I][app:059]: setup() finished successfully!
[09:37:22][D][sensor:117]: 'SH1106 Si7021 RSSI': Sending state -64.00000 dBm with 0 decimals of accuracy
[09:37:22][I][app:105]: ESPHome version 1.19.2 compiled on Jun 21 2021, 09:36:03
[09:37:22][C][wifi:484]: WiFi:
[09:37:22][C][wifi:344]:   SSID: 'skynet'
[09:37:22][C][wifi:345]:   IP Address: 192.168.1.41
[09:37:22][C][wifi:347]:   BSSID: XXXXXXXXXXXXX
[09:37:22][C][wifi:348]:   Hostname: 'sh1106_si7021'
[09:37:22][C][wifi:352]:   Signal strength: -64 dB ▂▄▆█
[09:37:22][C][wifi:356]:   Channel: 6
[09:37:22][C][wifi:357]:   Subnet: 255.255.255.0
[09:37:22][C][wifi:358]:   Gateway: 192.168.1.1
[09:37:22][C][wifi:359]:   DNS1: (IP unset)
[09:37:22][C][wifi:360]:   DNS2: (IP unset)
[09:37:22][C][i2c:028]: I2C Bus:
[09:37:22][C][i2c:029]:   SDA Pin: GPIO4
[09:37:22][C][i2c:030]:   SCL Pin: GPIO5
[09:37:22][C][i2c:031]:   Frequency: 50000 Hz
[09:37:22][I][i2c:033]: Scanning i2c bus for active devices...
[09:37:22][I][i2c:040]: Found i2c device at address 0x40
[09:37:22][C][spi:081]: SPI bus:
[09:37:22][C][spi:082]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[09:37:22][C][spi:084]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[09:37:22][C][spi:085]:   Using HW SPI: YES
[09:37:22][D][api.connection:644]: Client 'Home Assistant 2021.4.6 (192.168.1.220)' connected successfully!
[09:37:22][C][logger:189]: Logger:
[09:37:22][C][logger:190]:   Level: DEBUG
[09:37:22][C][logger:191]:   Log Baud Rate: 115200
[09:37:22][C][logger:192]:   Hardware UART: UART0
[09:37:22][D][time:040]: Synchronized time: Mon Jun 21 09:37:23 2021
[09:37:22][C][htu21d:027]: HTU21D:
[09:37:22][C][htu21d:028]:   Address: 0x40
[09:37:22][C][htu21d:032]:   Update Interval: 60.0s
[09:37:22][C][htu21d:033]:   Temperature 'SH1106 Si7021 Temperature'
[09:37:22][C][htu21d:033]:     Device Class: 'temperature'
[09:37:22][C][htu21d:033]:     State Class: 'measurement'
[09:37:22][C][htu21d:033]:     Unit of Measurement: '°C'
[09:37:22][C][htu21d:033]:     Accuracy Decimals: 1
[09:37:22][C][htu21d:034]:   Humidity 'SH1106 Si7021 Humidity'
[09:37:22][C][htu21d:034]:     Device Class: 'humidity'
[09:37:22][C][htu21d:034]:     State Class: 'measurement'
[09:37:22][C][htu21d:034]:     Unit of Measurement: '%'
[09:37:22][C][htu21d:034]:     Accuracy Decimals: 1
[09:37:22][C][homeassistant.time:010]: Home Assistant Time:
[09:37:22][C][homeassistant.time:011]:   Timezone: 'CET-1CEST-2,M3.4.0/2,M10.5.0/3'
[09:37:22][C][ssd1306_spi:019]: SPI SSD1306
[09:37:22][C][ssd1306_spi:019]:   Rotations: 0 °
[09:37:22][C][ssd1306_spi:019]:   Dimensions: 128px x 64px
[09:37:22][C][ssd1306_spi:020]:   Model: SH1106 128x64
[09:37:22][C][ssd1306_spi:021]:   CS Pin: GPIO15 (Mode: OUTPUT)
[09:37:22][C][ssd1306_spi:022]:   DC Pin: GPIO0 (Mode: OUTPUT)
[09:37:22][C][ssd1306_spi:023]:   Reset Pin: GPIO2 (Mode: OUTPUT)
[09:37:22][C][ssd1306_spi:024]:   External VCC: NO
[09:37:22][C][ssd1306_spi:025]:   Update Interval: 1.0s
[09:37:22][C][ota:029]: Over-The-Air Updates:
[09:37:22][C][ota:030]:   Address: 192.168.1.41:8266
[09:37:22][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[09:37:22][C][api:095]: API Server:
[09:37:22][C][api:096]:   Address: 192.168.1.41:6053
[09:37:22][C][wifi_signal.sensor:009]: WiFi Signal 'SH1106 Si7021 RSSI'
[09:37:22][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[09:37:22][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[09:37:22][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[09:37:22][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[09:37:57][D][htu21d:052]: Got Temperature=29.5°C Humidity=62.8%
```

7h30n3, 2021-06-27:
The Error still persists in 1.19.4.

closed 2021-11-09 by github-actions[bot]