esphome/issues #2162: PZEM-004T V3 Modbus CRC Check failed!

Pelbru, 2021-06-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.): **

 Debian GNU/Linux 10 (buster) / Version Docker 20.10.6 / Version supervisor-2021.06.3 / Version core-2021.6.5

**ESP (ESP32/ESP8266, Board/Sonoff):**
 ESP8266 NodeMcu

**ESPHome version (latest production, beta, dev branch)**
 ESPHome v1.19.1

**Affected component:**
https://esphome.io/components/sensor/pzemac.html

**Description of problem:**
No reading of 3 pzem data. Modbus CRC Check failed! in the logs

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: pzem_ecs
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "secret! esphome_ssid"
  password: "secret! esphome_password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Pzem Ecs Fallback Hotspot"
    password: "GZVhGXrhstXD"

captive_portal:

# Enable logging
logger:
  level: debug
  baud_rate: 0
  
# Enable Home Assistant API
api:

ota:

uart:
  - id: ubus
    tx_pin: GPIO1     # TXD0
    rx_pin: GPIO3     # RXD0
    baud_rate: 9600
    stop_bits: 2      

modbus:
 id: mbus
 uart_id: ubus
  
time:
  platform: homeassistant
  id: homeassistant_time

  on_time:
      - seconds: 57
        minutes: 59
        hours: 23
        then:  
          - sensor.template.publish:
              id: conso_el_ecs_veille
              state: !lambda return id(conso_el_ecs_du_jour).state;
          - sensor.template.publish:
[... 655 lines cut ...]
[11:35:06][D][sensor:117]: 'Puissance Réactive Ph2': Sending state nan VAr with 2 decimals of accuracy
[11:35:06][D][sensor:117]: 'Energie cumulée Ph2': Sending state nan kWh with 1 decimals of accuracy
[11:35:06][D][sensor:117]: 'Up': Sending state 131986.03125 s with 0 decimals of accuracy
[11:35:06][D][sensor:117]: 'Puissance Réactive Ph1': Sending state nan VAr with 2 decimals of accuracy
[11:35:06][D][sensor:117]: 'Energie cumulée Ph3': Sending state nan kWh with 1 decimals of accuracy
[11:35:09][D][sensor:117]: 'Puissance Wifi Pzem': Sending state -60.00000 dBm with 0 decimals of accuracy
[11:35:10][D][text_sensor:015]: 'Uptime Pzem FCS': Sending state '1j 12h 39m 46s'
[11:35:10][D][sensor:117]: 'Energie cumulée Ph1': Sending state nan kWh with 1 decimals of accuracy
[11:35:11][D][sensor:117]: 'Energie cumulée Ph2': Sending state nan kWh with 1 decimals of accuracy
[11:35:11][D][sensor:117]: 'Up': Sending state 131991.03125 s with 0 decimals of accuracy
[11:35:11][D][sensor:117]: 'Energie cumulée Ph3': Sending state nan kWh with 1 decimals of accuracy
[11:35:12][D][sensor:117]: 'Puissance Totale': Sending state nan W with 2 decimals of accuracy
[11:35:12][D][sensor:117]: 'Puissance Apparente Ph2': Sending state nan VA with 2 decimals of accuracy
[11:35:15][D][text_sensor:015]: 'Uptime Pzem FCS': Sending state '1j 12h 39m 51s'
[11:35:15][D][sensor:117]: 'Energie cumulée Ph1': Sending state nan kWh with 1 decimals of accuracy
[11:35:15][D][sensor:117]: 'Puissance Réactive Ph3': Sending state nan VAr with 2 decimals of accuracy
[11:35:15][D][sensor:117]: 'Puissance Apparente Ph3': Sending state nan VA with 2 decimals of accuracy
[11:35:16][D][sensor:117]: 'Puissance Apparente Ph1': Sending state nan VA with 2 decimals of accuracy
[11:35:16][D][sensor:117]: 'Puissance Réactive Ph2': Sending state nan VAr with 2 decimals of accuracy
[11:35:16][D][sensor:117]: 'Energie cumulée Ph2': Sending state nan kWh with 1 decimals of accuracy
[11:35:16][D][sensor:117]: 'Up': Sending state 131996.03125 s with 0 decimals of accuracy
[11:35:16][D][sensor:117]: 'Puissance Réactive Ph1': Sending state nan VAr with 2 decimals of accuracy
[11:35:16][D][sensor:117]: 'Energie cumulée Ph3': Sending state nan kWh with 1 decimals of accuracy

**Additional information and things you've tried:**

jesserockz, 2021-06-22:
Can you enable `VERY_VERBOSE` logging and capture a section where the data comes in and then gives the error?

Pelbru, 2021-06-22:
Hello,
Here are the logs
Best regards
INFO Waiting for result...INFO OTA successfulINFO Successfully uploaded program.INFO Starting log output from pzem_ecs.local using esphome APIINFO Connecting to pzem_ecs.local:6053 (192.168.0.31)INFO Successfully connected to pzem_ecs.local[07:30:36][I][app:105]: ESPHome version 1.19.2 compiled on Jun 22 2021, 07:28:53[07:30:36][C][wifi:484]: WiFi:[07:30:36][C][wifi:344]:   SSID: 'TP-Link_AP_NonoNet'[redacted][07:30:36][C][wifi:345]:   IP Address: 192.168.0.31[07:30:36][C][wifi:347]:   BSSID: D8:47:32:8F:D3:B6[redacted][07:30:36][C][wifi:348]:   Hostname: 'pzem_ecs'[07:30:36][C][wifi:352]:   Signal strength: -59 dB ▂▄▆█[07:30:36][V][wifi:354]:   Priority: 0.0[07:30:36][C][wifi:356]:   Channel: 1[07:30:36][C][wifi:357]:   Subnet: 255.255.255.0[07:30:36][C][wifi:358]:   Gateway: 192.168.0.254[07:30:36][C][wifi:359]:   DNS1: 192.168.0.250[07:30:36][C][wifi:360]:   DNS2: (IP unset)[07:30:36][VV][api.service:099]: send_list_entities_sensor_response: ListEntitiesSensorResponse {
   object_id: 'consommation_ph3'
   key: 2605271820
   name: 'Consommation Ph3'
   unique_id: 'pzem_ecssensorconsommation_ph3'
   icon: ''
   unit_of_measurement: 'Wh'
   accuracy_decimals: 1
   force_update: NO
   device_class: 'energy'
   state_class: STATE_CLASS_NONE
}[07:30:36][C][uart_esp8266:075]: UART Bus:[07:30:36][C][uart_esp8266:077]:   TX Pin: GPIO1[07:30:36][C][uart_esp8266:080]:   RX Pin: GPIO3[07:30:36][C][uart_esp8266:081]:   RX Buffer Size: 256[07:30:36][C][uart_esp8266:083]:   Baud Rate: 9600 baud[07:30:36][C][uart_esp8266:084]:   Data Bits: 8[07:30:36][C][uart_esp8266:085]:   Parity: NONE[07:30:36][C][uart_esp8266:086]:   Stop bits: 2[07:30:36][C][uart_esp8266:088]:   Using hardware serial interface.[07:30:36][VV][api.service:099]: send_list_entities_sensor_response: ListEntitiesSensorResponse {
   object_id: 'frquence_ph3'
   key: 4148657984
   name: 'Fréquence Ph3'
   unique_id: 'pzem_ecssensorfrquence_ph3'
   icon: 'mdi:current-ac'
   unit_of_measurement: 'Hz'
   accuracy_decimals: 2
   force_update: NO
   device_class: ''
   state_class: STATE_CLASS_MEASUREMENT
}[07:30:36][C][modbus:101]: Modbus:[07:30:36][VV][api.service:099]: send_list_entities_sensor_response: ListEntitiesSensorResponse {
   object_id: 'facteur_de_puissance_ph3'
   key: 2613146311
   name: 'Facteur de puissance Ph3'
   unique_id: 'pzem_ecssensorfacteur_de_puissance_ph3'
   icon: ''
   unit_of_measurement: ''
   accuracy_decimals: 2
   force_update: NO
   device_class: 'power_factor'
   state_class: STATE_CLASS_MEASUREMENT
}[07:30:36][C][template.sensor:021]: Template Sensor 'Energie cumulée Ph1'[07:30:36][C][template.sensor:021]:   State Class: ''[07:30:36][C][template.sensor:021]:   Unit of Measurement: 'kWh'[07:30:36][C][template.sensor:021]:   Accuracy Decimals: 1[07:30:36][C][template.sensor:021]:   Icon: 'mdi:power'[07:30:36][C][template.sensor:022]:   Update Interval: 5.0s[07:30:36][VV][api.service:111]: send_list_entities_switch_response: ListEntitiesSwitchResponse {
   object_id: 'restart_pzem_ecs'
   key: 3368780035
   name: 'Restart Pzem ECS'
   unique_id: 'pzem_ecsswitchrestart_pzem_ecs'
   icon: 'mdi:restart'
   assumed_state: NO
}[07:30:36][C][template.sensor:021]: Template Sensor 'Consommation Ph1 de la veille'[07:30:36][C][template.sensor:021]:   State Class: ''[07:30:36][C][template.sensor:021]:   Unit of Measurement: 'kwh'[07:30:36][C][template.sensor:021]:   Accuracy Decimals: 2[07:30:36][C][template.sensor:021]:   Icon: 'mdi:power'[07:30:36][C][template.sensor:022]:   Update Interval: 10.0s[07:30:36][C][template.sensor:021]: Template Sensor 'Puissance Apparente Ph1'[07:30:36][C][template.sensor:021]:   State Class: ''[07:30:36][C][template.sensor:021]:   Unit of Measurement: 'VA'[07:30:36][C][template.sensor:021]:   Accuracy Decimals: 2[07:30:36][C][template.sensor:021]:   Icon: 'mdi:power'[07:30:36][C][template.sensor:022]:   Update Interval: 10.0s[07:30:36][VV][api.service:125]: send_list_entities_text_sensor_response: ListEntitiesTextSensorResponse {
   object_id: 'version_pzem_ecs'
   key: 1196846780
   name: 'Version Pzem ECS'
   unique_id: 'e8db84af36f6-version'
   icon: 'mdi:new-box'
}[07:30:36][C][template.sensor:021]: Template Sensor 'Puissance Réactive Ph1'[07:30:36][C][template.sensor:021]:   State Class: ''[07:30:36][C][template.sensor:021]:   Unit of Measurement: 'VAr'[07:30:36][C][template.sensor:021]:   Accuracy Decimals: 2[07:30:36][C][template.sensor:021]:   Icon: 'mdi:power'[07:30:36][C][template.sensor:022]:   Update Interval: 10.0s[07:30:36][VV][scheduler:152]: Running interval 'update' with interval=5000 last_execution=10939 (now=15942)[07:30:36][VV][uart_esp8266:111]:     Wrote 0b00000011 (0x03)[07:30:36][VV][uart_esp8266:111]:     Wrote 0b00000100 (0x04)[07:30:36][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)[07:30:36][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)[07:30:36][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)[07:30:36][VV][uart_esp8266:111]:     Wrote 0b00001010 (0x0A)[07:30:36][VV][uart_esp8266:111]:     Wrote 0b01110001 (0x71)[07:30:36][VV][uart_esp8266:111]:     Wrote 0b11101111 (0xEF)[07:30:36][VV][uart_esp8266:182]:     Flushing...[07:30:36][VV][api.service:125]: send_list_entities_text_sensor_response: ListEntitiesTextSensorResponse {
   object_id: 'uptime_pzem_fcs'
   key: 771846725
   name: 'Uptime Pzem FCS'
   unique_id: 'pzem_ecstext_sensoruptime_pzem_fcs'
   icon: 'mdi:clock-start'
[... cut ...] rval=5000 last_execution=45310 (now=50310)[07:31:10][VV][uart_esp8266:111]:     Wrote 0b00000001 (0x01)[07:31:10][VV][uart_esp8266:111]:     Wrote 0b00000100 (0x04)[07:31:10][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)[07:31:10][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)[07:31:10][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)[07:31:10][VV][uart_esp8266:111]:     Wrote 0b00001010 (0x0A)[07:31:10][VV][uart_esp8266:111]:     Wrote 0b01110000 (0x70)[07:31:10][VV][uart_esp8266:111

ssieb, 2021-06-24:
Have you ever had multiple pzem devices working on one uart?  How are they wired?

Pelbru, 2021-06-24:
No, the 3 devices on the same UART never worked. With just one it works.
I addressed the pzem with ESP.Easy.Flasher.exe and the firmware 
ESP_Easy_mega-20191003_PZEM_core_ESP8266_4M1M.bin
I tried with and without the diodes and it's the same problem

------ Message d'origine ------

(#2162)

Pelbru, 2021-06-24:
------ Message d'origine ------

(#2162)

ssieb, 2021-06-24:
Without the diodes, you are likely to fry something and you also need a pullup.  First make sure that a single pzem works with your config.  Then you can use the circuit in https://github.com/xoseperez/espurna/issues/1051 to add the others.

ssieb, 2021-06-24:
[image]

Pelbru, 2021-06-24:
Thank you for the new wiring plan
I test with the new wiring and I inform you

------ Message d'origine ------

(#2162)

ipetrovits, 2021-06-24:
Hi,
I'm utilizing 3 of these on the same UART of an ESP01 module.

Logger is set to 0 as above, but uart config is different:
`
uart:
  baud_rate: 9600
  stop_bits: 1
`
I have diodes as the picture above showing, and also modded the resistors on all 3 boards.
Tried using a level shifter before without success, because the high-low levels are a different (originally it is using 5V levels and I converted this to lower to 3.3V). Some say 8266s should tolerate 5V, i don't know. never tried.

ssieb, 2021-06-24:
Some people have found that 2 stop bits are needed and there's no harm in having an extra one.
The 8266 is officially 5V tolerant on the GPIO pins.

Pelbru, 2021-06-24:
It works!
I set 1 stop bit (for ESP32 you need 2 stop bit but not for ESP 8266)
I thought my pzem were addressed but they weren't there.
Addressing with ESP.Easy.Flasher.exe and the firmware 
ESP_Easy_mega-20191003_PZEM_core_ESP8266_4M1M.bin does not work.
I set the address of the pzem with software Pzem014 Master and a USB-TTL 
FTDI compatible with the CH340 driver and now everything works.
Thanks to all of you who helped solve my problem
Best regards

------ Message d'origine ------

(#2162)

closed 2021-06-24 by ssieb