esphome/issues #2163: v1.19.2 (maybe 1.19.X) has broken the esp32_ble_tracker for Xiaomi Mijia BLE Sensors

MrAlucardDante, 2021-06-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 20.04, pip 20.0.2, python 3.8

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.19.2

**Affected component:**

https://esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc

**Description of problem:**
Since updating from 1.18.0 to 1.19.2, my ESP32 stopped working. If I remove esp32_ble_tracker and all the BLE sensors, the ESP32 starts up, but if I add those sensors back, I get the ERR111, connection refused, even though I see the ESP32 connected to my router with the correct IP I assigned them in my DHCP as well as my config.

I downgraded my ESPHome install to 1.18.0, recomplied and reuploaded and everything is back to normal.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  esphome_name: first_floor_thermostat_node
  temp_label: Temperature
  humid_label: Humidite
  batt_label: Batterie
  living_label: Salon
  kitchen_label: Cuisine
  entrance_label: Entree
  shower_label: Douche
  study_label: Bureau

esphome:
  name: ${esphome_name}
  platform: ESP32
  board: esp-wrover-kit

time:
  - platform: homeassistant

wifi:
  power_save_mode: none
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    static_ip: !secret first_floor_ip
    gateway: !secret gateway_ip
    subnet: !secret subnet

esp32_ble_tracker:

sensor:
  - platform: uptime
    name: "Uptime de ${esphome_name}"
    
  - platform: wifi_signal
    name: "WiFi de ${esphome_name}"
    update_interval: 60s
    
  - platform: xiaomi_lywsd03mmc
    mac_address: !secret mac_living
    bindkey: !secret fake_bind_key
    temperature:
      name: "${temp_label} ${living_label}"
    humidity:
      name: "${humid_label} ${living_label}"
    battery_level:
      name: "${batt_label} ${living_label}"
  
  - platform: xiaomi_lywsd03mmc
    mac_address: !secret mac_kitchen
    bindkey: !secret fake_bind_key
    temperature:
      name: "${temp_label} ${kitchen_label}"
    humidity:
      name: "${humid_label} ${kitchen_label}"
    battery_level:
      name: "${batt_label} ${kitchen_label}"
  
  - platform: xiaomi_lywsd03mmc
    mac_address: !secret mac_entrance
    bindkey: !secret fake_bind_key
    temperature:
      name: "${temp_label} ${entrance_label}"
    humidity:
      name: "${humid_label} ${entrance_label}"
    battery_level:
      name: "${batt_label} ${entrance_label}"
  
  - platform: xiaomi_lywsd03mmc
    mac_address: !secret mac_shower
    bindkey: !secret fake_bind_key
    temperature:
      name: "${temp_label} ${shower_label}"
    humidity:
      name: "${humid_label} ${shower_label}"
    battery_level:
      name: "${batt_label} ${shower_label}"
  
  - platform: xiaomi_lywsd03mmc
    mac_address: !secret mac_study
    bindkey: !secret fake_bind_key
    temperature:
      name: "${temp_label} ${study_label}"
    humidity:
      name: "${humid_label} ${study_label}"
    battery_level:
      name: "${batt_label} ${study_label}"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret ota_api_password
  reboot_timeout: 60min

ota:
  password: !secret ota_api_password

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

agners, 2021-06-21:
Looks like a duplicate of https://github.com/esphome/issues/issues/2141.

jesserockz, 2021-06-22:
Closing as a duplicate

closed 2021-06-22 by jesserockz