esphome/issues #2165: What is the proper way of using ESP32-SOLO-1?

5tan, 2021-06-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 20.04, esphome installed using `pip`.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, ESP32-SOLO-1

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

N/A

**Description of problem:**
I own ESP32-SOLO-1. When I was trying to use 
```
  platform: ESP32
  board: esp32doit-devkit-v1
```
 I was getting instant reboots, with error message:
```
E (352) cpu_start: Running on single core chip, but application is built with dual core support.
E (352) cpu_start: Please enable CONFIG_FREERTOS_UNICORE option in menuconfig.
```
The same as described here https://github.com/esphome/issues/issues/1560 

When I have tried workaround from the issue https://github.com/esphome/issues/issues/1560 , meaning:
```
esphome:
  name: test
  platform: ESP32
  board: esp32doit-devkit-v1
  platformio_options:
    platform: espressif32@1.11.0
    platform_packages: |-4
          framework-arduinoespressif32 @ https://github.com/pauln/arduino-esp32.git#solo-no-mac-crc/1.0.4
```
I sometimes get 
```
[17:36:29]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[17:36:29]Core 0 register dump:
[17:36:29]PC      : 0x400e328e  PS      : 0x00060930  A0      : 0x800e3326  A1      : 0x3ffb1be0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d aeolus_uv_lamp_esp32 -t idedata
WARNING Decoded 0x400e328e: AsyncEventSourceClient::_runQueue() at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
```

Any tip how should I proceed with ESP32-SOLO-1 to get esphome ruinning & stable?

**Problem-relevant YAML-configuration entries:**
N/A

**Logs (if applicable):**

N/A

**Additional information and things you've tried:**

agners, 2021-06-21:
I think the CPU clock of the solo variant officially is specified to be 160MHz. I also made my own build of the Arduino SDK, which is really just Arduino SDK 1.0.6 rebuilt with `CONFIG_FREERTOS_UNICORE=y`. In my test this proved to be stable for a 12h+:

```yaml
esphome:
  platform: ESP32
  board: esp32dev
  arduino_version: '1.0.6'
  platformio_options:
    board_build.f_cpu: 160000000L
    platform_packages: |
      framework-arduinoespressif32 @ https://github.com/agners/arduino-esp32.git#1.0.6-solo
```

5tan, 2021-06-22:
Hello @agner, thanks for sharing this

Unfortunately for me it is crashing as well:
```
 (5506) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[09:03:19]E (5506) task_wdt:  - IDLE0 (CPU 0)
[09:03:19]E (5506) task_wdt: Tasks currently running:
[09:03:19]E (5506) task_wdt: CPU 0: loopTask
[09:03:19]E (5506) task_wdt: Aborting.
[09:03:19]abort() was called at PC 0x40152570 on core 0
[09:03:19]
[09:03:19]ELF file SHA256: 0000000000000000
[09:03:19]
[09:03:19]Backtrace: 0x40087fa7:0x3ffbfa60 0x4008820d:0x3ffbfa80 0x40152570:0x3ffbfaa0 0x400869b2:0x3ffbfac0 0x400eaacd:0x3ffb1970 0x400e8e91:0x3ffb1990 0x400e91d2:0x3ffb19b0 0x400e9221:0x3ffb19d0 0x400d4cf6:0x3ffb19f0 0x400d4e71:0x3ffb1a20 0x400dd641:0x3ffb1a40 0x400dd66e:0x3ffb1a70 0x400d50d2:0x3ffb1ad0 0x400d5056:0x3ffb1b10 0x40168191:0x3ffb1b30 0x4016824d:0x3ffb1b50 0x400dca35:0x3ffb1b70 0x400e067e:0x3ffb1bc0 0x400eb14a:0x3ffb1fb0 0x400891c6:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40087fa7: invoke_abort at /opt/esp/idf/components/esp32/panic.c:715
WARNING Decoded 0x4008820d: abort at /opt/esp/idf/components/esp32/panic.c:715
WARNING Decoded 0x40152570: task_wdt_isr at /opt/esp/idf/components/esp32/task_wdt.c:252
WARNING Decoded 0x400869b2: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x400eaacd: uartWriteBuf at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:464
WARNING Decoded 0x400e8e91: HardwareSerial::write(unsigned char const*, unsigned int) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:190
WARNING Decoded 0x400e91d2: Print::write(char const*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:261
WARNING Decoded 0x400e9221: Print::print(char const*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:261
 (inlined by) Print::println(char const*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:195
WARNING Decoded 0x400d4cf6: esphome::logger::Logger::log_message_(int, char const*, int) at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/src/esphome/components/logger/logger.cpp:175
WARNING Decoded 0x400d4e71: esphome::logger::Logger::log_vprintf_(int, char const*, int, char const*, __va_list_tag) at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/src/esphome/components/logger/logger.cpp:175
WARNING Decoded 0x400dd641: esphome::esp_log_vprintf_(int, char const*, int, char const*, __va_list_tag) at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/src/esphome/core/log.cpp:32
WARNING Decoded 0x400dd66e: esphome::esp_log_printf_(int, char const*, int, char const*, ...) at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/src/esphome/core/log.cpp:14
WARNING Decoded 0x400d50d2: esphome::ota::OTAComponent::dump_config() at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/src/esphome/components/ota/ota_component.cpp:356
WARNING Decoded 0x400d5056: esphome::ota::OTAComponent::setup() at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/src/esphome/components/ota/ota_component.cpp:356
WARNING Decoded 0x40168191: esphome::Component::call_setup() at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1754
WARNING Decoded 0x4016824d: esphome::Component::call() at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/src/esphome/core/component.cpp:82
WARNING Decoded 0x400dca35: esphome::Application::setup() at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/src/esphome/core/application.cpp:38
WARNING Decoded 0x400e067e: setup() at /home/z00lwix/test/my_ws/test_uv/test_dev/config/test_dev_esp32/lamp_esp32.yaml:128
WARNING Decoded 0x400eb14a: loopTask(void*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
WARNING Decoded 0x400891c6: vPortTaskWrapper at /opt/esp/idf/components/freertos/port.c:403
```

agners, 2021-06-22:
It does look like the watchdog triggers. What kind of components etc. are you using (full configuration)?

5tan, 2021-06-22:
* wifi
* logger
* ota
* web server
* interval
* binary sensor
* sensor
* script (some custom scripts for automation)
* switch
* output 
* light

5tan, 2021-06-22:
@agners 

Here is a minimal example to cause watchdog trigger (it's just a few components, and periodic LED blink):
```yaml
esphome:
  name: abc
  platform: ESP32
  board: esp32dev
  arduino_version: '1.0.6'
  platformio_options:
    board_build.f_cpu: 160000000L
    platform_packages: |
      framework-arduinoespressif32 @ https://github.com/agners/arduino-esp32.git#1.0.6-solo

wifi:
  ssid: ...
  password: ...

logger:
  level: DEBUG

ota:

web_server:
  port: 80
  auth:
    username: user
    password: pass

interval:
  - interval: 1s
    then:
      - light.toggle:
          id: light_led

output:
  - platform: gpio
    id: out_led
    pin: GPIO19

light:
  - platform: binary
    id: light_led
    name: "LED"
    output: out_led
```
It will cause issues like:
```
[11:24:22]E (5477) task_wdt:  - IDLE0 (CPU 0)
[11:24:22]E (5477) task_wdt: Tasks currently running:
[11:24:22]E (5477) task_wdt: CPU 0: loopTask
[11:24:22]E (5477) task_wdt: Aborting.
[11:24:22]abort() was called at PC 0x4014be30 on core 0
[11:24:22]
[11:24:22]ELF file SHA256: 0000000000000000
[11:24:22]
[11:24:22]Backtrace: 0x40087fa7:0x3ffbfa50 0x4008820d:0x3ffbfa70 0x4014be30:0x3ffbfa90 0x400869b2:0x3ffbfab0 0x400e5be0:0x3ffb1c40 0x400e41ed:0x3ffb1c60 0x400e452e:0x3ffb1c80 0x400e457d:0x3ffb1ca0 0x400d3ce6:0x3ffb1cc0 0x400d3e61:0x3ffb1cf0 0x400da899:0x3ffb1d10 0x400da8c6:0x3ffb1d40 0x400d40c2:0x3ffb1da0 0x400d4046:0x3ffb1de0 0x40160201:0x3ffb1e00 0x401602bd:0x3ffb1e20 0x400d9e31:0x3ffb1e40 0x400dbb4b:0x3ffb1e90 0x400e625a:0x3ffb1fb0 0x400891c6:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40087fa7: invoke_abort at /opt/esp/idf/components/esp32/panic.c:715
WARNING Decoded 0x4008820d: abort at /opt/esp/idf/components/esp32/panic.c:715
WARNING Decoded 0x4014be30: task_wdt_isr at /opt/esp/idf/components/esp32/task_wdt.c:252
WARNING Decoded 0x400869b2: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x400e5be0: uartWriteBuf at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-uart.c:464
WARNING Decoded 0x400e41ed: HardwareSerial::write(unsigned char const*, unsigned int) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/HardwareSerial.cpp:190
WARNING Decoded 0x400e452e: Print::write(char const*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:261
WARNING Decoded 0x400e457d: Print::print(char const*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:261
 (inlined by) Print::println(char const*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:195
WARNING Decoded 0x400d3ce6: esphome::logger::Logger::log_message_(int, char const*, int) at /home/z00lwix/test/abc/src/esphome/components/logger/logger.cpp:175
WARNING Decoded 0x400d3e61: esphome::logger::Logger::log_vprintf_(int, char const*, int, char const*, __va_list_tag) at /home/z00lwix/test/abc/src/esphome/components/logger/logger.cpp:175
WARNING Decoded 0x400da899: esphome::esp_log_vprintf_(int, char const*, int, char const*, __va_list_tag) at /home/z00lwix/test/abc/src/esphome/core/log.cpp:32
[... 12 lines cut ...]
[11:24:33]E (5428) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[11:24:33]E (5428) task_wdt:  - IDLE0 (CPU 0)
[11:24:33]E (5428) task_wdt: Tasks currently running:
[11:24:33]E (5428) task_wdt: CPU 0: loopTask
[11:24:33]E (5428) task_wdt: Aborting.
[11:24:33]abort() was called at PC 0x4014be30 on core 0
[11:24:33]
[11:24:33]ELF file SHA256: 0000000000000000
[11:24:33]
[11:24:34]Backtrace: 0x40087fa7:0x3ffbfa50 0x4008820d:0x3ffbfa70 0x4014be30:0x3ffbfa90 0x400869b2:0x3ffbfab0 0x401601ee:0x3ffb1e00 0x401602bd:0x3ffb1e20 0x400d9ecd:0x3ffb1e40 0x400dbb4b:0x3ffb1e90 0x400e625a:0x3ffb1fb0 0x400891c6:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x40087fa7: invoke_abort at /opt/esp/idf/components/esp32/panic.c:715
WARNING Decoded 0x4008820d: abort at /opt/esp/idf/components/esp32/panic.c:715
WARNING Decoded 0x4014be30: task_wdt_isr at /opt/esp/idf/components/esp32/task_wdt.c:252
WARNING Decoded 0x400869b2: _xt_lowint1 at /opt/esp/idf/components/freertos/xtensa_vectors.S:1154
WARNING Decoded 0x401601ee: esphome::Component::call_loop() at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1754
WARNING Decoded 0x401602bd: esphome::Component::call() at /home/z00lwix/test/abc/src/esphome/core/component.cpp:82
WARNING Decoded 0x400d9ecd: esphome::Application::setup() at /home/z00lwix/test/abc/src/esphome/core/application.cpp:50 (discriminator 2)
WARNING Decoded 0x400dbb4b: setup() at /home/z00lwix/test/abc/src/main.cpp:150
WARNING Decoded 0x400e625a: loopTask(void*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
WARNING Decoded 0x400891c6: vPortTaskWrapper at /opt/esp/idf/components/freertos/port.c:403
[11:24:34]
[11:24:34]Rebooting...
```
periodically

agners, 2021-06-22:
Is that the Espressif Development board ESP32-SOLO-1? It does work with the same configuration on my end.

Can you try:

```yaml
wifi:
  ssid: ...
  password: ...
  output_power: 11 dB
```

5tan, 2021-06-25:
> Is that the Espressif Development board ESP32-SOLO-1? It does work with the same configuration on my end.
> Can you try:

No this was a custom board. Unfortunately I am unable to test output_power setting you have recommended, since I don't have it anymore.

Now I am trying the very same yaml on ESP-WROOM-32 (ESP DevKit v1) and the "watchdog issue" doesn't happen on this board.

I am also observing Guru Meditation Error from above on ESP-WROOM-32. I have opened separate ticket for it: https://github.com/esphome/issues/issues/2191

I am closing this issue with following conclusion:
* there is no official support for ESP32-SOLO-1
* there are some ArduinoSDK custom tweaks you can use to run esphome ESP32-SOLO-1, but...
* ...currently it is not stable. Watchdog issues are expected.

closed 2021-06-25 by 5tan

jesserockz, 2021-09-23:
After esphome/esphome#2303 was merged you can now compile ESPHome for single core based boards using the following configuration. Note that not all components are supported by esp-idf yet, mainly webserver, captive portal and mqtt.

```yaml
esphome:
  name: ...

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
```

cryptotomte, 2022-01-18:
I have just started with EspHome and uses a ESP32-SOLO-1 as well. The default esp32dev board configuration compiled but gave the same error message as the thread started had. I found this thread and tried jesserockz example code. I then got this error message:

INFO Reading configuration /config/esphome/esp32test.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32test (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-ninja 1.7.1 
Reading CMake configuration...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Configuring incomplete, errors occurred!
See also "/data/esp32test/.pioenvs/esp32test/CMakeFiles/CMakeOutput.log".

fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
CMake Error at /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:189 (message):
  Failed to resolve component 'cxx'.
Call Stack (most recent call first):
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:215 (__build_resolve_and_add_req)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/build.cmake:435 (__build_expand_requirements)
  /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 1.36 seconds ==========================

My yaml is as follows:

esphome:
  name: esp32test

esp32:
  board: esp32dev
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
      
# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "3cfdd33a98d13ab5c4afab6941534e9e"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: <deleted before posting>

What can be the problem here?

// P