esphome/issues #2166: Treatlife DS03 not able to control fan on any 1.19.X version -- always resorts to speed level 4.

bkvargyas, 2021-06-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 1M

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.19.x (Any version of 1.19)

**Affected component:**

https://esphome.io/components/fan/tuya.html

**Description of problem:**
Anytime the fan speed selection is pressed, the speed will always default back to speed level 4.   Press the "-" on the wall switch two times, and it will go back to level 4.   Also, in the Home Assistant GUI, any attempt to try a speed more then 100%, will have it jump back to 100%.    The light control portion of the switch works just fine.

**Problem-relevant YAML-configuration entries:**
```yaml

substitutions:
  devicename: officefan
  lightname: officefan
  ssid: moonlight_IoT
  ssidpass: xxxxxxxx

esphome:
  name: $devicename
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: $ssid
  password: $ssidpass
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $lightname wifi
    password: xxxxx

captive_portal:

api:
  password: xxxxxxx

# Disable logging via the UART
logger:
  level: DEBUG
  baud_rate: 0

ota:
   password: xxxxxxx
   
web_server:
  port: 80
  
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 115200

tuya:

sensor:
  - platform: wifi_signal
    name: $lightname WiFi Signal
    update_interval: 60s

  - platform: uptime
    name: $lightname Uptime

light:
  - platform: "tuya"
    name: $lightname-light
    dimmer_datapoint: 10
    switch_datapoint: 9
    min_value: 100
    max_value: 1000

fan:
  - platform: "tuya"
    name: $lightname-fan
    switch_datapoint: 1
    speed_datapoint: 3
    speed_count: 4
```

**Logs (if applicable):**

```

[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 2
[11:31:28][D][tuya:271]: Datapoint 3 update to 2
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 3
[11:31:28][D][tuya:271]: Datapoint 3 update to 3
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 4
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 2
[11:31:28][D][tuya:271]: Datapoint 3 update to 4
[11:31:28][C][tuya.fan:022]: Speed has invalid value 4
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 2
[11:31:28][D][tuya:271]: Datapoint 3 update to 2
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 3
[11:31:28][D][tuya:271]: Datapoint 3 update to 2
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 3
[11:31:28][D][tuya:271]: Datapoint 3 update to 3
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 4
[11:31:28][D][tuya:271]: Datapoint 3 update to 3
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 4
[11:31:28][D][tuya:271]: Datapoint 3 update to 4
[11:31:28][C][tuya.fan:022]: Speed has invalid value 4
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 4
[11:31:28][D][tuya:271]: Datapoint 3 update to 4
[11:31:28][C][tuya.fan:022]: Speed has invalid value 4
[11:31:28][D][tuya:402]: Setting datapoint 1 to 1
[11:31:28][D][tuya:402]: Setting datapoint 3 to 4
```

**Additional information and things you've tried:**

Have not tried going back to 1.18 yet, as reports that 1.18 worked.   I started with 1.19 on the DS03, and have 15 of them to flash.   Was working fine on Tasmota, but wanted ESPHome for tighter integration.    It appears to be any attempt to try and set the speed less then 100% forces things back to 100%.   Even if I try the speed_count setting of 3 instead of 4....  same issues.

bkvargyas, 2021-06-21:
Ok, I'm attempting to roll back to V1.18 in Home Assistant, through the configuration settings:
esphome_version: v1.18.0

However --- I'm getting the following errors when trying to start up: ( I do see it download v1.18.0 )....

[12:18:19] INFO: Starting ESPHome dashboard...
usage: esphome [-h] [-v] [-q] [-s key value]
               [configuration [configuration ...]]
               {config,compile,upload,logs,run,clean-mqtt,wizard,mqtt-fingerprint,version,clean,dashboard,vscode,update-all}
               ...
esphome: error: argument command: invalid choice: '/config/esphome' (choose from 'config', 'compile', 'upload', 'logs', 'run', 'clean-mqtt', 'wizard', 'mqtt-fingerprint', 'version', 'clean', 'dashboard', 'vscode', 'update-all')

pcartwright81, 2021-06-22:
Unfortunately I figured out this yesterday, tried to revert to 18.0 also, and had the same issues.  There was a 19.0 dev branch that worked, which was not named.  I was hoping to fork and cherry picking some commits to make the ha addon work.  None of the betas work, and I went through most of the files since the original commit adding speed count.  Nothing looked out of place.

agners, 2021-06-22:
Maybe downgrading via custom components works?

```yaml
external_components:
  - source: github://esphome/esphome@v1.18.0
    components: [fan]
```

bkvargyas, 2021-06-22:
> Maybe downgrading via custom components works?
> 
> ```yaml
> external_components:
>   - source: github://esphome/esphome@v1.18.0
>     components: [fan, tuya]
> ```

So I did try this, didn't realize you could do this...and it worked!   Here is the debug output

```
[06:15:03][D][tuya.fan:022]: MCU reported speed of: 1
[06:15:03][D][tuya.fan:074]: Setting switch: ON
[06:15:03][D][tuya.fan:098]: Setting speed: 1
[06:15:14][D][tuya.fan:074]: Setting switch: ON
[06:15:14][D][tuya.fan:098]: Setting speed: 2
[06:15:14][D][tuya.fan:022]: MCU reported speed of: 2
[06:15:14][D][tuya.fan:074]: Setting switch: ON
[06:15:14][D][tuya.fan:098]: Setting speed: 2
[06:15:23][D][tuya.fan:074]: Setting switch: ON
[06:15:23][D][tuya.fan:098]: Setting speed: 3
[06:15:23][D][tuya.fan:022]: MCU reported speed of: 3
```
Something something changed in the V1.19 version of these components that broke this integration.

bkvargyas, 2021-06-22:
Ok, just using the following settings worked, so it appears to just be the tuya component that's not working correctly in 1.19

```
external_components:
  - source: github://esphome/esphome@v1.18.0
    components: [tuya]
```

grrraham, 2021-06-28:
The Tuya Fan / Light wall switches I have are not specially Treatlife DS03, however they have the same issue when compiled with esphome V1.20 dev.

agners work around has restored functionality for me as well.

bdr99, 2021-07-03:
I also have a Treatlife DS03 and I can confirm this issue. ESPHome 1.18.0 works, but ESPHome 1.19.4 is not working.

trvrnrth, 2021-07-03:
It would be helpful to get some [verbose logs](https://esphome.io/components/logger.html#log-levels) showing the problem, ideally annotated with the times at which the switches are pressed as these might shed some light on the issue.

pcartwright81, 2021-07-03:
VERY_VERBOSE shows the actual issue.  I think this is because of data point validation being broken.
[15:31:19][VV][preferences:051]: SAVE 43: 0=0x00000001 1=0x00000004 (Type=625718363, CRC=0x563EFBF5)
[15:31:19][V][tuya.fan:069]: Setting switch: ON
[15:31:19][D][tuya:402]: Setting datapoint 1 to 1
[15:31:19][E][tuya:405]: Attempt to set unknown datapoint 1
[15:31:19][V][tuya.fan:082]: Setting speed: 4
[15:31:19][D][tuya:402]: Setting datapoint 3 to 4
[15:31:19][E][tuya:405]: Attempt to set unknown datapoint 3

trvrnrth, 2021-07-04:
@pcartwright81 Thanks for the extra log; that was helpful.

I think I've spotted the problem and have PRed a fix. Classic off-by-one gotcha! If you're able to test it by adding the fix in as an external component and reporting back that would be great:
```
external_components:
  - source: github://trvrnrth/esphome@fix-tuya-fan-speed
    components: [tuya]
```

I haven't dug into it but the logs about attempts to set unknown datapoints are likely to relate to the fan component trying to set values before the tuya MCU has reported it's initial state which is used to determine the datapoint types as well as initial values. It's not ideal but probably won't cause any issues in real use.

closed 2021-07-04 by jesserockz

pcartwright81, 2021-07-04:
Pulled from dev, and this fix works.

bkvargyas, 2021-07-04:
I too pulled from from dev and confirmed this fix works.  After I saw what was changed and looked back at 1.18 it was apparent what happened.