esphome/issues #2167: Auto_uart warnings continue /  HA supervisor with 1.19.2

lumper5, 2021-06-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HA Superior / Core:  2021.06.3 / 2021.6.6

ver: 1.19.2

**Affected component:**

I've been seeing the following warning for several months.  I have waited to see if an update would "fix" whatever error was being caught.  I'm currently:
core: core-2021.6.6
sup: supervisor-2021.06.3
ESPHome ver: 1.19.2

 This is my attempt to "report it to the maintainer of ESPHome" as indicated in the error messages.  While my devices and HA seem to work without issue, I keep seeing these error messages periodically - several times a day and at each reboot/restart.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```

`21-06-21 20:47:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome (dev)
21-06-21 20:47:43 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/15ef4d2f/template/config.yaml: Invalid Add-on config!. Got ordereddict([('base', ordereddict([('url', 'https://esphome.io/'), ('webui', 'http://[HOST]:[PORT:6052]'), ('startup', 'application'), ('boot', 'auto'), ('arch', ['amd64', 'armv7', 'aarch64']), ('hassio_api', True), ('auth_api', True), ('hassio_role', 'default'), ('homeassistant_api', False), ('host_network', True), ('ingress', True), ('ingress_port', 0), ('panel_icon', 'mdi:chip'), ('auto_uart', True), ('ports', ordereddict([('6052/tcp', None)])), ('ports_description', ordereddict([('6052/tc...
21-06-21 20:47:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-06-21 20:47:43 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome (beta)
21-06-21 20:47:44 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome (dev)
21-06-21 20:47:44 WARNING (MainThread) [supervisor.store.data] Can't read /data/addons/git/15ef4d2f/template/config.yaml: Invalid Add-on config!. Got ordereddict([('base', ordereddict([('url', 'https://esphome.io/'), ('webui', 'http://[HOST]:[PORT:6052]'), ('startup', 'application'), ('boot', 'auto'), ('arch', ['amd64', 'armv7', 'aarch64']), ('hassio_api', True), ('auth_api', True), ('hassio_role', 'default'), ('homeassistant_api', False), ('host_network', True), ('ingress', True), ('ingress_port', 0), ('panel_icon', 'mdi:chip'), ('auto_uart', True), ('ports', ordereddict([('6052/tcp', None)])), ('ports_description', ordereddict([('6052/tc...
21-06-21 20:47:44 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome
21-06-21 20:47:44 WARNING (MainThread) [supervisor.addons.validate] Add-on config 'auto_uart' is deprecated, use 'uart'. Please report this to the maintainer of ESPHome (beta)`

```

**Additional information and things you've tried:**

jesserockz, 2021-06-22:
This has been fixed a while ago now.

Also that file it is complaining about `Can't read /data/addons/git/15ef4d2f/template/config.yaml` is an old file that has been renamed too.

So it seems like the `https://github.com/esphome/hassio` repo is not updating on your instance.

Perhaps you could try to remove the repo and add it back again?

lumper5, 2021-07-19:
Don't know what this would have to do this issue, but upon adding a proxy range to http: in config.yaml, the issue is no longer happening.  Maybe something to do with a docker issue in supervisor vis a vis a reverse proxy?  Sounds unlikely, but there it is.

closed 2021-12-02 by github-actions[bot]