esphome/issues #2168: Inconsistent naming convention

3ative, 2021-06-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

N/A
**ESPHome version (latest production, beta, dev branch)**

v1.19.2
**Affected component:**

**Description of problem:**
Now that we are _forced_ to use dashes "-" instead of underscores "_" in our device names, why don't the rest of the files follow suit... or, even better, revert to underscores in the name, please. 💕
# E.G.
[image]

Yet after a page refresh, this title changes to underscores:
[image]
and the 'Filename:' remains with dashes.

Obviously the order of entries on the ESPHome frontend is using the filename to index, which throws out the ordering:
[image]

More-over, there is inconsistent naming on the file structure, in the /esphome folder:
[image]

It would be great to revert back to using underscores in the Name, which would be consistent with coding  / file structure / etc... 👍 

🙄 OR are you going to force us to re-do / rename / convert ALL our other devices in ESPHome?

Cheers. 😎
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

3ative, 2021-06-22:
### Additional:
I see ESPHome will allow changing dashes to underscores, after the fact, on the filename in the /esphome folder. 🎁
I.E. "freshmatic-bedroom-2.yaml" to "freshmatic_bedroom_2.yaml" 👍

So I guess, it's just the initial naming rules that need to be re-refinded. 😎

dougle03, 2021-06-22:
I don't understand why these things get played with? For what purpose. There are many many elements of ESPHome that could use attention, this really was not one of them, especially without any explanation as to why it was monkeyed with. It feels like a personal preference was implemented badly....

bojanpotocnik, 2021-06-22:
> I don't understand why these things get played with? For what purpose.

Simply because ESPHome shall conform to specifications of the used protocols. Underscore characters are not permitted in domain names in accordance with RFC-1035, which only allows letters, digits and hyphens.

**RFC-1035** _DOMAIN NAMES - IMPLEMENTATION AND SPECIFICATION_ - [**2.3.1.**](https://datatracker.ietf.org/doc/html/rfc1035#section-2.3.1) Preferred name syntax
> The labels must follow the rules for ARPANET host names. They must start with a letter, end with a letter or digit, and have as interior characters only letters, digits, and hyphen. There are also some restrictions on the length. Labels must be 63 characters or less.

`my_smart_sensor` may work on most routers supporting mDNS, but it is not required to - `my-smart-sensor`, however, must work. This can cause confusion between users ("it works fine on my network"), because non-conforming domain name is probably not the first thing you would think of when e.g. your Home Assistant ESPHome add-on cannot update your sensor. Which is normal, because as it turns out, most users are not aware of this RFC rules.

3ative, 2021-06-22:
@bojanpotocnik 
I'm ok with consistency and if indeed it was to conform to the standards, why hasn't it been implemented across the various file and folder structures. (see my screen caps for the mixed use dashes and underscores) 
My main point, as per the title, _"Inconsistent naming convention (Underscores vs Dashes)"_ 😉

OttoWinter, 2021-06-23:
> Now that we are forced to use dashes "-" instead of underscores "_" in our device names

Just as an FYI, you _can_ still use underscores. Just edit `esphome->name` and the filename, then everything works fine. The wizard prevents underscores nowadays because we don't want people to encounter hard-to-diagnose issues with the mentioned RFC (both for them, and for us for support burden).

This is a good compromise IMHO because new users don't fall in this trap, and existing users can continue to use it _if_ it works for them (and the router is lenient)

> Yet after a page refresh, this title changes to underscores:

Can you post the exact steps with which this happens? AFAIK, the name is exactly what you have in your `esphome->name`

> More-over, there is inconsistent naming on the file structure, in the /esphome folder:

Again, that folder name _should_ be just what you have in `esphome->name`. If not that's probably a bug somewhere.

3ative, 2021-06-23:
> Just as an FYI, you _can_ still use underscores. Just edit `esphome->name` and the filename, then everything works fine.

Yeah, I noticed that too, see my second post/reply: https://github.com/esphome/issues/issues/2168#issuecomment-865501641
So that's cool - I'm just wondering why it HAS to be dashes - But I guess the 'Title' of the device (in ESPHome) is taken from that. 

> This is a good compromise IMHO because new users don't fall in this trap, and existing users can continue to use it _if_ it works for them (and the router is lenient)

**I get it, after it was pointed out 😉 this is domain naming compliancy - I understand - I withdraw this part now. 👍**

> > Yet after a page refresh, this title changes to underscores:
> 
> Can you post the exact steps with which this happens? AFAIK, the name is exactly what you have in your `esphome->name`

**I'm not able to replicate, as seen in my OP - I assume it happened as some sort of caching - So disregard this too. 👍**

# So to drill down the OP. 
If we are now complying to conventions by using dashes, then shouldn't everything follow suit? 
- Having **Folder Names** follow **File Names** (_use dashes, instead of underscores_) would seem logical and organisationally  consistent.

Thank you for your reply. Cheers.

ssieb, 2021-06-24:
The folder name follows the device name, not the yaml file name.

3ative, 2021-06-24:
> The folder name follows the device name, not the yaml file name.

Hence me opening an issue... "Inconsistent naming convention" 🤣

zechnkaas, 2021-07-01:
mhm would be nice to have it as option. i also used one variable as my device name and used the same for the IDs too. IDs no dashes are allowed

3ative, 2021-07-24:
Closing this now as after copying, deleting and re-doing all 51 of my ESPHome devices... 🤦‍♀️
I have consistent naming across the board. 👌
_- Dashes in the .yaml files, dashes in the folder names and dashes all over the ESPHome Dashboard! and my DHCP loved it too👍👍_

Thanks to everyone that contributed and left comments. ❤

closed 2021-07-24 by 3ative

3ative, 2021-07-24:
**Additional:**
A little while ago I made a tutorial on how to change the names... 
This is what I did for all my ESPHome devices:
https://youtu.be/ekOOb8qAe7c

ruddj, 2021-08-25:
Just a followup question, does this mean we can no longer use a device name with dashes in the ID: field?
e.g. 
```
substitutions:
  devicename: light-kitchen

esphome:
  name: $devicename
  platform: ESP32
  board: esp32dev

3ative, 2021-08-25:
For mine, I use just room name in substitutions and prefix/append entity names & ID's with underscores.

I.E.:
```yaml
substitutions:
  room: bedroom

esphome:
  name: fan-${room}

....

switch:
  - platform: template
    id: ${room}_master
    icon: mdi:fan
    name: Fan ${room} master
  ```

ruddj, 2021-09-03:
Thanks, Just realised a better option is to remove the device name fully from ID, since ID is only used internally to the device. Only need to include substitutions for _name:_ rather than for _id:_