esphome/issues #2171: EZO Sensors not working properly

jcastro, 2021-06-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io latest version core-2021.6.6
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini ESP32
**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.19.2

**Affected component:**

https://esphome.io/components/sensor/ezo.html

**Description of problem:**
I have connected through I2C an Atlas Scientific ph probe and a electrical conductivity probe. These are working correctly as I tested them in other software (Mycodo).

However when connecting them to my ESP32 I'm experiencing 2 main problems
1. Both devices are detected and I can see the green light on the breakout board which means is trying to read from it
```[11:39:27][I][app:105]: ESPHome version 1.19.2 compiled on Jun 22 2021, 11:32:56
[11:39:27][C][wifi:484]: WiFi:
[11:39:27][C][wifi:344]:   SSID: [redacted]
[11:39:27][C][wifi:345]:   IP Address: 10.0.10.125
[11:39:27][C][wifi:347]:   BSSID: [redacted]
[11:39:27][C][wifi:348]:   Hostname: 'hwms03'
[11:39:27][C][wifi:352]:   Signal strength: -53 dB ▂▄▆█
[11:39:27][C][wifi:356]:   Channel: 1
[11:39:27][C][wifi:357]:   Subnet: 255.255.255.0
[11:39:27][C][wifi:358]:   Gateway: 10.0.10.1
[11:39:27][C][wifi:359]:   DNS1: 0.0.0.0
[11:39:27][C][wifi:360]:   DNS2: 0.0.0.0
[11:39:27][C][i2c:028]: I2C Bus:
[11:39:27][C][i2c:029]:   SDA Pin: GPIO21
[11:39:27][C][i2c:030]:   SCL Pin: GPIO22
[11:39:27][C][i2c:031]:   Frequency: 50000 Hz
[11:39:27][I][i2c:033]: Scanning i2c bus for active devices...
[11:39:28][I][i2c:040]: Found i2c device at address 0x63
[11:39:28][I][i2c:040]: Found i2c device at address 0x64
[11:39:28][C][logger:189]: Logger:
[11:39:28][C][logger:190]:   Level: DEBUG
[11:39:28][C][logger:191]:   Log Baud Rate: 115200
[11:39:28][C][logger:192]:   Hardware UART: UART0
[11:39:28][C][ezo.sensor:014]: EZO 'ph_ezo'
[11:39:28][C][ezo.sensor:014]:   State Class: ''
[11:39:28][C][ezo.sensor:014]:   Unit of Measurement: 'pH'
[11:39:28][C][ezo.sensor:014]:   Accuracy Decimals: 0
[11:39:28][C][ezo.sensor:015]:   Address: 0x63
[11:39:28][C][ezo.sensor:018]:   Update Interval: 5.0s
[11:39:28][C][ezo.sensor:014]: EZO 'EC'
[11:39:28][C][ezo.sensor:014]:   State Class: ''
[11:39:28][C][ezo.sensor:014]:   Unit of Measurement: 'uS/cm'
[11:39:28][C][ezo.sensor:014]:   Accuracy Decimals: 0
[11:39:28][C][ezo.sensor:015]:   Address: 0x64
[11:39:28][C][ezo.sensor:018]:   Update Interval: 5.0s
[11:39:28][C][web_server:136]: Web Server:
[11:39:28][C][web_server:137]:   Address: 10.0.10.125:80
[11:39:28][C][ota:029]: Over-The-Air Updates:
[11:39:28][C][ota:030]:   Address: 10.0.10.125:3232
[11:39:28][C][api:095]: API Server:
[11:39:28][C][api:096]:   Address: 10.0.10.125:6053
[11:39:31][D][sensor:117]: 'EC': Sending state 0.00000 uS/cm with 0 decimals of accuracy
[11:39:32][D][sensor:117]: 'ph_ezo': Sending state 3.90200 pH with 0 decimals of accuracy
[11:39:36][D][sensor:117]: 'EC': Sending state 0.00000 uS/cm with 0 decimals of accuracy
[11:39:37][D][sensor:117]: 'ph_ezo': Sending state 3.89800 pH with 0 decimals of accuracy
[11:39:41][D][sensor:117]: 'EC': Sending state 0.00000 uS/cm with 0 decimals of accuracy
[11:39:42][D][sensor:117]: 'ph_ezo': Sending state 3.90000 pH with 0 decimals of accuracy````

2. However the EC probe is not showing any value but it shows as a sensor on HA while the pH is showing value on the logs but not creating a sensor in HA. Below is how it looks
![Screenshot 2021-06-22 at 11 42 19](https://user-images.githubusercontent.com/190036/122902624-ed796d80-d34e-11eb-97ae-591015e49621.png)
![Screenshot 2021-06-22 at 11 42 44](https://user-images.githubusercontent.com/190036/122902680-fa965c80-d34e-11eb-868b-bfb3ce9bfc35.png)
 

**Problem-relevant YAML-configuration entries:**
```yaml
```esphome:
  name: hwms03
  platform: ESP32
  board: wemos_d1_mini32

wifi:
  ssid: "IoT"
  password: "redacted"
  #fast_connect: True
  #power_save_mode: none
  manual_ip:
    static_ip: 10.0.10.125
    gateway: 10.0.10.1
    subnet: 255.255.255.0

logger:
api:
ota:
web_server:

i2c:
  sda: 21
  scl: 22
  scan: True
  id: bus_a

sensor:
  - platform: ezo
    id: ph_ezo
    address: 99
    unit_of_measurement: "pH"
    #accuracy_decimals: 2
    update_interval: 15s
    
  - platform: ezo
    id: ec_ezo
    name: "EC"
    address: 100
    #accuracy_decimals: 2
    unit_of_measurement: "uS/cm"
    update_interval: 15s```

```

**Logs (if applicable):**

```INFO Reading configuration /config/esphome/hwms03.yaml...
INFO Starting log output from 10.0.10.125 using esphome API
INFO Connecting to 10.0.10.125:6053 (10.0.10.125)
INFO Successfully connected to 10.0.10.125
[11:43:50][I][app:105]: ESPHome version 1.19.2 compiled on Jun 22 2021, 11:40:43
[11:43:50][C][wifi:484]: WiFi:
[11:43:50][C][wifi:344]:   SSID: [redacted]
[11:43:50][C][wifi:345]:   IP Address: 10.0.10.125
[11:43:50][C][wifi:347]:   BSSID: [redacted]
[11:43:50][C][wifi:348]:   Hostname: 'hwms03'
[11:43:50][C][wifi:352]:   Signal strength: -53 dB ▂▄▆█
[11:43:50][C][wifi:356]:   Channel: 1
[11:43:50][C][wifi:357]:   Subnet: 255.255.255.0
[11:43:50][C][wifi:358]:   Gateway: 10.0.10.1
[11:43:50][C][wifi:359]:   DNS1: 0.0.0.0
[11:43:50][C][wifi:360]:   DNS2: 0.0.0.0
[11:43:50][C][i2c:028]: I2C Bus:
[11:43:50][C][i2c:029]:   SDA Pin: GPIO21
[11:43:50][C][i2c:030]:   SCL Pin: GPIO22
[11:43:50][C][i2c:031]:   Frequency: 50000 Hz
[11:43:50][I][i2c:033]: Scanning i2c bus for active devices...
[11:43:50][I][i2c:040]: Found i2c device at address 0x63
[11:43:50][I][i2c:040]: Found i2c device at address 0x64
[11:43:50][C][logger:189]: Logger:
[11:43:50][C][logger:190]:   Level: DEBUG
[11:43:50][C][logger:191]:   Log Baud Rate: 115200
[11:43:50][C][logger:192]:   Hardware UART: UART0
[11:43:50][C][ezo.sensor:014]: EZO 'ph_ezo'
[11:43:50][C][ezo.sensor:014]:   State Class: ''
[11:43:50][C][ezo.sensor:014]:   Unit of Measurement: 'pH'
[11:43:50][C][ezo.sensor:014]:   Accuracy Decimals: 0
[11:43:50][C][ezo.sensor:015]:   Address: 0x63
[11:43:50][C][ezo.sensor:018]:   Update Interval: 15.0s
[11:43:51][C][ezo.sensor:014]: EZO 'EC'
[11:43:51][C][ezo.sensor:014]:   State Class: ''
[11:43:51][C][ezo.sensor:014]:   Unit of Measurement: 'uS/cm'
[11:43:51][C][ezo.sensor:014]:   Accuracy Decimals: 0
[11:43:51][C][ezo.sensor:015]:   Address: 0x64
[11:43:51][C][ezo.sensor:018]:   Update Interval: 15.0s
[11:43:51][C][web_server:136]: Web Server:
[11:43:51][C][web_server:137]:   Address: 10.0.10.125:80
[11:43:51][C][ota:029]: Over-The-Air Updates:
[11:43:51][C][ota:030]:   Address: 10.0.10.125:3232
[11:43:51][C][api:095]: API Server:
[11:43:51][C][api:096]:   Address: 10.0.10.125:6053
[11:43:51][D][sensor:117]: 'EC': Sending state 0.00000 uS/cm with 0 decimals of accuracy
[11:43:52][D][sensor:117]: 'ph_ezo': Sending state 3.90000 pH with 0 decimals of accuracy
[11:44:05][D][sensor:117]: 'EC': Sending state 0.00000 uS/cm with 0 decimals of accuracy
[11:44:07][D][sensor:117]: 'ph_ezo': Sending state 3.89800 pH with 0 decimals of accuracy
```

**Additional information and things you've tried:**

jcastro, 2021-06-22:
I think I found "the bug" (maybe it's just expected behaviour). I've added salt to the water and now the EC readings shows.

Previously the EC of the water was around 80 uS/cm, now it's ~617 uS/cm and it's showing up.

So this is either on purpose or we could maybe lower the minimum that should report, IMHO having this such precise lab tools I would like to see whatever the sensor is reading, even if it's below XX amount

[image]

ssieb, 2021-06-23:
There is nothing in the code to limit what values will be reported.  If you set the logger level to very_verbose, you can see what data is being transferred.  Get it to the state where it's only reporting 0 like before and post the logs showing the data.

jcastro, 2021-06-23:
Sure, here's the output when the probe is out of water

```[22:45:53][VV][scheduler:152]: Running interval 'update' with interval=2000 last_execution=31952 (now=33952)
[22:45:53][VV][i2c:056]: Beginning Transmission to 0x64:
[22:45:53][VV][i2c:093]:     Writing 0b01010010 (0x52)
[22:45:53][VV][i2c:061]:     Transmission ended. Status code: 0x00
[22:45:54][VV][i2c:083]: Requesting 20 bytes from 0x64:
[22:45:54][VV][i2c:113]:     Received 0b00000001 (0x01)
[22:45:54][VV][i2c:113]:     Received 0b00110000 (0x30)
[22:45:54][VV][i2c:113]:     Received 0b00101110 (0x2E)
[22:45:54][VV][i2c:113]:     Received 0b00110000 (0x30)
[22:45:54][VV][i2c:113]:     Received 0b00110000 (0x30)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:45:54][V][sensor:023]: 'EC': Received new state 0.000000
[22:45:54][D][sensor:117]: 'EC': Sending state 0.00000 uS/cm with 2 decimals of accuracy
[22:45:54][VV][api.service:105]: send_sensor_state_response: SensorStateResponse {
  key: 1819748205
  state: 0
  missing_state: NO
}
[22:45:54][VV][api.service:220]: on_ping_request: PingRequest {}
[22:45:54][VV][api.service:032]: send_ping_response: PingResponse {}```

This is when I put the probe into water. Surprisingly now it seems to read! Anyway I'm adding the output in case you see something strange
```
[22:47:02][VV][api.service:220]: on_ping_request: PingRequest {}
[22:47:02][VV][api.service:032]: send_ping_response: PingResponse {}
[22:47:02][VV][scheduler:152]: Running interval 'update' with interval=2000 last_execution=21578 (now=23579)
[22:47:03][VV][i2c:056]: Beginning Transmission to 0x64:
[22:47:03][VV][i2c:093]:     Writing 0b01010010 (0x52)
[22:47:03][VV][i2c:061]:     Transmission ended. Status code: 0x00
[22:47:03][VV][i2c:083]: Requesting 20 bytes from 0x64:
[22:47:03][VV][i2c:113]:     Received 0b00000001 (0x01)
[22:47:03][VV][i2c:113]:     Received 0b00111001 (0x39)
[22:47:03][VV][i2c:113]:     Received 0b00110101 (0x35)
[22:47:03][VV][i2c:113]:     Received 0b00101110 (0x2E)
[22:47:03][VV][i2c:113]:     Received 0b00110000 (0x30)
[22:47:03][VV][i2c:113]:     Received 0b00110001 (0x31)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
[22:47:03][V][sensor:023]: 'EC': Received new state 95.010002
[22:47:03][D][sensor:117]: 'EC': Sending state 95.01000 uS/cm with 2 decimals of accuracy
[22:47:03][VV][api.service:105]: send_sensor_state_response: SensorStateResponse {
  key: 1819748205
  state: 95.01
  missing_state: NO
}```

ssieb, 2021-06-23:
That is clearly the device doing that.
```
[22:45:54][VV][i2c:113]:     Received 0b00000001 (0x01)
[22:45:54][VV][i2c:113]:     Received 0b00110000 (0x30)
[22:45:54][VV][i2c:113]:     Received 0b00101110 (0x2E)
[22:45:54][VV][i2c:113]:     Received 0b00110000 (0x30)
[22:45:54][VV][i2c:113]:     Received 0b00110000 (0x30)
[22:45:54][VV][i2c:113]:     Received 0b00000000 (0x00)
```
0.00
```
[22:47:03][VV][i2c:113]:     Received 0b00000001 (0x01)
[22:47:03][VV][i2c:113]:     Received 0b00111001 (0x39)
[22:47:03][VV][i2c:113]:     Received 0b00110101 (0x35)
[22:47:03][VV][i2c:113]:     Received 0b00101110 (0x2E)
[22:47:03][VV][i2c:113]:     Received 0b00110000 (0x30)
[22:47:03][VV][i2c:113]:     Received 0b00110001 (0x31)
[22:47:03][VV][i2c:113]:     Received 0b00000000 (0x00)
```
95.01

There's nothing I can do about it, unless there's some configuration that can be sent to the device to change the behaviour.

jcastro, 2021-06-23:
Thanks for looking at it!

ssieb, 2021-06-23:
Feel free to re-open or comment if you find any way to change the behaviour of the sensor.

closed 2021-06-23 by ssieb