esphome/issues #2172: OTA and name_add_mac_suffix

atrent, 2021-06-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** ESP8266 (Wemos D1 mini)

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch):** 1.19.2

**Affected component:** OTA

**Description of problem:**

when using 'name_add_mac_suffix: True' the DHCP name becomes '<APP_NAME>-<PARTIALMAC>  but the OTA update still tries only '<APP_NAME>.local' (i.e., without mac address)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: iotdial
  platform: ESP8266
  board: d1_mini
  name_add_mac_suffix: True
```

**Logs (if applicable):**
INFO Successfully compiled program.
Found multiple options, please choose one:
  [1] /dev/ttyUSB0 (USB2.0-Serial)
  [2] Over The Air (iotdial.local)
(number): 2
INFO Resolving IP address of iotdial.local
ERROR Error resolving IP address of iotdial.local. Is it connected to WiFi?
ERROR (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
ERROR Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -2] Name or service not known

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2021-06-23:
Please see the note here: https://esphome.io/components/esphome.html#adding-the-mac-address-as-a-suffix-to-the-device-name
ESPHome can't know what device to send to, as it doesn't know the address it should send to (the address contains the MAC, which ESPHome doesn't know).

atrent, 2021-06-23:
> Please see the note here: https://esphome.io/components/esphome.html#adding-the-mac-address-as-a-suffix-to-the-device-name
> ESPHome can't know what device to send to, as it doesn't know the address it should send to (the address contains the MAC, which ESPHome doesn't know).

thanks, I missed the 'use_address' option

closed 2021-06-23 by atrent