esphome/issues #2174: Devices in Safari for Mac show as a grey bar status on main dashboard

erikpendragon, 2021-06-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

 Add-on version: 1.19.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 6.1  (amd64 / qemux86-64)
 Home Assistant Core: 2021.6.6
 Home Assistant Supervisor: 2021.06.4

**ESP (ESP32/ESP8266, Board/Sonoff):**

all of my devices

**ESPHome version (latest production, beta, dev branch)**

v1.19.0, v1.19.2, v1.20.0-dev

**Affected component:**

**Description of problem:**
Safari for mac Version 13.1.2 after upgrading to 1.19.0+ all of my devices come up with a grey status bar
and the install button doesn't do anything when clicked

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
1.19.2 LOGS------
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 1.19.2
 You are running the latest version of this add-on.
 System: Home Assistant OS 6.1  (amd64 / qemux86-64)
 Home Assistant Core: 2021.6.6
 Home Assistant Supervisor: 2021.06.4
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-esphome.sh: executing... 
[cont-init.d] 30-esphome.sh: exited 0.
[cont-init.d] 40-migrate.sh: executing... 
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[13:41:15] INFO: Waiting for dashboard to come up...
[13:41:15] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[13:41:16] INFO: Starting NGINX...
INFO 101 GET /ace (0.0.0.0) 1.19ms
[... 82 lines cut ...]
[cont-init.d] 40-migrate.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[13:27:17] INFO: Waiting for dashboard to come up...
[13:27:17] INFO: Starting ESPHome dashboard...
INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[13:27:18] INFO: Starting NGINX...
INFO 101 GET /ace (0.0.0.0) 1.61ms
INFO Running command 'esphome --dashboard -q vscode /config/esphome --ace'
INFO 304 GET / (0.0.0.0) 20.28ms
INFO 304 GET /static/css/esphome.css?hash=e183fe6e (0.0.0.0) 5.74ms
INFO 304 GET /static/css/esphome.css?hash=e183fe6e (0.0.0.0) 1.47ms

```

**Additional information and things you've tried:**

i have tried setting status_use_ping both true and false - didn't work
clearing cache of the browser - didn't work
setting different user agent - didn't work
open on iOS home assistant app - works
open in chrome on mac - works

[image]

slimcdk, 2021-06-22:
Can you inspect network usage on Safari? Can you see it querying for `ping`?
What do you see if you access `<esphome ip>/ping`?

erikpendragon, 2021-06-22:
i am not sure how i would do that.
If i follow this guide [here](https://jun711.github.io/web/how-to-inspect-network-request-and-response-headers-on-safari/)

If i go to http://homeassistant.local:8123/a0d7b954_esphome/dashboard i get 1 error on a file index-98a3f8c8.js
and 2 warnings
Summary
URL: http://homeassistant.local:8123/api/hassio_ingress/CtOVCQYxMUI9kWbqU4k4m099cyidauntT2y7fXM84xs/static/js/esphome/index-98a3f8c8.js
Status: 401 Unauthorized
Source: Network

Request
Origin: http://homeassistant.local:8123
Accept: */*
Referer: http://homeassistant.local:8123/api/hassio_ingress/CtOVCQYxMUI9kWbqU4k4m099cyidauntT2y7fXM84xs/
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_13_6) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/13.1.2 Safari/605.1.15

Response
Content-Type: text/plain
Content-Length: 17
Date: Tue, 22 Jun 2021 18:42:05 GMT
Server: Python/3.9 aiohttp/3.7.4.post0

[Warning] The main 'lit-element' module entrypoint is deprecated. Please update your imports to use the 'lit' package: 'lit' and 'lit/decorators.ts' or import from 'lit-element/lit-element.ts'. (app.bcb38ac1.js, line 849)
[Warning] The main 'lit-element' module entrypoint is deprecated. Please update your imports to use the 'lit' package: 'lit' and 'lit/decorators.ts' or import from 'lit-element/lit-element.ts'. (entrypoint.0d3c68f7.js, line 4171)

if i go to http://homeassistant.local:8123/a0d7b954_esphome/ping
i get two warnings
[Warning] The main 'lit-element' module entrypoint is deprecated. Please update your imports to use the 'lit' package: 'lit' and 'lit/decorators.ts' or import from 'lit-element/lit-element.ts'. (app.bcb38ac1.js, line 849)
[Warning] The main 'lit-element' module entrypoint is deprecated. Please update your imports to use the 'lit' package: 'lit' and 'lit/decorators.ts' or import from 'lit-element/lit-element.ts'. (entrypoint.0d3c68f7.js, line 4171)

bkvargyas, 2021-06-22:
Works for me on Safari....   MacOS Big Sur 11.4.   If you happen to be running UniFi access points, reboot them.   Seen that before (usually red bars though)

[image]

erikpendragon, 2021-06-22:
Edit, Validate, Install and Logs buttons don't do anything either.
I am on a older version of mac os high sierra with safari version Version 13.1.2 (13609.3.5.1.5)
I believe this to be a front end issue with the new 1.19.x+

erikpendragon, 2021-06-22:
ok, i have found a way to duplicate the failure. it i right click on my tab and say "duplicate tab" the next tab will some times come up correctly some times i have had to duplicate 6 times to get one that works.
After i have one that works i can go back to all of the ones that didn't work and refresh them and they do work.
but if i clear my cache and refresh any of the tabs that do work then it doesn't work any more.

erikpendragon, 2021-06-22:
and if i look at my network traffic i don't get the error with index-98a3f8c8.js any more

Summary
URL: http://192.168.1.96:8123/api/hassio_ingress/CtOVCQYxMUI9kWbqU4k4m099cyidauntT2y7fXM84xs/static/js/esphome/index-98a3f8c8.js
Status: 200 OK
Source: Memory Cache

Request
No request, served from the memory cache.

Response
X-Content-Type-Options: nosniff
Content-Type: application/javascript
Last-Modified: Fri, 18 Jun 2021 00:16:22 GMT
ETag: "39227d1da6c789bf865bcf3d644c66655d7db241e5caba82e7cc1a743d4f9592c2d9510e814c092678539f5674aab360d8a3a85dc041580eaa2d5f2d772421a5"
Accept-Ranges: bytes
Date: Tue, 22 Jun 2021 21:04:48 GMT
X-XSS-Protection: 1; mode=block
Content-Length: 16793
Connection: keep-alive
Server: nginx
X-Robots-Tag: none

erikpendragon, 2021-06-23:
ohh, but even after i "get it to work" with the duplication trick the Install button still doesn't work, Edit, Validate, and Logs buttons do

erikpendragon, 2021-06-23:
everything works if i go to the ip:port of esphome https://192.168.1.xx:6052, it just doesn't work in a frame within home assistant

erikpendragon, 2021-07-02:
fix listed here @OttoWinter 

https://github.com/esphome/dashboard/issues/22#issuecomment-869336124

closed 2021-11-15 by github-actions[bot]