esphome/issues #2176: ESP8266 resets or does not receive sensor data from HA after updating from 1.18

TheStork239, 2021-06-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, Wemos D1 Mini

**ESPHome version (latest production, beta, dev branch)**

1.19.2 (issue started after upgrading from 1.18, worked fine there)

**Affected component:**

Waveshare e-ink display
https://esphome.io/components/display/waveshare_epaper.html

**Description of problem:**
The code for my waveshare 4.2in e-ink display worked perfectly in ESPHome 1.18, but after upgrading to 1.19.1 and 1.19.2 the same yaml configuration causes the ESP to either reset (sometimes repeatedly) or become unresponsive to Home Assistant.  Quite often the ESP will fetch only one of the HA sensor values to be displayed, and become unresponsive after that, as is the case in the example log file below.  (In 1.18 it fetches and displays values from several different sensors and updates them on the display every 600 seconds).

I have the previous bin file created in 1.18, and have tried compiling the same yaml configuration in 1.19.2 (and before then in 1.19.1).  The bin file compiled with the 1.19.x versions have this issue, while the bin file created with 1.18 using the same yaml config file works fine without any resets or becoming unresponsive. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: epaper-display
  platform: ESP8266
  board: esp01_1m
  on_boot:
    priority: -10
    then:
      - delay: 10s
      - component.update: epaperdisplay

wifi:
  ssid: "removed"
  password: "removed"

logger:

# Enable Home Assistant API
api:

ota:
  password: "removed"

font:

  - file: 'fonts/Roboto-Medium.ttf'
    id: roboto_med_30
    size: 30
  - file: 'fonts/Roboto-Regular.ttf'
    id: footer_font
    size: 15
    glyphs:
      ['&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
       'g', 'h', 'i', 'j', 'k', 'l', 'm', 'n', 'o', 'p', 'q', 'r', 's', 't',
       'u', 'v', 'w', 'x', 'y', 'z','å', 'ä', 'ö', '/','º','µ','³']

  - file: 'fonts/Roboto-Regular.ttf'
    id: main_sensor_unit
    size: 20
    glyphs:
      ['&', '@', '!', ',', '.', '"', '%', '(', ')', '+', '-', '_', ':', '°', '0',
       '1', '2', '3', '4', '5', '6', '7', '8', '9', 'A', 'B', 'C', 'D', 'E',
       'F', 'G', 'H', 'I', 'J', 'K', 'L', 'M', 'N', 'O', 'P', 'Q', 'R', 'S',
       'T', 'U', 'V', 'W', 'X', 'Y', 'Z', ' ', 'a', 'b', 'c', 'd', 'e', 'f',
[... 495 lines cut ...]
[17:26:26][C][homeassistant.sensor:029]:   Unit of Measurement: ''
[17:26:26][C][homeassistant.sensor:029]:   Accuracy Decimals: 1
[17:26:26][C][homeassistant.sensor:030]:   Entity ID: 'sensor.woking_probability_of_precipitation'
[17:26:26][C][homeassistant.sensor:029]: Homeassistant Sensor 'wind_gust'
[17:26:26][C][homeassistant.sensor:029]:   State Class: ''
[17:26:26][C][homeassistant.sensor:029]:   Unit of Measurement: ''
[17:26:26][C][homeassistant.sensor:029]:   Accuracy Decimals: 1
[17:26:26][C][homeassistant.sensor:030]:   Entity ID: 'sensor.woking_wind_gust'
[17:26:26][C][homeassistant.sensor:029]: Homeassistant Sensor 'river_level'
[17:26:26][C][homeassistant.sensor:029]:   State Class: ''
[17:26:26][C][homeassistant.sensor:029]:   Unit of Measurement: ''
[17:26:26][C][homeassistant.sensor:029]:   Accuracy Decimals: 1
[17:26:26][C][homeassistant.sensor:030]:   Entity ID: 'sensor.woking_south_water_level_stage'
[17:26:26][C][homeassistant.text_sensor:022]: Homeassistant Text Sensor 'wind_dir'
[17:26:26][C][homeassistant.text_sensor:023]:   Entity ID: 'sensor.woking_wind_direction'
[17:26:26][C][homeassistant.text_sensor:022]: Homeassistant Text Sensor 'weather_status'
[17:26:26][C][homeassistant.text_sensor:023]:   Entity ID: 'weather.met_office_woking'
[17:26:26][D][homeassistant.sensor:023]: 'sensor.awair_element_air_quality_humidity': Got state 53.94
[17:26:26][D][sensor:117]: 'humid_indoor': Sending state 53.94000  with 1 decimals of accuracy
[17:26:35][I][WiFi:374]: Good

```

**Additional information and things you've tried:**
The bin file created in 1.18 works fine, while the same yaml configuration compiled in 1.19.2 (and before then in 1.19.1) have this issue of resets and broken fetching of sensor data.

OttoWinter, 2021-06-23:
> last failed alloc call: 4020F4F5(24)

your device is running out of memory - this could be because your last config was just close to the limit, and changes in 1.19 just bumped it over the edge.

It may help to go into the `fonts` glyphs (for example in Roboto-bold) and manually enabled _just_ the glyphs you really need (the default is the whole ASCII latin set, so quite a few characters).

TheStork239, 2021-06-23:
Thanks @OttoWinter, I think you are right.  I reduced the number of fonts / glyphs and that got it working again.  Interestingly it wasn't shown as high RAM usage at time of compiling, and the fact that it had worked with a previous ESPHome version really threw me, as I incorrectly assumed it was not my code causing it.

closed 2021-06-23 by TheStork239