esphome/issues #2177: ESP32 - Arduino espressif compile warning

chemelli74, 2021-06-22:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.19.2

**Affected component:**

https://github.com/espressif/arduino-esp32/releases/tag/1.0.6

**Description of problem:**
Compile issue

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

```

**Additional information and things you've tried:**

agners, 2021-06-23:
This code comes from the Arduino SDK. It seems that it has been addressed upstream: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-spi.c#L1274

So this will solve itself with the next update.

OttoWinter, 2021-06-23:
And to be clear: This is just a compiler warning for code we don't control. You can safely ignore it.

closed 2021-06-24 by chemelli74