esphome/issues #2178: Uptime: device_class ignored

chemelli74, 2021-06-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

v1.19.2

**Affected component:**

uptime

**Description of problem:**
device_class even if accepted by "esphome config xxx.yaml" is not honored

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: uptime
    name: ${device_name} Uptime
    device_class: timestamp
```

**Logs (if applicable):**

```
[02:33:26][C][uptime.sensor:030]: Uptime Sensor 'Gateway BLE Salotto Uptime'
[02:33:26][C][uptime.sensor:030]:   Device Class: 'timestamp'
[02:33:26][C][uptime.sensor:030]:   State Class: ''
[02:33:26][C][uptime.sensor:030]:   Unit of Measurement: 's'
[02:33:26][C][uptime.sensor:030]:   Accuracy Decimals: 0
[02:33:26][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
```

**Additional information and things you've tried:**

`[02:35:30][D][sensor:117]: 'Gateway BLE Salotto Uptime': Sending state 813.99701 s with 0 decimals of accuracy`

OttoWinter, 2021-06-23:
What do you mean by ignored? The logs look just fine to me.

chemelli74, 2021-06-23:
> What do you mean by ignored? The logs look just fine to me.

Value is shown as `seconds` not as `timestamp`

Simone

OttoWinter, 2021-06-23:
Yes... the `seconds` is because the unit_of_measurement is set to `s`. If you want to change the unit to timestamp you can do so using `unit_of_measurement`.

Note that unit as well as device_class makes no semantic difference for esphome, it just reports those settings straight to HA.

chemelli74, 2021-06-24:
> Yes... the `seconds` is because the unit_of_measurement is set to `s`. If you want to change the unit to timestamp you can do so using `unit_of_measurement`.

Usually if you specify a `device_class` there is no need for the `unit_of measurement`; 
at least this is the way HA works, or am I wrong ?

And the value should be represented in a different way:

- `unit_of_measurement: s` => 50
- `device_class: timestamp` => 1624542425.771909

Simone

OttoWinter, 2021-06-24:
> at least this is the way HA works, or am I wrong ?

No, esphome works differently. the integration sets the unit of measurement/device_class. You are given the option to _change_ the unit of measurement, but that's only for cosmetic purposes and doesn't influence any semantics in ESPHome.

> And the value should be represented in a different way:

I think I finally understand what you mean. No, `device_class: timestamp` does not work because `uptime` measures uptime in seconds since device start - if you change `device_class` that has no semantic difference in ESPHome whatsoever (the uptime sensor still reports uptime in seconds). Anything different would be difficult anyway, because ESPs _don't_ have real time clocks, so they don't even know what time to measure off of (only when you add `time` component, but `uptime` is independent of that).

chemelli74, 2021-06-24:
Got it.
Then you should ignore `device_timestamp` if applied to uptime sensor so when a user execute "esphome config xxx.yaml" this param is not there and is clear is ignored ;-)

For it to be compliant with other HA sensors, will open a issue there.

Simone

chemelli74, 2021-06-25:
Workaround from Frenk:

```yaml
time:
  - platform: homeassistant
    id: homeassistant_time
    timezone: "Europe/Rome"
    on_time_sync:
      - component.update: sensor_uptime_timestamp

sensor:
  - platform: uptime
    name: Internal Uptime
    id: sensor_uptime
    internal: true

  - platform: template
    id: sensor_uptime_timestamp
    name: ${device_name} Uptime
    device_class: "timestamp"
    accuracy_decimals: 0
    update_interval: never
    lambda: |-
      static float timestamp = (
        id(homeassistant_time).utcnow().timestamp - id(sensor_uptime).state
      );
      return timestamp;
```

closed 2021-06-25 by chemelli74