esphome/issues #2179: Changes to configuration in ESPHome is not automatically updated in Home Assistant

henkburnett, 2021-06-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io Core: core-2021.6.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, D1-Mini
**ESPHome version (latest production, beta, dev branch)**

1.19.2
**Affected component:**

Device entities are not updated in Home Assistant

**Description of problem:**
When you make changes to the configuration yaml of a device, it doesn't update Home Assistant automatically.
The only way you are able to push the changes through to Home Assistant is to delete the device from the ESPHome integation and restart Home Assistant. Home Assistant then discovers the changes and creates the applicable sensors/switches

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
I see no errors in the ESPhome addin logs.

```

**Additional information and things you've tried:**

I have 2 instances of Home assistant. The second instance does not have the ESPHome addon installed, but does have the ESPHome module enabled under integrations. This instance automatically updates the entities in it's Home Assistant withou having the need to replace it.

OttoWinter, 2021-06-23:
Can you describe what steps you took in detail?

In general, the workflow is this:
* In the ESPHome **add-on** (the grid interface where you can edit the node configs), edit the configuration
* Still in the add-on click **install** (formerly upload)
* When that is complete, your HA instance will reconnect to the node, and update the changes. Any new entities are added automatically.

henkburnett, 2021-06-23:
Hi Otto
In ESPHome add-on I edit a device config (like changing the name: field of an entity.
I then go Save and Install and the change is compiled and uploaded to the device.
The changes are saved correctly to the device itself, but the name is not changed in my Home Assistant instance.

As I have noted before, I do run 2 Home Assistant instances, both with the ESPHome integration. (My second device does NOT have the ESPHome Addin installed though)

Once the device reboots, I could normally see BOTH my Home Assistant instances connecting to my device in the logs windows.
After I had updated only the ESPHome addin, on that same Home Assistant instance is where I no longer see the changes automatically pushed to Home Assistant.
The changes do push to my secondary Home Assistant instance.

On the reboot, if I monitor the logs, I only see my secondary Home Assistant instance communicate via the API connection. I used to see both Home Assistants connect to my device.
[12:09:30][V][api.connection:628]: Hello from client: 'Home Assistant 2021.6.4 (192.168.0.100)'
[12:09:30][D][api.connection:644]: Client 'Home Assistant 2021.6.4 (192.168.0.100)' connected successfully!

I hope this makes sense

henkburnett, 2021-06-30:
Good day
On further testing, I have established that if  I make changes to the configuration in the ESP-home addon, I can go to the ESPhome Integration and disable the specific device. When I then re enable the device in the Integration, the changes I had made pushes through to Home Assistant.

OttoWinter, 2021-07-10:
Please enable verbose logging on the HA instance that is not reconnecting (see `logger` section in HA configuration.yaml) and post those logs.

Also, do you mean _just_ the device updates aren't pushed to HA anymore or is the device showing as unavailable?

closed 2021-11-22 by github-actions[bot]