esphome/issues #2180: MQTT component not recognized when included using package

depuits, 2021-06-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Manjaro linux AUR and Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 but I don't think the board is relevant

**ESPHome version (latest production, beta, dev branch)**

v1.19.2

**Affected component:**

https://esphome.io/components/mqtt.html

**Description of problem:**
Adding retain option and presumable other mqtt options to components produce an error when the mqtt component was set up using packages.

**Problem-relevant YAML-configuration entries:**
```yaml
# main file (test.yaml)
substitutions:
  devicename: test
  topic: 'test'

packages:
  device_base: !include common/esp32.yaml

binary_sensor:
  - platform: gpio
    name: "Doorbell"
    retain: False
    pin: GPIO27

# common/esp32.yaml
esphome:
  name: $devicename
  platform: ESP32
  board: wemos_d1_mini32

<<: !include base.yaml

# common/base.yaml
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: $devicename
    password: !secret wifi_password

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password
  topic_prefix: $topic
  
captive_portal:
logger:

ota:
  password: !secret device_password
   
status_led:
  pin:
    number: GPIO2
    inverted: False
```

**Logs (if applicable):**

```
esphome compile test.yaml 
INFO Reading configuration test.yaml...
Failed config

binary_sensor.gpio: [source test.yaml:9]
  platform: gpio
  name: Doorbell
  
  This option requires component mqtt.
  retain: False [source test.yaml:11]
  pin: GPIO27
```

**Additional information and things you've tried:**

whi-tw, 2021-07-07:
Same as #1487

whi-tw, 2021-07-07:
Workaround recommendation:

```yaml
substitutions:
  devicename: garage
  upper_devicename: Garage

packages:
  sonoff_mini: !include common/sonoff_mini.yaml

<<: !include common/mqtt.config.yaml
```

paulmonigatti, 2021-09-17:
An interim fix was implemented in esphome/esphome#2210, and a final fix in esphome/esphome#2193. Should all be working now!