esphome/issues #2181: esp32 with wifi and bluetooth enabled go into bootloop

Haringstad, 2021-06-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**
esp32 esp-wrover-kit
**ESPHome version (latest production, beta, dev branch)**
latest production v 1.19.2
**Affected component:**
wifi
**Description of problem:**
esp goes into bootloop, ending in safe mode due to error:
**Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!**

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: 3rd_floor_multisensor
  upper_devicename: "3rd Floor multi Sensor"

esphome:
  name: ${devicename}
  platform: ESP32
  board: esp32doit-devkit-v1
#  board_flash_mode: dout
  
wifi:
  ssid: "HASSIO_IoT"
  password: "xxxxx"
  manual_ip:
    static_ip: 192.168.x.x
    gateway: 192.168.x.x
    subnet: 255.255.255.0
    dns1: 192.168.x.x
    dns2: 192.168.x.x
  domain: .xxxx.loc
  power_save_mode: none
  output_power: 17.5db  
  fast_connect: false
  ap:
    ssid: "3rd floor sensor Hotspot"
    password: "datisgeheim"  

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 0s
  password: "datisookgeheim"

ota:
  password: "endezeook"

captive_portal:  

time:
  - platform: sntp
    id: sntp_time
    timezone: "Europe/Amsterdam"
    servers: 
      - 192.168.x.x
      - 192.168.x.x
    on_time:
      - seconds: 0
        minutes: 0
        hours: 4
        days_of_week: MON-SUN
        then:
           - switch.toggle: restart_switch

esp32_ble_tracker:
  scan_parameters:
    active: false

switch:
  - platform: restart
    name: "Reboot 3rd Floor Multisensor"
    id: restart_switch
    
sensor:
  - platform: wifi_signal
    name: "3rd Floor Multisensor WiFi Signal"
    update_interval: 120s
    id: third_floor_multisensor_wifi_signal
    filters:
      - filter_out: nan
      - heartbeat: 60s
      - sliding_window_moving_average:
         window_size: 2
         send_every: 2
         send_first_at: 1
  - platform: uptime
    name: "3rd Floor Multisensor Uptime"
    id: third_floor_multisensor_uptime
    update_interval: 1h
  - platform: esp32_hall
    name: "3rd Floor Mulltisensor Hall Sensor"
    update_interval: 60s
  - platform: xiaomi_cgg1
    mac_address: "x.x.x.x.x"
    temperature:
      name: "Mijia 3rd Floor Temperature"
    humidity:
      name: "Mijia 3rd Floor Humidity"
    battery_level:
      name: "Mijia 3rd Floor Battery Level"
  - platform: ble_rssi
    mac_address: x:x:x:x
    name:  "P40"

text_sensor:
  - platform: version
    name: "3rd Floor Multisensor ESPHome Version"
    id: bathroom_switch_firmware
  - platform: wifi_info
    ip_address:
      name: "3rd Floor Multisensor IP"
    ssid:
      name: "3rd Floor Multisensor SSID"
    bssid:
      name: "3rd Floor Multisensor BSSID"
#  - platform: ble_scanner
#    name: "BLE Devices Scanner"
    
binary_sensor:
  - platform: status
    name: "3rd Floor Multisensor Status"
    id: third_floor_multisensor_status

web_server:
  port: 80  

```

**Logs (if applicable):**

```
INFO Reading configuration /config/3rd_floor_sensor.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 27 March 02:00:00 to 30 October 03:00:00
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[10:57:07]ets Jun  8 2016 00:22:57
[10:57:07]
[10:57:07]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:57:07]configsip: 0, SPIWP:0xee
[10:57:07]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:57:07]mode:DIO, clock div:2
[10:57:07]load:0x3fff0018,len:4
[10:57:07]load:0x3fff001c,len:1044
[10:57:07]load:0x40078000,len:8896
[10:57:07]load:0x40080400,len:5828
[10:57:07]entry 0x400806ac
[10:57:07][I][logger:170]: Log initialized
[10:57:07][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[10:57:07][I][app:029]: Running through setup()...
[10:57:08][D][esp32_ble_tracker:180]: Starting scan...
[10:57:08][D][text_sensor:015]: '3rd Floor multi Sensor ESPHome Version': Sending state '1.19.2 Jun 23 2021, 12:37:25'
[10:57:08][D][binary_sensor:034]: '3rd Floor multi Sensor Status': Sending initial state OFF
[10:57:08][C][wifi:037]: Setting up WiFi...
[10:57:08][D][esp-idf:000]: E (1851) wifi:
[10:57:08][D][esp-idf:000]: Error! Should enable WiFi modem sleep when both WiFi and Bluetooth are enabled!!!!!!
[10:57:08]
[10:57:08][D][esp-idf:000]: 
[10:57:08]
[10:57:08]abort() was called at PC 0x40127a3f on core 0
[10:57:08]
[10:57:08]ELF file SHA256: 0000000000000000
[10:57:08]
[10:57:08]Backtrace: 0x4008f8cc:0x3ffe64a0 0x4008fb49:0x3ffe64c0 0x40127a3f:0x3ffe64e0 0x400ffb50:0x3ffe6500 0x400fe746:0x3ffe6520 0x4012a56e:0x3ffe6540 0x40090bbe:0x3ffe6570
WARNING Found stack trace! Trying to decode it
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /config/3rd_floor_multisensor -t idedata
UnknownPackage: Detected unknown package 'tool-unity':
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/main.py", line 220:
    projenv.DumpIDEData()  # pylint: disable=undefined-variable
  File "/root/.platformio/packages/tool-scons/script/../engine/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/pioide.py", line 161:
    "includes": _dump_includes(env),
  File "/usr/local/lib/python3.6/dist-packages/platformio/builder/tools/pioide.py", line 62:
    unity_dir = get_core_package_dir("tool-unity")
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/core.py", line 89:
    return pm.install(name, requirements)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/core.py", line 55:
    PackageManager.install(self, name, requirements, *args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/package.py", line 704:
    pkg_dir = self._install_from_piorepo(name, requirements)
  File "/usr/local/lib/python3.6/dist-packages/platformio/managers/package.py", line 458:
    name + (". Error -> %s" % last_exc if last_exc else "")
========================== [FAILED] Took 5.29 seconds ==========================

```

**Additional information and things you've tried:**

When I use versions pre v1.19.*, everything works fine.

CarlosGS, 2021-06-23:
Linking relevant issue "power_save_mode logic is twisted in ESP8266" https://github.com/esphome/issues/issues/1532
TL/DR: Modem sleep is actually "high" (the current "light" sleep also reduces CPU frequency and causes dropouts)

Haringstad, 2021-06-23:
But, adding `power_save_mode: high` did solve my problem......

ssieb, 2021-06-24:
Yes, you can't use `power_save_mode: none` with bluetooth.  A change has been added to fail the compile in that case.
https://github.com/esphome/esphome/pull/1950

closed 2021-06-24 by ssieb