esphome/issues #2183: "LOGS" Button Does Not Function in MS Edge in ESPHome Add-On

mbaran5, 2021-06-23:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 2021.6.4 / Home Assistant OS 6.0 / ESPHome 1.19.3 running on VMware
Browser MS Edge 91.0.864.48 on Windows 10 20H2

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**ESPHome version (latest production, beta, dev branch)**

ESP Home 1.19.3

**Affected component:**

https://esphome.io/guides/getting_started_hassio.html

**Description of problem:**

When clicking on the LOGS button in ESPHome add-on, nothing occurs. The behavior works fine in Chrome 91.0.4472.114

**Problem-relevant YAML-configuration entries:**
N/A

**Logs (if applicable):**

N/A

**Additional information and things you've tried:**

DaleLJefferson, 2021-06-24:
I'm seeing the same issue in Brave browser, same error message.

ESPHome 1.19.4
Brave V1.26.67

renmatheus, 2021-06-25:
Same issue

andste97, 2021-06-25:
I have the same issue in Google Chrome.

ESPHome version on Home Assistant: 1.19.4
Chrome version: 91.0.4472.124

jasperslot, 2021-06-27:
Same issue here... on Mac OS with Chrome 91. It works on Safari however...

MarcHagen, 2021-06-27:
The problem is here:
https://github.com/esphome/dashboard/blob/ed3eed004e99068e2cb3453f7fe8820ec72a0e9a/src/api/serial-ports.ts#L12

Then fetchApi doesn't appear to return a javascript array but regular text.
Which is wierd because this is checked here: https://github.com/esphome/dashboard/blob/ed3eed004e99068e2cb3453f7fe8820ec72a0e9a/src/api/index.ts#L13-L15

This is not a browser cache problem.
`/serial-ports` does indeed return a json text, and the `content-type: application/json; charset=utf-8` header.

In my testing, `resp.headers.get("content-type")` returned: `application/json; charset=utf-8` which !== `application/json`
Which is where all hell goes lose :) I'm not quite sure where the `charset=utf-8` came from. Nor why this isn't propperly filtered out.

balloob, 2021-06-28:
Thanks for finding the issue @MarcHagen 

Weird that the header would be changed. It's set explicitely to `application/json`: 

https://github.com/esphome/esphome/blob/3b940b1c04ebabf520d1494a36c719e5ab27caad/esphome/dashboard/dashboard.py#L356

By any chance anyone running anything in front of their dashboard, nginx or anything ?

closed 2021-06-28 by balloob

mbaran5, 2021-06-28:
I know this is closed, but yes HAProxy in front of mine. I didn’t think to test direct.

MarcHagen, 2021-06-28:
Me too, using nginx with minimum Config.

I've came across this issue in my own apps too, pure JS doet what you want add some nginx, haproxy or even browsers and wierf stuff happens 😅

Thanks for merging!

juicejuice, 2021-08-02:
I see the fix was merged over a month ago but I'm still getting this exact issue in ESPHome 1.20.3. I can't seem to find how to track what commits went into that build :( anyone know?

juicejuice, 2021-08-02:
Thanks @MarcHagen! Your instincts were correct - the "Empty cache and hard refresh" option fixed it. I wonder how to get the browser to invalidate the cache for such files to avoid the issue (beyond my working knowledge)?