esphome/issues #2184: Hostname always "espressif" when using Ethernet on ESP32

ljaco, 2021-06-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 (WT32-ETH01)

**ESPHome version (latest production, beta, dev branch)**

ESPHome version v1.19.2

**Affected component:**

https://esphome.io/components/ethernet.html

**Description of problem:**

When using ethernet, the hostname is not set correctly and is always "espressif" (instead of "esp32-test-ethernet" as defined in the example below). Therefore, communication between HA and the board does not work correctly. Accessing the board works only with "espressif" as hostname or directly via its IP. In the logs, it shows the correct hostname and no indication that it has failed to set it. When using wifi instead of ethernet everything works fine and the hostname is set correctly.

Here the DHCP Lease:
192.168.1.47	a8:03:2a:21:f3:dc		espressif		2021/06/23 09:29:02	2021/06/23 11:29:02	online	active

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: esp32-test-ethernet
  devicename_underscore: esp32_test_ethernet
  devicename_pretty: ESP32 Test Ethernet

esphome:
  name: $devicename
  platform: ESP32
  board: nodemcu-32s

# wifi:
#   ssid: !secret wifi_ssid
#   password: !secret wifi_password
#   domain: !secret mydomain
#   ap:
#     ssid: $devicename
#     password: !secret ap_password
    
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16
  domain: !secret mydomain

web_server:
  port: !secret web_server_port

logger:
  level: VERY_VERBOSE

api:
  password: !secret api_password

ota:
  password: !secret ota_password

sensor:
- platform: uptime
  name: $devicename_pretty Uptime
  update_interval: 60s

```
[... 140 lines cut ...]
[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=19285 (now=29285)
[VV][scheduler:152]: Running interval 'update' with interval=60000 last_execution=4294937678 (now=30385)
[V][sensor:023]: 'ESP32 Test Ethernet Uptime': Received new state 30.386999
[D][sensor:117]: 'ESP32 Test Ethernet Uptime': Sending state 30.38700 s with 0 decimals of accuracy
[VV][api.service:220]: on_ping_request: PingRequest {}
[VV][api.service:032]: send_ping_response: PingResponse {}
[VV][api.service:220]: on_ping_request: PingRequest {}
[VV][api.service:032]: send_ping_response: PingResponse {}
[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=29285 (now=39285)
[VV][api.service:220]: on_ping_request: PingRequest {}
[VV][api.service:032]: send_ping_response: PingResponse {}
[VV][api.service:220]: on_ping_request: PingRequest {}
[VV][api.service:032]: send_ping_response: PingResponse {}
[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=39285 (now=49285)
[VV][api.service:220]: on_ping_request: PingRequest {}
[VV][api.service:032]: send_ping_response: PingResponse {}
[VV][api.service:220]: on_ping_request: PingRequest {}
[VV][api.service:032]: send_ping_response: PingResponse {}
[VV][scheduler:152]: Running interval '' with interval=10000 last_execution=49285 (now=59285)
[VV][api.service:220]: on_ping_request: PingRequest {}
[VV][api.service:032]: send_ping_response: PingResponse {}

```

**Additional information and things you've tried:**

meijie-jesse, 2021-07-04:
I have the same problem as you

flacjacket, 2021-07-12:
I was seeing this issue as well, and was able to put together something that fixes this for me. If you are able to test if esphome/esphome#2010 is able to fix this for you as well, I'd appreciate it!

ljaco, 2021-07-13:
I'm unfortunately not able to test it as I don't have a development environment set up.

closed 2021-07-15 by jesserockz