esphome/issues #2185: cant flash wemos d1 mini

divemasterjm, 2021-06-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
hass.io
Please provide details about your environment below this line. -->

**ESP (ESP32/ESP8266, Board/Sonoff):**
wemos d1 mini
Please provide details about which ESP you're using below.
-->

**ESPHome version (latest production, beta, dev branch)**
latest production
ESPHome version like v1.14 or 1.15-dev
-->

**Affected component:**

**Description of problem:**
after flashing, i've got flash ok but when reboots i have the following error

[09:54:00]
[09:54:00] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[09:54:00]
[09:54:00]wdt reset
[09:54:00]load 0x4010f000, len 3584, room 16 
[09:54:00]tail 0
[09:54:00]chksum 0xb0
[09:54:00]csum 0xb0
[09:54:00]v2843a5ac
[09:54:00]~ld
[09:54:00][I][logger:170]: Log initialized
[09:54:00][C][ota:366]: There have been 7 suspected unsuccessful boot attempts.
[09:54:00][I][app:029]: Running through setup()...
[09:54:00][C][wifi:037]: Setting up WiFi...
[09:54:00][D][wifiStarting scan...

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: malo
  platform: ESP8266
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "312dbae27a27b360eba44e3d1e47b358"

wifi:
  ssid: "ssid"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Malo Fallback Hotspot"
    password: "jSqhFrajQYBg"

captive_portal:
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

i've tried flashing tasmota and works ok, but any attent to flash esphome is failed, flash is ok but after reboot not working
tried 
esphome add-on
esphome
esphome-flasher

ssieb, 2021-06-24:
Try using a different power source.

divemasterjm, 2021-06-24:
Using 'COM4' as serial port.
Connecting....
Detecting chip type... ESP8266
Connecting....

Chip I(94 %)Writing at 0x00040000... (100 %)Wrote 393984 bytes (273851 compressed) at 0x00000000 in 6.9 seconds (effective 458.4 kbit/s)...
Hash of data verified.

Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[22:48:47][I][logger:170]: Log initialized
[22:48:47][C][ota:366]: There have been 5 suspected unsuccessful boot attempts.
[22:48:47][I][app:029]: Running through setup()...
[22:48:47][C][wifi:037]: Setting up WiFi...
[22:48:47][D][wifi:365]: Starting scan...
[22:48:55]
[22:48:55] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[22:48:55]
[22:48:55]wdt reset
[22:48:55]load 0x4010f000, len 3584, room 16 
[22:48:55]tail 0
[22:48:55]chksum 0xb0
[22:48:55]csum 0xb0
[22:48:55]v2843a5ac
[22:48:55]~ld
[22:48:56][I][logger:170]: Log initialized
[22:48:56][C][ota:366]: There have been 6 suspected unsuccessful boot attempts.
[22:48:56][I][app:029]: Running through setup()...
[22:48:56][C][wifi:037]: Setting up WiFi...
[22:48:56][D][wifi:Mхѥ scan...
[22:49:04]
[22:49:04] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[22:49:04]
[22:49:04]wdt reset
[22:49:04]load 0x4010f000, len 3584, room 16 
[22:49:04]tail 0
[22:49:04]chksum 0xb0
[22:49:04]csum 0xb0
[22:49:04]v2843a5ac
[22:49:04]~ld
[22:49:04][I][logger:170]: Log initialized
[22:49:04][C][ota:366]: There have been 7 suspected unsuccessful boot attempts.
[22:49:04][I][app:029]: Running through setup()...
[22:49:04][C][wifi:037]: Setting up WiFi...
[22:49:04][D][wifi:Starting scan...
[22:49:04]

gb53smith, 2021-06-25:
Ideally, the board should be "d1_mini" , although mine did work with "esp01_1m".
Some of the clones have a 150 mA voltage regulator on board which has been report to result in an unreliable WiFi connection.
I got rid of
```
  ap:
    ssid: "Malo Fallback Hotspot"
    password: "jSqhFrajQYBg"

captive_portal:
```
which comes by default.  It is not needed for Home Assistant integration.

divemasterjm, 2021-06-25:
i had 4 wemos, 3 went ok, only this one having issues with esphome, if i flash tasmota it works, i've tried with d1_mini having same result

gb53smith, 2021-06-25:
If "3 went OK" do you just have a one defective wemos?

divemasterjm, 2021-06-25:
this was my first thought, but why is working with tasmota without issues?

closed 2021-11-08 by github-actions[bot]