esphome/issues #2186: Issue with temperature readings after upgrade to 1.19.x

keith-gajeski, 2021-06-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io
Please provide details about your environment below this line. -->

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32
Please provide details about which ESP you're using below.
ESP32 Devkit

**ESPHome version (latest production, beta, dev branch)**
latest production
ESPHome version like v1.14 or 1.15-dev
1.19

**Affected component:**
NTC Sensor
Please add the link to the documentation at https://esphome.io/index.html of the component in question.
https://esphome.io/components/sensor/ntc.html

**Description of problem:**
Good Afternoon. I have an ESP32 that is monitoring the temperature of my cold and hot water in my home. Since the update the temperature readings have been really off. I have reverted to 1.18 and everything is working fine. I will include my yaml and the logs from both 1.18 version and 1.19.x
I marked the issues in the log in bold print, to make it easier to find. Also, this is my first Github post, so if I did something wrong, please let me know!

Thanks!
Keith

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: 618_blueshutters_basement
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: <snip>
  password: <snip>
  
captive_portal:  

switch:
  - platform: gpio
    pin: 16
    inverted: yes
    id: MainWaterOn
    interlock: [MainWaterOff]
  - platform: template
    id: MainWaterOnTP
    turn_on_action:
    - switch.turn_on: MainWaterOn
    - delay: 6 s
    - switch.turn_off: MainWaterOn

  - platform: gpio
    pin: 17
    inverted: yes
    id: MainWaterOff
    interlock: [MainWaterOn]
  - platform: template
    id: MainWaterOffTP
    turn_on_action:
    - switch.turn_on: MainWaterOff
    - delay: 6 s
    - switch.turn_off: MainWaterOff    

  - platform: gpio
    pin: 25
    inverted: yes
    id: HydroponicsOn
    interlock: [HydroponicsOff]
  - platform: template
[... 914 lines cut ...]
[12:08:50][D][sensor:099]: 'Cold Water Temperature': Sending state 14.38544 °C with 1 decimals of accuracy
[12:09:00][D][adc:058]: 'Postfilter': Got voltage=0.07V
[12:09:00][D][sensor:099]: 'Postfilter': Sending state 0.06930 V with 2 decimals of accuracy
[12:09:01][D][adc:058]: 'hot_sensor': Got voltage=0.68V
[12:09:01][D][sensor:099]: 'hot_sensor': Sending state 0.67961 V with 2 decimals of accuracy

[12:09:01][D][resistance:037]: 'Hot_Water_Temp' - Resistance 38557.3Ω
[12:09:01][D][sensor:099]: 'Hot_Water_Temp': Sending state 38557.30859 Ω with 1 decimals of accuracy
[12:09:01][D][ntc:026]: 'Hot Water Temperature' - Temperature: 30.7°C

[12:09:01][D][sensor:099]: 'Hot Water Temperature': Sending state 30.65219 °C with 1 decimals of accuracy
[12:09:07][D][pulse_counter:159]: 'Cold Water Flow': Retrieved counter: 0.00 pulses/min
[12:09:07][D][sensor:099]: 'Cold Water Flow': Sending state 0.00000 GPM with 2 decimals of accuracy
[12:09:10][D][pulse_counter:159]: 'Water Heater Status': Retrieved counter: 144.00 pulses/min
[12:09:11][D][adc:058]: 'Prefilter': Got voltage=0.21V
[12:09:11][D][sensor:099]: 'Prefilter': Sending state 0.20576 V with 2 decimals of accuracy
[12:09:11][D][pulse_counter:159]: 'Hot Water Flow': Retrieved counter: 0.00 pulses/min
[12:09:11][D][sensor:099]: 'Hot Water Flow': Sending state 0.00000 GPM with 2 decimals of accuracy
[12:09:12][D][sensor:099]: 'Water Heater Status': Sending state 144.00000 pulses/min with 2 decimals of accuracy

```

**Additional information and things you've tried:**

Reverted back to 1.18. I've tried all branches of 1.19.

agners, 2021-06-24:
Hm, I guess this could be related to ntc, resistance or adc component. You can try to use their 1.18 counter part separately using

```yaml
external_components:
  - source: github://esphome/esphome@v1.18.0
    components: [ntc] # try resistance and adc as well
```

ssieb, 2021-06-24:
I can verify that the adc readings on esp32 seem to be about half what they should be.  Nothing significant has changed in the esphome code.  I suspect some change in the Arduino layer or the Espressiv SDK.  I'll try to investigate a bit more later.

jesserockz, 2021-06-24:
Trying with both versions produces the same voltage result for me.

```yaml
esphome:
  ...
  # arduino_version: 1.0.4
  # arduino_version: 1.0.6
  
sensor:
  - platform: adc
    pin: 34
    id: adc_pin
    attenuation: 11dB
    update_interval: 1s
```

jesserockz, 2021-06-24:
ok, at `attenuation: 6dB` I get different results, around half the voltage reported

Perhaps linked to:
https://github.com/espressif/arduino-esp32/pull/3377

jesserockz, 2021-06-24:
Multimeter shows 0.822V

4 analog pins wired together with the different attenuation values and attached to above voltage

ESP32 Arduino Version 1.0.4:
```
[11:49:23][D][sensor:117]: 'adc_pin_11db': Sending state 0.80762 V with 2 decimals of accuracy
[11:49:23][D][sensor:117]: 'adc_pin_0db': Sending state 0.84803 V with 2 decimals of accuracy
[11:49:23][D][sensor:117]: 'adc_pin_2_5db': Sending state 0.86154 V with 2 decimals of accuracy
[11:49:23][D][sensor:117]: 'adc_pin_6db': Sending state 0.90310 V with 2 decimals of accuracy
```

ESP32 Arduino Version 1.0.6:
```
[11:53:35][D][sensor:117]: 'adc_pin_11db': Sending state 0.80762 V with 2 decimals of accuracy
[11:53:35][D][sensor:117]: 'adc_pin_2_5db': Sending state 0.31136 V with 2 decimals of accuracy
[11:53:35][D][sensor:117]: 'adc_pin_0db': Sending state 0.22725 V with 2 decimals of accuracy
[11:53:36][D][sensor:117]: 'adc_pin_6db': Sending state 0.45773 V with 2 decimals of accuracy
```

jesserockz, 2021-06-25:
Ok, so the above linked PR is the cause and it seems that the `analogRead` now returns a pre-scaled value. Previously it was not scaled and ESPHome was scaling, and now its doubling up basically, causing the bad readings.

jesserockz, 2021-06-25:
See also: https://github.com/espressif/arduino-esp32/issues/5041

SeByDocKy, 2021-06-28:
Ha now I understand why I have bad readings wuth my NTC :)

closed 2021-07-20 by jesserockz