esphome/issues #2188: Substitutions not working for some entities id

chemelli74, 2021-06-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

v1.19.2

**Affected component:**

https://esphome.io/components/sensor/wifi_signal.html
https://esphome.io/components/text_sensor/

**Description of problem:**
Substitutions are operated on the name but not on the entity id for both `wifi_signal` and `text_sensor`.
Work fine instead for `template`.

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  device_name: "Gateway BLE Salotto"

sensor:
  - platform: template
    id: sensor_uptime_timestamp
    name: ${device_name} Uptime
    device_class: "timestamp"
    accuracy_decimals: 0
    update_interval: never
    lambda: |-
      static float timestamp = (
        id(homeassistant_time).utcnow().timestamp - id(sensor_uptime).state
      );
      return timestamp;

  - platform: wifi_signal
    name: ${device_name} RSSI
    update_interval: 60s

text_sensor:
  - platform: version
    name: ${device_name} Version
    hide_timestamp: true
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

[image]

OttoWinter, 2021-06-28:
`entity_id` is generated by HA from the `name` property (esphome doesn't generate these)

So probably what happened here is that you first flashed a fw with the substitution undefined/or with a typo. -> `device_name RSSI` name was sent to HA, which it converted to an entity_id.

Then you updated the device, but because of the unique_id the old entity_id is kept.

chemelli74, 2021-07-01:
Strange as the change was done at once for all 3 sensors, and one successufully update.

This is the data in recreated `esphome.xxxx` storage file after removing the config, restarting HA and adding it back:
```
           {
                "object_id": "gateway_ble_salotto_uptime",
                "key": 3424092583,
                "name": "Gateway BLE Salotto Uptime",
                "unique_id": "esp32-salotto-ble-gatewaysensorgateway_ble_salotto_uptime",
                "icon": "",
                "device_class": "timestamp",
                "unit_of_measurement": "",
                "accuracy_decimals": 0,
                "force_update": false,
                "state_class": 0
            },
```
```
           {
                "object_id": "gateway_ble_salotto_rssi",
                "key": 916315470,
                "name": "Gateway BLE Salotto RSSI",
                "unique_id": "246f28b47d1c-wifisignal",
                "icon": "",
                "device_class": "signal_strength",
                "unit_of_measurement": "dBm",
                "accuracy_decimals": 0,
                "force_update": false,
                "state_class": 1
            },
```
I solved the issue the hard way: remove + re-add.

But I still think there is a issue under the wood.

Simone

OttoWinter, 2021-07-01:
Maybe, but I'd guess that's on the HA side then. ESPHome literally just tells HA "this is the name & unique_id of this entity" and HA then handles the entity_id conversion etc. 

So if you're getting the right `name` in your HA view, that means the ESPHome side works fine.

chemelli74, 2021-07-12:
After updating both ESPHome and HomeAssistant, now all entities have the right naming schema.

Simone

closed 2021-07-12 by chemelli74