esphome/issues #2191: Toggling switch in Web Server GUI can trigger Guru Meditation Error: Core  1 panic'ed (LoadProhibited)

5tan, 2021-06-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 20.04, esphome installed using pip.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, ESP-WROOM-32 on ESP DevKit v1

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

web_server

**Description of problem:**
I can trigger Guru Meditation error simply by toggling a switch in GUI. Please see video below:
https://user-images.githubusercontent.com/15819543/123428234-e14b1580-d5c5-11eb-96bd-29a8eea57f07.mp4

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: abc
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: ...
  password: ...

debug:

logger:
  level: DEBUG

ota:

web_server:
  port: 80
  auth:
    username: user
    password: pass

switch:
  - platform: gpio
    id: real_switch
    name: real switch
    pin: GPIO23
```

**Logs (if applicable):**

```
[14:54:25]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[14:54:25]Core 1 register dump:
[14:54:25]PC      : 0x400dcd60  PS      : 0x00060c30  A0      : 0x800dcdf6  A1      : 0x3ffb1d60  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d abc -t idedata
WARNING Decoded 0x400dcd60: AsyncEventSourceClient::_runQueue() at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
[14:54:25]A2      : 0x3ffd30bc  A3      : 0x70abba12  A4      : 0x81000000  A5      : 0x3ffd4192  
[14:54:25]A6      : 0x0000000a  A7      : 0x6c61663a  A8      : 0x81000000  A9      : 0x3ffb1d40  
[14:54:25]A10     : 0x00810000  A11     : 0x0000004f  A12     : 0x0000004f  A13     : 0x3ffd409c  
[14:54:25]A14     : 0x00000000  A15     : 0x007d6573  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[14:54:25]EXCVADDR: 0x00810008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[14:54:25]
[14:54:25]Backtrace: 0x400dcd60:0x3ffb1d60 0x400dcdf3:0x3ffb1d80 0x400dce0f:0x3ffb1da0 0x400dce65:0x3ffb1dc0 0x400d45e7:0x3ffb1df0 0x4015b30a:0x3ffb1e30 0x400d3cb5:0x3ffb1e50 0x400d1ed9:0x3ffb1e80 0x400d3bb2:0x3ffb1ed0 0x400d3e9d:0x3ffb1ef0 0x40081172:0x3ffb1f10 0x400d7c05:0x3ffb1f60 0x400da03e:0x3ffb1f90 0x400e4a29:0x3ffb1fb0 0x40088e7d:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400dcd60: AsyncEventSourceClient::_runQueue() at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400dcdf3: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400dce0f: AsyncEventSourceClient::write(char const*, unsigned int) at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400dce65: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400d45e7: esphome::web_server::WebServer::on_switch_update(esphome::switch_::Switch*, bool) at /home/z00lwix/test/abc/src/esphome/components/web_server/web_server.cpp:68
WARNING Decoded 0x4015b30a: std::_Function_handler<void (bool), esphome::Controller::setup_controller()::{lambda(bool)#1}>::_M_invoke(std::_Any_data const&, bool&&) at /home/z00lwix/test/abc/src/esphome/core/controller.cpp:35
 (inlined by) _M_invoke at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400d3cb5: std::function<void (bool)>::operator()(bool) const at /home/z00lwix/test/abc/src/esphome/components/switch/switch.cpp:53
 (inlined by) esphome::CallbackManager<void (bool)>::call(bool) at /home/z00lwix/test/abc/src/esphome/core/helpers.h:212
 (inlined by) esphome::switch_::Switch::publish_state(bool) at /home/z00lwix/test/abc/src/esphome/components/switch/switch.cpp:46
WARNING Decoded 0x400d1ed9: esphome::gpio::GPIOSwitch::write_state(bool) at /home/z00lwix/test/abc/src/esphome/components/gpio/switch/gpio_switch.cpp:101
WARNING Decoded 0x400d3bb2: esphome::switch_::Switch::toggle() at /home/z00lwix/test/abc/src/esphome/components/switch/switch.cpp:53
WARNING Decoded 0x400d3e9d: std::_Function_handler<void (), esphome::web_server::WebServer::handle_switch_request(AsyncWebServerRequest*, esphome::web_server::UrlMatch)::{lambda()#1}>::_M_invoke(std::_Any_data const&) at /home/z00lwix/test/abc/src/esphome/components/web_server/web_server.cpp:68
 (inlined by) _M_invoke at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x40081172: std::function<void ()>::operator()() const at /home/z00lwix/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:932
 (inlined by) esphome::Scheduler::call() at /home/z00lwix/test/abc/src/esphome/core/scheduler.cpp:158
WARNING Decoded 0x400d7c05: esphome::Application::loop() at /home/z00lwix/test/abc/src/esphome/core/application.cpp:70
WARNING Decoded 0x400da03e: loop() at /home/z00lwix/test/abc/src/main.cpp:126
WARNING Decoded 0x400e4a29: loopTask(void*) at /home/z00lwix/.platformio/packages/framework-arduinoespressif32@3.10004.210126/cores/esp32/main.cpp:19
WARNING Decoded 0x40088e7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[14:54:25]
[14:54:25]Rebooting...
```

**Additional information and things you've tried:**

5tan, 2021-06-25:
I wonder if this issue might be linked to one of these logging issues:
* https://github.com/esphome/issues/issues/2190
* https://github.com/esphome/issues/issues/2189

5tan, 2021-06-25:
One more observation: The issue never happened to me with website off, simply by making REST calls to toggle the switch: 
```
while [ true ] ; do time curl -u user:pass http://192.168.0.16/switch/real_switch/toggle -i -X POST ; sleep 0.5 ; done
```
So maybe issue is really related to the website logging?

5tan, 2021-06-25:
Yet another observation: error never happens if I change log level to INFO, and disable `debug` component (even with website/GUI shown).

5tan, 2021-06-25:
I think the issue is here: https://github.com/esphome/esphome/blob/dev/esphome/components/web_server/web_server.cpp#L296
```
void WebServer::on_switch_update(switch_::Switch *obj, bool state) {
  this->events_.send(this->switch_json(obj, state).c_str(), "state"); // <--- !!!
}
std::string WebServer::switch_json(switch_::Switch *obj, bool value) {
  return json::build_json([obj, value](JsonObject &root) {
    root["id"] = "switch-" + obj->get_object_id();
    root["state"] = value ? "ON" : "OFF";
    root["value"] = value;
  });
}
```
`c_str()` is returning pointer to string data that can be already deleted (by std::string destructor) when `AsyncEventSourceClient` is trying to send it?

vazir, 2023-02-08:
It seems totally unrelated. It is enough to have a few switches and a few sensors and try trigger switches over MQTT intensively. That will happen. 
[10:35:58][D][switch:056]: 'water-controller-test-switch sw 4': Sending state ON
[10:35:58]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[10:35:58]
[10:35:58]Core  0 register dump:
[10:35:58]PC      : 0x400df993  PS      : 0x00060630  A0      : 0x800dfa11  A1      : 0x3ffd8530  
WARNING Decoded 0x400df993: esphome::Scheduler::cancel_item_(esphome::Component*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, esphome::Scheduler::SchedulerItem::Type) at /config/.esphome/build/water-controller-test-switch/src/esphome/core/scheduler.cpp:266 (discriminator 2)
[10:35:58]A2      : 0x3ffc2e08  A3      : 0x3ffaf34c  A4      : 0x3ffd8604  A5      : 0x00000000  
[10:35:58]A6      : 0x00000000  A7      : 0x3ffb4b10  A8      : 0x00000000  A9      : 0x3ffb4b4c  
[10:35:59]A10     : 0x00018c2e  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00009b00  
[10:35:59]A14     : 0x00646e65  A15     : 0x00646e65  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[10:35:59]EXCVADDR: 0x00000000  LBEG    : 0x40084651  LEND    : 0x40084659  LCOUNT  : 0x00000027  
[10:35:59]
[10:35:59]
[10:35:59]Backtrace:0x400df990:0x3ffd85300x400dfa0e:0x3ffd8560 0x400dfb8a:0x3ffd8580 0x400df429:0x3ffd85b0 0x400d7e64:0x3ffd85f0 0x400d9e65:0x3ffd8640 0x400d47bd:0x3ffd8670 0x400d9cc1:0x3ffd86c0 0x400d7d48:0x3ffd86e0 0x400d6082:0x3ffd8720 0x400d60ee:0x3ffd8740 0x400d5926:0x3ffd8790 0x400ea1e1:0x3ffd87d0 0x4016bf1a:0x3ffd8840 0x400eaf76:0x3ffd8870 0x400e9f66:0x3ffd88c0 0x400e9f7e:0x3ffd8930 0x40159399:0x3ffd8950 0x40159411:0x3ffd8980 0x40159bae:0x3ffd89a0