esphome/issues #2193: Logs button not working

goofy2k, 2021-06-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome V1.19.4 under Docker on a Synology NAS

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 NodeMCU and ESP8266 NodeMCU

**ESPHome version (latest production, beta, dev branch)**

V1.19.4, Docker image

**Affected component:**

All

**Description of problem:**
No response whatsoever after pressing Logs
For all nodes.
All other actions work fine.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
After a firmware update with Install >> WiFi  log info appears in the Install screen as soon as the device has connected successfully.

MarcHagen, 2021-06-27:
See: #2183

balloob, 2021-06-28:
Closing because it's a duplicate.

closed 2021-06-28 by balloob