esphome/issues #2198: Sonoff basic not working from 1.19 and forward

j-k-olofsson, 2021-06-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Debian 10.10

**ESP (ESP32/ESP8266, Board/Sonoff):**
Sonoff

**ESPHome version (latest production, beta, dev branch)**
ESPHome v1.19,x

**Affected component:**

**Description of problem:**
After updating from 1.18.0 to 1.19.0 (and following releases) my Sonoff Basic devices are not able to start correctly, kepps rebooting and turning relay on and off.

**Problem-relevant YAML-configuration entries:**
substitutions:
  node_name: esphome_02
  log_level: DEBUG
esphome:
  name: esphome_02
  comment: Sonoff Basic generic node - Growlight
  platform: ESP8266
  board: esp8285
  arduino_version: platformio/espressif8266@1.8.0
  build_path: builds/esphome_02
  platformio_options: {}
  esp8266_restore_from_flash: false
  board_flash_mode: dout
  includes: []
  libraries: []
  name_add_mac_suffix: false
wifi:
  fast_connect: true
  ap:
    ssid: esphome_02 FallbackAP
    password: !secret 'MQTT_PWD'
    ap_timeout: 1min
  enable_mdns: true
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  output_power: 20.0
  networks:
  - ssid: !secret 'WIFI_SSID'
    password: !secret 'MQTT_PWD'
    priority: 0.0
  use_address: esphome_02.local
captive_portal: {}
logger:
  level: DEBUG
  logs:
    mqtt.component: DEBUG
    mqtt.client: ERROR
  baud_rate: 115200
  tx_buffer_size: 512
  hardware_uart: UART0
  esp8266_store_log_strings_in_flash: true
web_server:
  port: 80
  css_url: https://esphome.io/_static/webserver-v1.min.css
  js_url: https://esphome.io/_static/webserver-v1.min.js
ota:
  password: !secret 'OTA_PWD'
  safe_mode: true
  port: 8266
  reboot_timeout: 5min
  num_attempts: 10
mqtt:
  broker: !secret 'MQTT_IP'
  username: !secret 'MQTT_USER'
  password: !secret 'MQTT_PWD'
  topic_prefix: homeassistant
  log_topic:
    topic: homeassistant/node/esphome_02/log/state
    qos: 0
    retain: true
  birth_message:
    topic: homeassistant/node/esphome_02/availability/state
    payload: online
    qos: 0
    retain: true
  will_message:
    topic: homeassistant/node/esphome_02/availability/state
    payload: offline
    qos: 0
    retain: true
  on_message:
  - topic: homeassistant/node/esphome_02/cmd/state
    then:
    - logger.log:
        format: esphome_02 is alive!
        args: []
        level: DEBUG
        tag: main
    qos: 0
  port: 1883
  discovery: true
  discovery_retain: true
  discovery_prefix: homeassistant
  keepalive: 15s
  reboot_timeout: 15min
  shutdown_message:
    topic: homeassistant/status
    payload: offline
    qos: 0
    retain: true
status_led:
  pin:
    number: 13
    inverted: true
    mode: OUTPUT
switch:
- platform: gpio
  name: esphome_02 Relay state
  pin:
    number: 12
    mode: OUTPUT
    inverted: false
  id: relay
  restore_mode: RESTORE_DEFAULT_OFF
  interlock_wait_time: 0ms
- platform: restart
  id: my_reset
  name: esphome_02 Reboot
  icon: mdi:restart
binary_sensor:
- platform: gpio
  id: device_button
  pin:
    number: 0
    mode: INPUT_PULLUP
    inverted: true
  on_press:
  - then:
    - switch.toggle:
        id: relay
  on_double_click:
  - min_length: 50ms
    max_length: 350ms
    then:
    - switch.turn_on:
        id: my_reset
  name: device_button
  internal: true

**Logs (if applicable):**
Using 'COM5' as serial port.
Showing logs:
[13:27:47]+ښjmumݥ344]:   SSID: 'OLOFSSON_WIFI'
[13:27:47][C][wifi:345]:   IP Address: 192.168.0.227
[13:27:47][C][wifi:347]:   BSSID: 2C:FD:A1:13:56:78
[13:27:47][C][wifi:348]:   Hostname: 'esphome_test_basic'
[13:27:47][C][wifi:352]:   Signal strength: -45 dB ▂▄▆█
[13:27:47][C][wifi:356]:   Channel: 11
[13:27:47][C][wifi:357]:   Subnet: 255.255.255.0
[13:27:47][C][wifi:358]:   Gateway: 192.168.0.1
[13:27:47][C][wifi:359]:   DNS1: 192.168.0.2
[13:27:47][C][wifi:360]:   DNS2: 192.168.0.1
[13:27:47][D][wifi:507]: Disabling AP...
[13:27:47]
[13:27:47]Exception (0):
[13:27:47]epc1=0xa8c09222 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[13:27:47]
[13:27:47]ctx: cont 
[13:27:47]sp: 3ffff9c0 end: 3fffffd0 offset: 01a0
[13:27:47]
[13:27:47]>>>stack>>>
[13:27:47]3ffffb60:  00000000 3ffe8304 3ffffc10 40225954  
[13:27:47]3ffffb70:  00000004 ffffffff ffffffff 00000003  
[13:27:47]3ffffb80:  00000003 00000016 00302075 ffff0208  
[13:27:47]3ffffb90:  3ffffd20 3fff161c 00000014 0000000b  
[13:27:47]3ffffba0:  3fff1~~~~~~~~~3ffe8304 3ffffc60 40225954  
[13:27:47]3ffffbc0:  3fff2c32 0000002e 00000000 402234ed  
[13:27:47]3ffffbd0:  3ffffd00 0000000f 00302018 3fff0ce1  
[13:27:47]3ffffbe0:  3fff0cac 3fff0cac 3fff2bec 40227ef6  
[13:27:47]3ffffbf0:  4022828b 3fff2bec 3fff2c32 40228294  
[13:27:47]3ffffc00:  00000008 3ffffc30 000001f0 402234ed  
[13:27:47]3ffffc10:  3fff1642 0000002d 000001d9 3fff2c40  
[13:27:47]3ffffc20:  3fff2bec 3fff0f88 3fff0cac 4022e4e1  
[13:27:47]3ffffc30:  005e0001 0000fb00 00000008 4021c719  
[13:27:47]3ffffc40:  00000018 3fff0f88 3fff2bec 4022ef68  
[13:27:47]3ffffc50:  00000005 3ffe8304 000001da 00000000  
[13:27:47]3ffffc60:  3fff0cb0 00000001 00000000 00000002  
[13:27:47]3ffffc70:  3fff0c34 00000004 3fff0f84 4022f5d4  
[13:27:47]3ffffc80:  3ffffd20 000001f0 3fff15dc 3fff0f84  
[13:27:47]3ffffc90:  00000000 3fff2bec 3fff2c58 4022efa8  
[13:27:47]3ffffca0:  3fff0cac 3ffffcd0 00000004 3fff0f84  
[13:27:47]3ffffcb0:  00000016 3fff0cac 3fff0f84 4022e739  
[13:27:47]3ffffcc0:  3fff0cac 3ffffcd0 00000004 402282e8  
[13:27:47]3ffffcd0:  00000494 00000000 3fff0e04 4022839b  
[13:27:47]3ffffd40:  00000000 3fff05fc 3ffffd90 401006dc  
[13:27:47]3ffffd50:  3fff085c 000000df 00000000 4021da7c  
[13:27:47]3ffffd60:  00000000 4021da24 3fff06dc fb0000e0  
[13:27:47]3ffffd70:  00000000 3fff05fc 3fff06b0 00000000  
[13:27:47]3ffffd80:  00000000 3fff05fc 3fff06b0 4021982a  
[13:27:47]3ffffd90:  3fff1464 3fff1a8c 4021805c 402184bc  
[13:27:47]3ffffda0:  00000000 00000000 00000003 0000002c  
[13:27:47]3ffffdb0:  3ffe9067 00000003 3fff06b0 4020f586  
[13:27:47]3ffffdc0:  3ffe9067 00000003 3fff1a8c 4020acee  
[13:27:47]3ffffdd0:  40220ca8 e300a8c0 00000000 00000000  
[13:27:47]3ffffde0:  00000000 00000000 40106a32 36872b02  
[13:27:47]3ffffdf0:  00000000 3fff0f44 3fff24e4 00000100  
[13:27:47]3ffffe00:  3fff1a8c 00000f51 3fff1a8c 4020bfb0  
[13:27:47]3ffffe10:  00000f51 00000006 3fff04c4 4021f218  
[13:27:47]3ffffe20:  3fff0f44 00000006 3fff04c4 4021f29d  
[13:27:47]3ffffe30:  3fff1a8c 00000006 3fff04c4 4020df50  
[13:27:47]3ffffe40:  3fff0f44 00000007 00000000 40227876  
[13:27:47]3ffffe50:  00000100 00000000 00000000 4022722a  
[13:27:47]3ffffe60:  3fffff20 3fff04c4 00000000 3fff282c  
[13:27:47]3ffffe70:  3fff0598 3fff04c4 00000000 40211022  
[13:27:47]3ffffe80:  3fff1c68 00000000 00000000 3fff1c98  
[13:27:47]3ffffe90:  00000000 00000000 00000000 40220ca8  
[13:27:47]3ffffea0:  00000000 40220ca8 00000000 40220ca8  
[13:27:47]3ffffeb0:  00000000 40220ca8 00000000 40220ca8  
[13:27:47]3ffffec0:  00000000 00000000 3fff1bd8 00000000  
[13:27:47]3ffffed0:  00000000 3fff1bf8 00000000 00000000  
[13:27:47]3ffffee0:  00000000 40220ca8 00000000 40220ca8  
[13:27:47]3ffffef0:  00000000 40220ca8 00000000 40220ca8  
[13:27:47]3fffff00:  00000000 40220ca8 00000000 00000000  
[13:27:47]3fffff10:  3fff2bc0 3fff2b98 3fff2b98 00000000  
[13:27:47]3fffff20:  3fff299c 3fff2a5c 3fff2b24 3fffff20  
[13:27:47]3fffff30:  00000003 feefeffe feefeffe feefeffe  
[13:27:47]3fffff40:  feefeffe feefeffe feefeffe feefeffe  
[13:27:47]3fffff50:  feefeffe feefeffe feefeffe feefeffe  
[13:27:47]3fffff60:  feefeffe 3fffff20 00000001 3fffff20  
[13:27:47]3fffff70:  00000001 3fffff20 00000001 3fff282c  
[13:27:47]3fffff80:  3fff26ec 3fff2574 3fff246c feefeffe  
[13:27:47]3fffff90:  3fff2b24 3fff2a3c 3fff05c8 00000002  
[13:27:47]3fffffa0:  feefeffe feefeffe feefeffe 3fff07d8  
[13:27:47]3fffffb0:  3fffdad0 00000000 3fff07d1 4021dde8  
[13:27:47]3fffffc0:  feefeffe feefeffe 3ffe857c 40100721  
[13:27:47]<<<stack<<<
[13:27:47]
[13:27:47] ets Jan  8 2013,rst cause:1, boot mode:(3,0)
[13:27:47]
[13:27:47]load 0x4010f000, len 1384, room 16 
[13:27:47]tail 8
[13:27:47]chksum 0x2d
[13:27:47]csum 0x2d
[13:27:47]vbb28d4a3
[13:27:47]~ld

**Additional information and things you've tried:**
Also tested v1.19.1, 1.19.2, 1.19.3 and 1.19.4 with the same results. Downgrading to 1.18.0 solves the problem.

j-k-olofsson, 2021-06-30:
Same problem as #2205

j-k-olofsson, 2021-07-16:
STill the same problem after upgrading to esphome-1.20.0b2

KC-inDomus, 2021-07-19:
Same problem with a Sonoff Basic R1 board, no problem with a R3 board.

jesserockz, 2021-07-19:
I pulled out one of my old sonoff basics yesterday and it worked fine on the latest dev branch...
@j-k-olofsson Could you get a serial log from the machine that compiled esphome - `esphome run your-config.yaml` as this will attempt to decode the stacktrace, giving more information

j-k-olofsson, 2021-07-20:
As requested, this is the output from "esphome run esphome_02.yaml" using version 1.19.4

INFO Reading configuration esphome_02.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting builds/esphome_02/.pioenvs
INFO Deleting builds/esphome_02/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d builds/esphome_02
Processing esphome_02 (board: esp8285; framework: arduino; platform: platformio/espressif8266@1.8.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 423.98KB Flash
PACKAGES:
 - framework-arduinoespressif8266 2.20402.4 (2.4.2)
 - tool-esptool 1.413.0 (4.13)
 - toolchain-xtensa 1.40802.0 (4.8.2)
Library Manager: Installing ESPAsyncTCP-esphome @ 1.2.3
Library Manager: ESPAsyncTCP-esphome @ 1.2.3 has been installed!
Library Manager: Installing esphome/AsyncTCP-esphome
Library Manager: AsyncTCP-esphome @ 1.2.2 has been installed!
Library Manager: Installing ESPAsyncWebServer-esphome @ 1.2.7
Library Manager: ESPAsyncWebServer-esphome @ 1.2.7 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Library Manager: Installing AsyncMqttClient-esphome @ 0.8.4
Library Manager: AsyncMqttClient-esphome @ 0.8.4 has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing ArduinoJson-esphomelib @ 5.13.3
Library Manager: ArduinoJson-esphomelib @ 5.13.3 has been installed!
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS>
|   |-- <ESP8266WiFi> 1.0
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.0
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/esphome_02/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/json/json_util.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/logger/logger.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/custom_mqtt_device.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_binary_sensor.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_client.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_climate.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_component.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_cover.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_fan.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_light.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_sensor.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_switch.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt/mqtt_text_sensor.cpp.o
Compiling .pioenvs/esphome_02/src/esphome/components/mqtt_subscribe/text_sensor/mqtt_subscribe_text_sensor.cpp.o
[... 122 lines cut ...]
Compiling .pioenvs/esphome_02/FrameworkArduino/libb64/cdecode.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/libb64/cencode.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/libc_replacements.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/pgmspace.cpp.o
Compiling .pioenvs/esphome_02/FrameworkArduino/sntp-lwip2.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/spiffs/spiffs_cache.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/spiffs/spiffs_check.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/spiffs/spiffs_gc.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/spiffs/spiffs_hydrogen.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/spiffs/spiffs_nucleus.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/spiffs_api.cpp.o
Compiling .pioenvs/esphome_02/FrameworkArduino/spiffs_hal.cpp.o
Compiling .pioenvs/esphome_02/FrameworkArduino/time.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/uart.c.o
Compiling .pioenvs/esphome_02/FrameworkArduino/umm_malloc/umm_malloc.c.o
Archiving .pioenvs/esphome_02/libFrameworkArduino.a
Indexing .pioenvs/esphome_02/libFrameworkArduino.a
Linking .pioenvs/esphome_02/firmware.elf
Retrieving maximum program size .pioenvs/esphome_02/firmware.elf
Checking size .pioenvs/esphome_02/firmware.elf
Building .pioenvs/esphome_02/firmware.bin
RAM:   [====      ]  44.1% (used 36112 bytes from 81920 bytes)
Flash: [====      ]  38.8% (used 397572 bytes from 1023984 bytes)
===================================================================================== [SUCCESS] Took 17.96 seconds =====================================================================================
INFO Successfully compiled program.

HoLy1337, 2021-07-22:
Hello There,

I have the same problem with my Sonoff basic s20...

I tried to simplify my config to find out what is cousing this problem...

but even with this config it is not working since the update (ESPHome v1.19,x) still not fixed in 1.20.0....

esphome:
  name: test
  platform: ESP8266
  board: esp8285
  arduino_version: 2.4.2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

majdzik84, 2021-07-22:
I confirm, it still does not work in the latest version.

HoLy1337, 2021-07-27:
1.20.1 still the same problem...

Here the Log from flashing with an exception: 0 (IllegalInstructionCause)
Can someone decode the Stack?

[12:48:47][I][logger:170]: Log initialized
[12:48:47][C][ota:378]: There have been 3 suspected unsuccessful boot attempts.
[12:48:47][I][app:029]: Running through setup()...
[12:48:47][D][binary_sensor:034]: 'btn': Sending initial state OFF
[12:48:47][C][light:031]: Setting up light 'Rechte Lampe'...
[12:48:47][D][light:093]: 'Rechte Lampe' Setting:
[12:48:47][D][light:102]:   State: ON
[12:48:47][C][wifi:037]: Setting up WiFi...
[12:48:47][I][wifi:250]: WiFi Connecting to 'WLAN-DFFC12'...
[12:48:52][I][wifi:513]: WiFi Connected!
[12:48:52][C][wifi:359]:   SSID: 'WLAN-DFFC12'
[12:48:52][C][wifi:360]:   IP Address: 192.168.2.80
[12:48:52][C][wifi:362]:   BSSID: C0:25:06:xx:xx:xx
[12:48:52][C][wifi:363]:   Hostname: 'lampe_rechts'
[12:48:52][C][wifi:367]:   Signal strength: -37 dB ▂▄▆█
[12:48:52][C][wifi:371]:   Channel: 1
[12:48:52][C][wifi:372]:   Subnet: 255.255.255.0
[12:48:52][C][wifi:373]:   Gateway: 192.168.2.1
[12:48:52][C][wifi:374]:   DNS1: 192.168.2.1
[12:48:52][C][wifi:375]:   DNS2: 0.0.0.0
[12:48:52][D][wifi:522]: Disabling AP...
[12:48:52]
[12:48:52]Exception (0):
[12:48:52]epc1=0xa8c0b79e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[12:48:52]
[12:48:52]ctx: cont 
[12:48:52]sp: 3ffff9c0 end: 3fffffd0 offset: 01a0
[12:48:52]
[12:48:52]>>>stack>>>
[12:48:52]3ffffb60:  3ffffbd3 3ffea242 00302073 00000003  
[12:48:52]3ffffb70:  3ffea245 00000003 3ffffba0 3ffea242  
[12:48:52]3ffffb80:  00000000 3ffe8304 3ffffc30 40226678  
[12:48:52]3ffffb90:  00000004 ffffffff ffffffff 00000003  
[12:48:52]3ffffba0:  00000003 00000016 00302075 ffff0208  
[12:48:52]3ffffbb0:  3ffffd40 3fff2884 00000014 00000007  
[12:48:52]3ffffbc0:  3fff351a 0000002e 3fff0ab0 3fff2884  
[12:48:52]3ffffbd0:  00000000 3ffe8304 3ffffc80 3fff1f51  
[12:48:52]3ffffbe0:  3fff1f1c 3fff1f1c 3fff34d4 40228b2a  
[12:48:52]3ffffbf0:  40228ebf 3fff34d4 3fff351a 40228ec8  
[12:48:52]3ffffc00:  00000008 3ffffc30 3f302075 00000012  
[12:48:52]3ffffc10:  0000000d 000001f5 3fff2844 3fff3528  
[12:48:52]3ffffc20:  3fff34d4 3fff34c0 3fff1f1c 4022f119  
[12:48:52]3ffffc30:  005e0001 0000fb00 000001d9 ffff0208  
[12:48:52]3ffffc40:  00000018 3fff34c0 3fff34d4 4022fba0  
[12:48:52]3ffffc50:  3fff2893 0000000f 00000008 00000000  
[12:48:52]3ffffc60:  3fff1f20 00000001 00000000 00000002  
[12:48:52]3ffffc70:  3fff1ea4 00000004 3fff34bc 4023020c  
[12:48:52]3ffffc80:  3fff28b9 3fff288f 000001ca 3fff34bc  
[12:48:52]3ffffc90:  00000000 3fff34d4 3fff3540 4022fbe0  
[12:48:52]3ffffca0:  3fff1f1c 3ffffcd0 00000004 3fff34bc  
[12:48:52]3ffffcb0:  00000016 3fff1f1c 3fff34bc 4022f371  
[12:48:52]3ffffcc0:  3fff1f1c 3ffffcd0 00000004 40228f1c  
[12:48:52]3ffffcd0:  00000494 00000000 3fff2074 40228f58  
[12:48:52]3ffffce0:  3fff34c0 3fff1f1c 3fff21bc 4022f442  
[12:48:52]3ffffcf0:  3fff219c 4021eb70 00000010 00000020  
[12:48:52]3ffffd00:  3ffffd6c 3fff1f1c 3fff34bc 4022f61b  
[12:48:52]3ffffd10:  3ffffd6c 3ffebc04 3fff1f1c 4022f6a7  
[12:48:52]3ffffd20:  3ffea4bd 00000001 3ffffda0 00000001  
[12:48:52]3ffffd30:  3fff1920 3fff1870 3fff1920 4021b933  
[12:48:52]3ffffd40:  00000000 3fff1870 3ffffd90 401006dc  
[12:48:52]3ffffd50:  3fff1acc 0000004d 00000000 4021ebc8  
[12:48:52]3ffffd60:  00000000 4021eb70 3fff194c fb0000e0  
[12:48:52]3ffffd70:  00000000 3fff1870 3fff1920 00000000  
[12:48:52]3ffffd80:  00000000 3fff1870 3fff1920 4021cdc2  
[12:48:52]3ffffd90:  3fff2244 5a71446e 4021b5f4 4021ba54  
[... 133 lines cut ...]
[12:48:56]3fffff10:  00000000 402218c0 00000000 402218c0  
[12:48:56]3fffff20:  00000000 402218c0 00000000 00000000  
[12:48:56]3fffff30:  3fff3354 00000003 00000000 006e7462  
[12:48:56]3fffff40:  0100000a 402218c0 fcffffff 402218c0  
[12:48:56]3fffff50:  00000000 402218c0 00000000 3fff32dc  
[12:48:56]3fffff60:  0100000a 402218c0 0100000a 402218c0  
[12:48:56]3fffff70:  fcffffff 402218c0 00000000 402218c0  
[12:48:56]3fffff80:  00000000 3fff318c feefeffe feefeffe  
[12:48:56]3fffff90:  3ffffee8 3fff32dc 3fff2f94 00000000  
[12:48:56]3fffffa0:  feefeffe feefeffe feefeffe 3fff1a48  
[12:48:56]3fffffb0:  3fffdad0 00000000 3fff1a41 4021ef34  
[12:48:56]3fffffc0:  feefeffe feefeffe 3ffe8548 40100721  
[12:48:56]<<<stack<<<
[12:48:56]
[12:48:56] ets Jan  8 2013,rst cause:1, boot mode:(3,6)
[12:48:56]
[12:48:56]load 0x4010f000, len 1384, room 16 
[12:48:56]tail 8
[12:48:56]chksum 0x2d
[12:48:56]csum 0x2d
[12:48:56]vbb28d4a3
[12:48:56]~ld
[12:48:56][I][logger:170]: Log initialized
[12:48:56][C][ota:378]: There have been 5 suspected unsuccessful boot attempts.
[12:48:56][I][app:029]: Running through setup()...

oxan, 2021-08-09:
> Here the Log from flashing with an exception: 0 (IllegalInstructionCause)
> Can someone decode the Stack?

Unfortunately we can't do that without your `firmware.elf`. If you retrieve the logs during `esphome logs`, stack decoding should be done automatically though.

j-k-olofsson, 2021-08-10:
@oxan sorry for the missing log stack. I had to change device but this is also a Sonoff basic with exactly the same problem and this is the log:

[09:29:29][I][logger:170]: Log initialized
[09:29:29][C][ota:378]: There have been 8 suspected unsuccessful boot attempts.
[09:29:29][I][app:029]: Running through setup()...
[09:29:29][C][switch.gpio:011]: Setting up GPIO Switch 'esphome_01_test_ver Relay state'...
[09:29:29][D][switch:025]: 'esphome_01_test_ver Relay state' Turning OFF.
[09:29:29][D][switch:045]: 'esphome_01_test_ver Relay state': Sending state OFF
[09:29:29][D][switch:025]: 'esphome_01_test_ver Relay state' Turning OFF.
[09:29:29][D][binary_sensor:034]: 'device_button': Sending initial state OFF
[09:29:29][C][wifi:037]: Setting up WiFi...
[09:29:29][I][wifi:250]: WiFi Connecting to 'OLOFSSON_WIFI'...
[09:29:33][I][wifi:513]: WiFi Connected!
[09:29:33][C][wifi:359]:   SSID: 'OLOFSSON_WIFI'
[09:29:33][C][wifi:360]:   IP Address: 192.168.0.227
[09:29:33][C][wifi:362]:   BSSID: 2C:FD:A1:13:65:C8
[09:29:33][C][wifi:363]:   Hostname: 'esphome_01_test_ver'
[09:29:33][C][wifi:367]:   Signal strength: -58 dB ▂▄▆█
[09:29:33][C][wifi:371]:   Channel: 11
[09:29:33][C][wifi:372]:   Subnet: 255.255.255.0
[09:29:33][C][wifi:373]:   Gateway: 192.168.0.1
[09:29:33][C][wifi:374]:   DNS1: 192.168.0.2
[09:29:33][C][wifi:375]:   DNS2: 192.168.0.1
[09:29:33][D][wifi:522]: Disabling AP...
[09:29:33]
[09:29:33]Exception (0):
WARNING Exception type: Illegal instruction (Is the flash damaged?)
[09:29:33]epc1=0xa8c09222 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[09:29:33]
[09:29:33]ctx: cont
[09:29:33]sp: 3ffff9c0 end: 3fffffd0 offset: 01a0
[09:29:33]
[09:29:33]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[09:29:33]3ffffb60:  00000000 3ffe8304 3ffffc10 40225a38
WARNING Decoded 0x40225a38: _svfprintf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c:531
[09:29:33]3ffffb70:  00000004 ffffffff ffffffff 00000003
[09:29:33]3ffffb80:  00000003 00000016 00302075 ffff0208
[09:29:33]3ffffb90:  3ffffd20 3fff161c 00000014 0000000b
[09:29:33]3ffffba0:  3fff1625 00000008 3ffef7f0 3fff161c
[09:29:33]3ffffbb0:  00000000 3ffe8304 3ffffc60 40225a38
WARNING Decoded 0x40225a38: _svfprintf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c:531
[09:29:33]3ffffbc0:  3fff2c42 0000002e 00000000 402235d1
WARNING Decoded 0x402235d1: _vsnprintf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:73
[09:29:33]3ffffbd0:  3ffffd00 0000000f 00302018 3fff0ce1
[09:29:33]3ffffbe0:  3fff0cac 3fff0cac 3fff2bfc 40227fda
WARNING Decoded 0x40227fda: new_linkoutput at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:259
[09:29:33]3ffffbf0:  4022836f 3fff2bfc 3fff2c42 40228378
WARNING Decoded 0x4022836f: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:305
WARNING Decoded 0x40228378: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:305
[09:29:33]3ffffc00:  00000008 3ffffc30 000001f0 402235d1
WARNING Decoded 0x402235d1: _vsnprintf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:73
[09:29:33]3ffffc10:  3fff1642 0000002d 000001d9 3fff2c50
[09:29:33]3ffffc20:  3fff2bfc 3fff0f88 3fff0cac 4022e5c5
WARNING Decoded 0x4022e5c5: etharp_output_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c:893
[09:29:33]3ffffc30:  005e0001 0000fb00 00000008 4021c6bd
WARNING Decoded 0x4021c6bd: Print::write(char const*)
[09:29:33]3ffffc40:  00000018 3fff0f88 3fff2bfc 4022f04c
WARNING Decoded 0x4022f04c: ip4_output_if_opt_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:962
[09:29:33]3ffffc50:  00000005 3ffe8304 000001da 00000000
[09:29:33]3ffffc60:  3fff0cb0 00000001 00000000 00000002
[09:29:33]3ffffc70:  3fff0c34 00000004 3fff0f84 4022f6b8
WARNING Decoded 0x4022f6b8: mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:136
[09:29:33]3ffffc80:  3ffffd20 000001f0 3fff15dc 3fff0f84
[09:29:33]3ffffc90:  00000000 3fff2bfc 3fff2c68 4022f08c
WARNING Decoded 0x4022f08c: ip4_output_if_opt at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:788
[09:29:33]3ffffca0:  3fff0cac 3ffffcd0 00000004 3fff0f84
[09:29:33]3ffffcb0:  00000016 3fff0cac 3fff0f84 4022e81d
WARNING Decoded 0x4022e81d: igmp_send at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
[09:29:33]3ffffcc0:  3fff0cac 3ffffcd0 00000004 402283cc
[... 72 lines cut ...]
[09:29:34]3fffff10:  3fff2bd0 3fff2ba8 3fff2ba8 00000000
[09:29:34]3fffff20:  3fff29ac 3fff2a6c 3fff2b34 3fffff20
[09:29:34]3fffff30:  00000003 feefeffe feefeffe feefeffe
[09:29:34]3fffff40:  feefeffe feefeffe feefeffe feefeffe
[09:29:34]3fffff50:  feefeffe feefeffe feefeffe feefeffe
[09:29:34]3fffff60:  feefeffe 3fffff20 00000001 3fffff20
[09:29:34]3fffff70:  00000001 3fffff20 00000001 3fff283c
[... cut ...]
[09:29:34]3fffffa0:  feefeffe feefeffe feefeffe 3fff07d8
[09:29:34]3fffffb0:  3fffdad0 00000000 3fff07d1 4021dd8c
WARNING Decoded 0x4021dd8c: loop_wrapper() at core_esp8266_main.cpp
[09:29:34]3fffffc0:  feefeffe feefeffe 3ffe8548 40100721
WARNING Decoded 0x40100721: cont_wrapper
[09:29:34]<<<stack<<<
[09:29:34]
[09:29:34] ets Jan  8 2013,rst cause:4, boot mode:(3,0)
[09:29:34]
[09:29:34]wdt reset
[09:29:34]load 0x4010f000, len 1384, room 16
[09:29:34]tail 8
[09:29:34]chksum 0x2d
[09:29:34]csum 0x2d
[09:29:34]vbb28d4a3
[09:29:34]~ld

oxan, 2021-08-10:
I'll close this in favor of #2309 now. Please follow the instructions there if you have this problem.

Especially try removing the `arduino_version` option, as you all seem to be using ancient versions.

closed 2021-08-10 by oxan