esphome/issues #2200: Android HA Companion app BLE UUID is not detected with esp32_ble_tracker

fokcuk, 2021-06-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome Hass,io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 - esp-wrover-kit

**ESPHome version (latest production, beta, dev branch)**

1.19.4
**Affected component:**

esp32_ble_tracker

**Description of problem:**
UUID of the HA companion app is received in reverse order and not matched to the UUID filters.
Eg: 
UUID sent by HA Companion app:    b722b840-712c-4077-9cb7-0aee0e069000
iBeacon UUID registered in ESPHome:   0090060EEE0AB7-9C77-402C-7140-B822B7

**Problem-relevant YAML-configuration entries:**
```
esp32_ble_tracker:
  on_ble_advertise:
      then:
        - lambda: |-
            ESP_LOGD("ble_adv", "New BLE device");
            ESP_LOGD("ble_adv", "  address: %s", x.address_str().c_str());
            ESP_LOGD("ble_adv", "  name: %s", x.get_name().c_str());
            ESP_LOGD("ble_adv", "  RSSI: %i", x.get_rssi());
            ESP_LOGD("ble_adv", "  Advertised service UUIDs:");
            for (auto uuid : x.get_service_uuids()) {
                ESP_LOGD("ble_adv", "    - %s", uuid.to_string().c_str());
            }
            ESP_LOGD("ble_adv", "  Advertised service data:");
            for (auto data : x.get_service_datas()) {
                ESP_LOGD("ble_adv", "    - %s: (length %i)", data.uuid.to_string().c_str(), data.data.size());
            }
            ESP_LOGD("ble_adv", "  Advertised manufacturer data:");
            for (auto data : x.get_manufacturer_datas()) {
                ESP_LOGD("ble_adv", "    - %s: (length %i)", data.uuid.to_string().c_str(), data.data.size());
            }
        #registers reverse order of UUID
        
            if(x.get_ibeacon().has_value()) {
              auto ibeacon = x.get_ibeacon().value();
              ESP_LOGD("ble_adv", ibeacon.get_uuid().to_string().c_str());
              ESP_LOGD("ble_adv", "Major: %i", ibeacon.get_major());
              ESP_LOGD("ble_adv", "Minor: %i", ibeacon.get_minor());
            }
            
```

**Logs (if applicable):**

```
[10:57:43][D][ble_adv:029]: New BLE device
[10:57:43][D][ble_adv:030]:   address: 6B:28:44:21:BA:82
[10:57:43][D][ble_adv:031]:   name: 
[10:57:43][D][ble_adv:032]:   RSSI: -83
[10:57:43][D][ble_adv:033]:   Advertised service UUIDs:
[10:57:43][D][ble_adv:037]:   Advertised service data:
[10:57:43][D][ble_adv:041]:   Advertised manufacturer data:
[10:57:43][D][ble_adv:043]:     - 0x004C: (length 23)
[10:57:43][D][ble_adv:047]: 0090060EEE0AB7-9C77-402C-7140-B822B7
[10:57:43][D][ble_adv:048]: Major: 100
[10:57:43][D][ble_adv:049]: Minor: 1

```

**Additional information and things you've tried:**

kmdm, 2021-07-12:
Seems related to https://github.com/esphome/esphome/commit/a70a205ace089c30c0ffcf38b9793d7181595b04

kmdm, 2021-07-12:
FWIW, I currently fix this by re-implementing to_string() for LEN_128 and changing the loop to increment: 

```
            if (x.get_ibeacon().has_value()) {
                std::string uuid;
                esp_bt_uuid_t raw_uuid = x.get_ibeacon().value().get_uuid().get_uuid();

                char sbuf[64];
                char *bpos = sbuf;

                switch (raw_uuid.len) {
                    case ESP_UUID_LEN_128:
                        for (int8_t i = 0; i <= 15; i++) {
                            sprintf(bpos, "%02x", raw_uuid.uuid.uuid128[i]);
                            bpos += 2;

                            if (i == 6 || i == 8 || i == 10 || i == 12)
                                  sprintf(bpos++, "-");
                        }

                        sbuf[47] = '\0';
                        uuid.assign(sbuf);
                        break;
                    default:
                        uuid = x.get_ibeacon().value().get_uuid().to_string();
                        std::transform(uuid.begin(), uuid.end(), uuid.begin(), [](unsigned char c){ return std::tolower(c); });
                        break;
                }

```

I didn't check whether the other lengths are impacted (and I switched to lower case 'x' because I prefer lowercase for the UUID, similar to my hardware iBeacons). :-)

zhulik, 2021-07-21:
Any updates on this? I'd really appreciate if this was fixed

zhulik, 2021-11-19:
Bad bot, the problem is still there

litui, 2022-03-09:
Still an outstanding problem. Just encountered it here too.