esphome/issues #2202: ESPHome total daily energy sensor - count even if power sensor didn't change

krzysiek2788, 2021-06-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

core-2021.6.6
Home Assistant OS 6.0
supervisor-2021.06.6
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, Board LILYGO® TTGO T-Call V1.3 V1.4 ESP32
**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.19.4
**Affected component:**

When sensor which is set to be counted by total daily energy is stable (did not change) Total daily energy still changed please see screenshot's from homeassistant:
[https://esphome.io/components/sensor/total_daily_energy.html](https://esphome.io/components/sensor/total_daily_energy.html)
main sensors:
Import:
[image]

Export:
[image]

Daily energy sensor:
Import:
[image]
Export:
[image]

As you can see it counts if sensor is const (Import), also it counts wrongly (export) I have total exported energy about 466 kWh where Daily counts  2000+kWh. Which means it is not counting correctly.

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: home_lights_notify
  friendly_name: ttgo_home_lights_notify
  ip_address: !secret home_lights_notify

esphome:
  name: $devicename
  platform: ESP32
  board: esp-wrover-kit
  includes:
    - custom_components/sdm/SDM_Config_User.h
    - custom_components/sdm/SDM.h
    - custom_components/sdm/sdm_esphome.h

wifi:
  ssid: !secret wifi_id
  password: !secret wifi_pwd
  manual_ip:
    static_ip: ${ip_address}
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  #ap:
   # ssid: "Home Lights Notify"
    #password: "YiwdlGqxsCgP"

#captive_portal:

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0
  #level: VERBOSE

# Enable Home Assistant API
[... 346 lines cut ...]
    id: ${devicename}_Light_6
    name: ${devicename}_Light_6
    output: relay_6
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: binary
    id: ${devicename}_Light_7
    name: ${devicename}_Light_7
    output: relay_7
    restore_mode: RESTORE_DEFAULT_OFF
  - platform: binary
    id: ${devicename}_Light_8
    name: ${devicename}_Light_8
    output: relay_8
    restore_mode: RESTORE_DEFAULT_OFF

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2021-06-30:
No, that is correct.

`total_daily_energy` is for integrating _power_ data (with unit `W` or `kW`) over time to get _energy_ data with units `Wh`/`kWh`. You can think of it as summing up all values its base sensor puts out.

krzysiek2788, 2021-07-01:
Oh I thought it will work like Total daily in home Assistant. I thought I can create Total daily energy with kWh as well. Oh well maybe some day you will add this function as well.

closed 2021-07-01 by OttoWinter