esphome/issues #2204: Upgrading to latest version break my wifi devices connection.

trabogano, 2021-06-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

Esp32 heltec_wifi_kit_32

**ESPHome version (latest production, beta, dev branch)**

1.19.4

**Affected component:**

Wifi

**Description of problem:**

Today i have upgrade my esphome addon to 1.19.4 (stable)
The resulting firmware broke the wifi, my devices arent'able to connect via wifi anymore.

**Problem-relevant YAML-configuration entries:**
```yaml
wifi:
  ssid: "Mywifi"
  password: "MyPassword"

```

**Logs (if applicable):**

```
 Event: Disconnected ssid='Mywifi' bssid=[redacted] reason='4-Way Handshake Timeout'

```

**Additional information and things you've tried:**

It seems related wrong wifi wpa2 password but is correct same yaml file in a previous version of esphome work fine.

mkuoppa, 2021-06-30:
> **Operating environment/Installation (Hass.io/Docker/pip/etc.):**
>
> Hass.io
>
> **ESP (ESP32/ESP8266, Board/Sonoff):**
>
> Esp32 heltec_wifi_kit_32
>
> **ESPHome version (latest production, beta, dev [...]

I am experiencing similar issues with Sonoff devices (sonoff_s55 and sonoff_th16).
My base configuration is as follows:
`esphome:
  name: sonoff_th16
  platform: ESP8266
  board: esp8285
  arduino_version: 2.4.2
`

vincegre, 2021-07-01:
Have you found out the issue ? as I have a similar problem with some SONOFF ESP8266 devices that once flashed with latest esphome have no more Wifi :(

Jossie90, 2021-08-16:
I am having a very similar problem. I have so far upgraded two of my devices. At least one of them still connects to one of the accesspoints, but not the other. I have multiple (Ubiquiti) AP's with identical configuration.
Everything worked fine before upgrading.

randomsnack, 2021-08-20:
I got the esp32 device working by upgrading to the esphome beta. You can installed both. I stop the esphome addon and just use the beta until the next esphome upgrade. You will not lose any setting, but should backup your configuration, just to be safe.

https://esphome.io/guides/faq.html
How do I update to the latest beta release?

ESPHome has a beta release cycle so that new releases can easily be tested before the changes are deployed to the stable channel. You can help test ESPHome (and use new features) by installing the tested beta:

# For pip-based installs
pip3 install --pre -U esphome

# For docker-based installs
docker run [...] -it esphome/esphome:beta run livingroom.yaml

For Home Assistant supervised installs add the community addons beta repository by adding https://github.com/hassio-addons/repository-beta in Add-on store -> Repositories.

vincegre, 2021-08-20:
> Have you found out the issue ? as I have a similar problem with some SONOFF ESP8266 devices that once flashed with latest esphome have no more Wifi :(

Sorry for late answer, I have to try with newest release to check if problem still there ;)

trabogano, 2021-08-29:
Wifi now seems to work well in the latest stable version.

riddlecp, 2021-09-30:
Having the same issue.  Wifi working fine with a generic esp32, but not for the Heltec Wifi Kit 32