esphome/issues #2205: Sonoff basic keeps restarting after uploading 1.19.4 -  bootloop -

majdzik84, 2021-06-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home assistant - Addon.

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff basic

**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.19.4

**Affected component:**

**Description of problem:**

After the esphome uploaded to sonoff basic, it keeps restarting. I have uploaded many times with the latest espflasher but no changes.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: tuje
  platform: ESP8266
  board: esp8285
  arduino_version: 2.4.2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_pass
  use_address: 192.168.2.146
  manual_ip:
    static_ip: 192.168.2.146
    gateway: 192.168.2.1
    subnet: 255.255.255.0
    dns1: 192.168.2.1
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tuje lampa"
    password: !secret wifi_pass

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: gpio
    name: "Gniazdko basen"
    pin: GPIO12
    id: relay
    
  - platform: gpio
    name: "Lampa basen"
    pin: GPIO2
    id: relay2
    inverted: yes
    
status_led:
  pin:
    number: GPIO13
    inverted: yes

```

**Logs (if applicable):**

```
Using 'COM4' as serial port.
Showing logs:
[23:21:28][D][wifi:380]: Found networks:
[23:21:28][I][wifi:426]: - 'majdzik' (04:D4:C4:C1:80:E0) ▂▄▆█
[23:21:28][D][wifi:427]:     Channel: 3
[23:21:28][D][wifi:428]:     RSSI: -25 dB
[23:21:28][I][wifi:426]: - 'majdzik' (60:45:CB:59:B5:08) ▂▄▆█
[23:21:28][D][wifi:427]:     Channel: 11
[23:21:28][D][wifi:428]:     RSSI: -73 dB
[23:21:28][D][wifi:430]: - 'majdzik-gosc' (60:45:CB:59:B5:09) ▂▄▆█
[23:21:28][D][wifi:430]: - '' (00:1E:2A:61:68:E8) ▂▄▆█
[23:21:28][I][wifi:235]: WiFi Connecting to 'majdzik'...
[23:21:29][I][wifi:498]: WiFi Connected!
[23:21:29][C][wifi:344]:   SSID: 'majdzik'
[23:21:29][C][wifi:345]:   IP Address: 192.168.2.146
[23:21:29][C][wifi:347]:   BSSID: 04:D4:C4:C1:80:E0
[23:21:29][C][wifi:348]:   Hostname: 'tuje'
[23:21:29][C][wifi:352]:   Signal strength: -28 dB ▂▄▆█
[23:21:29][C][wifi:356]:   Channel: 3
[23:21:29][C][wifi:357]:   Subnet: 255.255.255.0
[23:21:29][C][wifi:358]:   Gateway: 192.168.2.1
[23:21:29][C][wifi:359]:   DNS1: 192.168.2.1
[23:21:29][C][wifi:360]:   DNS2: 0.0.0.0
[23:21:29][D][wifi:507]: Disabling AP...
[23:21:36]
[23:21:36] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[23:21:36]
[23:21:36]wdt reset
[23:21:36]load 0x4010f000, len 1384, room 16 
[23:21:36]tail 8
[23:21:36]chksum 0x2d
[23:21:36]csum 0x2d
[23:21:36]vbb28d4a3
[23:21:36]~ld
[23:21:37][I][logger:170]: Log initialized
[23:21:37][C][status_led:014]: Setting up Status LED...
[23:21:37][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[23:21:37][I][app:029]: Running through setup()...
[23:21:37][C][switch.gpio:011]: Setting up GPIO Switch 'Gniazdko basen'...
[23:21:37][D][switch:025]: 'Gniazdko basen' Turning OFF.
[23:21:37][D][switch:045]: 'Gniazdko basen': Sending state OFF
[23:21:37][D][switch:025]: 'Gniazdko basen' Turning OFF.
[23:21:37][C][switch.gpio:011]: Setting up GPIO Switch 'Lampa basen'...
[23:21:37][D][switch:025]: 'Lampa basen' Turning OFF.
[23:21:37][D][switch:045]: 'Lampa basen': Sending state ON
[23:21:37][D][switch:025]: 'Lampa basen' Turning OFF.
[23:21:37][C][wifi:037]: Setting up WiFi...
[23:21:37][D][wifi:365]: Starting scan...
[23:21:43][D][wifi:380]: Found networks:
[23:21:43][I][wifi:426]: - 'majdzik' (04:D4:C4:C1:80:E0) ▂▄▆█
[23:21:43][D][wifi:427]:     Channel: 3
[23:21:43][D][wifi:428]:     RSSI: -25 dB
[23:21:43][I][wifi:426]: - 'majdzik' (60:45:CB:59:B5:08) ▂▄▆█
[23:21:43][D][wifi:427]:     Channel: 11
[23:21:43][D][wifi:428]:     RSSI: -72 dB
[23:21:43][D][wifi:430]: - '' (00:1E:2A:61:68:E8) ▂▄▆█
[23:21:43][D][wifi:430]: - 'majdzik-gosc' (60:45:CB:59:B5:09) ▂▄▆█
[23:21:43][D][wifi:430]: - 'Patryk_NET' (24:4B:FE:BD:C8:78) ▂▄▆█
[23:21:43][D][wifi:430]: - 'Patryk_NET' (D4:6E:0E:2E:FC:C7) ▂▄▆█
[23:21:43][I][wifi:235]: WiFi Connecting to 'majdzik'...
[23:21:44][I][wifi:498]: WiFi Connected!
[23:21:44][C][wifi:344]:   SSID: 'majdzik'
[23:21:44][C][wifi:345]:   IP Address: 192.168.2.146
[23:21:44][C][wifi:347]:   BSSID: 04:D4:C4:C1:80:E0
[23:21:44][C][wifi:348]:   Hostname: 'tuje'
[23:21:44][C][wifi:352]:   Signal strength: -29 dB ▂▄▆█
[23:21:44][C][wifi:356]:   Channel: 3
[23:21:44][C][wifi:357]:   Subnet: 255.255.255.0
[23:21:44][C][wifi:358]:   Gateway: 192.168.2.1
[23:21:44][C][wifi:359]:   DNS1: 192.168.2.1
[23:21:44][C][wifi:360]:   DNS2: 0.0.0.0
[23:21:44][D][wifi:507]: Disabling AP...
[23:21:51]
[23:21:51] ets Jan  8 2013,rst cause:4, boot mode:(3,6)

```

**Additional information and things you've tried:**

whi-tw, 2021-07-07:
Are you also having trouble rolling back via the captive portal?

majdzik84, 2021-07-07:
yes, I had to upload the previous configuration file via usb

jesserockz, 2021-07-19:
Duplicate of #2198

closed 2021-07-19 by jesserockz

KC-inDomus, 2021-07-19:
Same problem with a Sonoff Basic R1 board, no problem with a R3 board.