esphome/issues #2206: ESPHome Crashes during compiling

ChrisHayes31, 2021-06-29:
When compiling a yaml file to do any type of install the system stops. Sometimes it auto reboots but most occasions I have to physically reboot the Pi.
Sometimes the compiling gets a little further but most occasions crashes.
Below is the last crash and the api.server.cpp is where it regularly crashes.

INFO Reading configuration /config/esphome/remote-control-wifi.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/remote-control-wifi
Processing remote-control-wifi (board: d1_mini; framework: arduino; platform: platformio/espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Library Manager: Installing Update
Library Manager: Already installed, built-in library
Dependency Graph
|-- ESPAsyncTCP-esphome 1.2.3
|   |-- ESP8266WiFi 1.0
|-- ESPAsyncWebServer-esphome 1.2.7
|   |-- ESPAsyncTCP-esphome 1.2.3
|   |   |-- ESP8266WiFi 1.0
|   |-- Hash 1.0
|   |-- ESP8266WiFi 1.0
|-- ESP8266WiFi 1.0
|-- ESP8266mDNS 1.2
|   |-- ESP8266WiFi 1.0
|-- Wire 1.0
|-- DNSServer 1.1.1
|   |-- ESP8266WiFi 1.0
Compiling /data/remote-control-wifi/.pioenvs/remote-control-wifi/src/esphome/components/api/api_connection.cpp.o
Compiling /data/remote-control-wifi/.pioenvs/remote-control-wifi/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/remote-control-wifi/.pioenvs/remote-control-wifi/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/remote-control-wifi/.pioenvs/remote-control-wifi/src/esphome/components/api/api_server.cpp.o

End report listing as this is where it crashes

I have tried reinstalling a new instance of Home Assistant, updated to latest versions and addons but nothing helps.
Please help resolve.
cs_hayes@hotmail.com

 

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

facaf, 2021-06-29:
I have the same problem! I tried with nodemcu esp8266 and esp32, with different yaml.

OttoWinter, 2021-06-30:
What is likely happening is that you're running out of memory, and the OS kills the compiler process.

Is this running on older RPi variants (v1,v2?), those have very little memory. Also make sure you're not running many other memory hungry addons

facaf, 2021-06-30:
> Lo que probablemente suceda es que se está quedando sin memoria y el sistema operativo mata el proceso del compilador.
> 
> ¿Se está ejecutando en variantes de RPi más antiguas (v1, v2?), Que tienen muy poca memoria. También asegúrese de que no está ejecutando muchos otros complementos que consumen mucha memoria

Hi! thanks for the reply.
This is not my case, I use a RPi 3B + with Ultra SD 32gb and HASSIO os. I have used RAM at 21% at most and SD at 25% (few add-ons installed and all were when ESPHome was running).
The problem started with the new updates (I don't know which, if from ESPhome or HASSIO).
I did the same tests as "ChrisHayes31", I even tested devices with YAML that were already working.
What remains for me to try is to do a new installation, but on an RPi4 and see what happens.

vellur, 2021-07-26:
I have the same problem. ESPHome was updated automatically and now i tried to update my yaml files prior that it always hangs at the same state. Need to unplug the power to resume HA.

Froosh, 2021-08-06:
Definitely a memory limitation on the Pi3(b).

I started by stepping back through previous ESPHome releases by using the configuraion options. v1.19.0 *almost* gets to complete the build for me, but still eventually exhausted memory and stopped; the root/supervisor SSH session (port 22222) hangs, observer stops responding, etc.

FYI: Trying to set configuration to use v1.18.0 doesn't work with the v1.20.4 container as it is attempting to start esphome with an unrecognised command line parameter (likely due to https://esphome.io/changelog/v1.19.0.html#command-line-interface). Alas, v1.18.0 was my last known good version.

While trying to build an image with ESPHome, it hangs when the memory usage is very high including the swap partition:
[image]

Memory info after a fresh power cycle:
```
# free -h
              total        used        free      shared  buff/cache   available
Mem:          912Mi       566Mi        18Mi       0.0Ki       327Mi       342Mi
Swap:         228Mi       9.0Mi       218Mi
```

So, I tried adding an extra swapfile on an external SSD to see what happened ... and yes, success! Seems like at least an extra 256-512MB are required to build ESPHome images with more recent versions.

```
dd if=/dev/zero of=/mnt/data/swapfile bs=1048576 count=1024
mkswap /mnt/data/swapfile
swapon /mnt/data/swapfile
```

Running Home Assistant OS 6.2, core-2021.8.1, supervisor-2021.06.8, ESPHome container v1.20.4

matteofranceschini, 2021-08-12:
I'm running HassOS in a virtual machine on my synology NAS and I experience the same issue, but only when compiling a yaml that has the code for a ESP32 camera.

By looking at the logs and stats, it seems that I don't have issues with the RAM space but, if I've understood correctly, with the Supervisor watchdog.

When the compiler starts, it eats up all the processing capabilities of my hardware, including (I guess) those needed to run Core: this causes missing answers to the Supervisor's watchdog and (correctly) triggers a reboot.

I don't know if it's possible, but we could either:
- [workaround] reduce the watchdog "readiness", allowing more time for Core to answer
- lower the priority of Esphome's container tasks (especially during compile)
- cap the maximum resources available for the esphome container

[image]
[image]

If more information is needed to dig deeper into the issue, let me know :)

ssieb, 2022-10-27:
A compile option has just been added to dev, so you can limit the number of processes.
https://github.com/esphome/esphome/pull/3952

closed 2022-10-27 by ssieb