esphome/issues #2208: external_component not finding main or default branch using short hand form

gb53smith, 2021-06-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini
Probably not board dependent.

**ESPHome version (latest production, beta, dev branch)**

v19.2

**Affected component:**

external_component:

**Description of problem:**
The external component can be found on my dev1 branch but not on @main or default by removing the @.
With @main the github.com/esphome/esphome/components/dht files are copied to the output source code directory.
Without @ there is a yaml error about missing __int__.py

**Problem-relevant YAML-configuration entries:**
```yaml

external_components:
  - source: github://gb53smith/esphome@main
    components: [dht]
```
Long form works with this
```
  source:
    type: git
    url: https://github.com://gb53smith/esphome
    ref: main
  components: [dht]
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

OttoWinter, 2021-06-30:
Please post what error ESPHome shows. Also note that if you added the `__init__.py` file later, it won't get pulled right away due to the `refresh` setting

gb53smith, 2021-06-30:
The error message is
```
Compiling /data/plants/.pioenvs/plants/src/esphome/components/dht/dht.cpp.o
Compiling /data/plants/.pioenvs/plants/src/main.cpp.o
src/esphome/components/dht/dht.cpp:246:26: error: no 'bool esphome::dht::DHT::read(bool)' member function declared in class 'esphome::dht::DHT'
 bool DHT::read(bool force) {                   ^
*** [/data/plants/.pioenvs/plants/src/esphome/components/dht/dht.cpp.o] Error 1
```
In my latest dht.h (main)  there is ```bool readSensor```.  The ```bool read``` was used on another branch in my repository.

So in summary what works is:
```
source: github://esphome/esphome@dev  #any branch other than main
   components: [dht]

# This long form reference to main
source:
    type: git
    url: https://github.com:/gb53smith/esphome
    ref: main
  components: [dht]
```
What does not work is:
```
source: github://gb53smith/esphome@main  #short form reference to main
    components: [dht]

source: github://gb53smith/esphome #short form reference to default branch
    components: [dht]

```
The empty __init__.py  filename was missing the second underscore in an earlier version but that was corrected and committed.

PS.
I found this external_component feature to be very useful when testing various versions of the DHT component

glmnet, 2021-06-30:
Try adding refresh: 1s

So it refreshes for sure. 

Default is 1 day

gb53smith, 2021-06-30:
The main branch was 2 days old when I tried the above compile. The long form finds the right files.
One should be able to replicate this by adding a DHT component to any yaml configuration with the failing external_component statements.

glmnet, 2021-07-01:
Indeed, all versions (long form, short form, and short form with explicity branch) ends up in different working copies.

All of them are giving me this error:
```
esphome compile .\test.issue_2208.yaml
INFO Reading configuration .\test.issue_2208.yaml...
Failed config

sensor.dht: [source .\test.issue_2208.yaml:30]

  Unknown value 'AUTO_DETECT', valid options are 'DHT11', 'DHT22'.
  platform: dht
  pin: GPIO2
  temperature:
    name: lele_sensor
```
if I add explicitely model either dht11 or dht22 then I get an error that [DHTNew.h is not found which is also correct
](https://github.com/gb53smith/esphome/blob/94abecb09f68e548d509dcf5340f790fe7d5b2dd/esphome/components/dht/dht.h#L8)

gb53smith, 2021-07-01:
Sorry, I should have mentioned that my version does not support AUTO_DETECT so either DHT11 or DHT22 must be specified.
To find  DHTNew.h, the esphome: section needs
```
  platformio_options:
    lib_deps: diaoul/DHTNew@1.0.0
```
The long form should now compile but not the short form.

(My code is an experiment to demonstrate a "wrapper" approach to creating a Esphome component, relying on a "known good" PlatformIO library.  I had problems with the Esphome DHT component the first time I tried Esphome.  My PlatformIO based node worked with the same hardware.)

I also tested ```github://esphome/esphome@dev``` and found that it fixed my original issue too.

glmnet, 2021-07-01:
> Sorry, I should have mentioned that my version does not support AUTO_DETECT so either DHT11 or DHT22 must be specified.
>
> To find  DHTNew.h, the esphome: section needs
>
> ``` [...]

Look for cg.add_library

gb53smith, 2021-07-01:
Do I add cg.add_library to sensor.py?  
What is the syntax? 
Does that elminate need to specify a paltaformio option?
How does that affect the external_component?

glmnet, 2021-07-01:
> Do I add cg.add_library to sensor.py?  
> 
Yes
> What is the syntax? 
> 
Search code base for examples 
> Does that elminate need to specify a paltaformio option?
> 
Yes
> How does that affect the external_component?  
> 
Should just work 
> 
> 
> 
> 
>

gb53smith, 2021-07-01:
Thanks @glmnet for the tip to use cg.add_library.  It does cleanup my component so the platformio option in the yaml configuration is no longer necessary.

The original issue reported remains unresolved.

OttoWinter, 2021-07-01:
Have you tried `refresh` yet?

gb53smith, 2021-07-01:
My local repository is on Windows 10.  I found no explicit "refresh" command.
From Github Desktop I fetched the origin and saw "refreshing" in the operation dialog.
From Git Bash I tried this command line:
```
$ git fetch --prune https://github.com/gb53smith/esphome.git
From https://github.com/gb53smith/esphome
 * branch            HEAD       -> FETCH_HEAD
```
Did that do a refresh?
I don't understand your code but can the short form be translated to the long form which works?

The other related error occurs when not using @main (ie requesting default branch) the configuration checker gives this error:
```
Cannot find the __init__.py file for the component dht.
```
I have an empty  __init__.py file in esphome/esphome/components/dht and nowhere else.

OttoWinter, 2021-07-01:
See refresh here: https://esphome.io/components/external_components.html - it's config option to add to the external_component. In your case `refresh: always` makes it so the repo is checked for changes on every compile

gb53smith, 2021-07-01:
I tried the external_component refresh.  The only option I found to pass the yaml validation was "1d".
Other options like "always", "1day", "0s" failed with message like this:
```
INFO Reading configuration /config/esphome/plants.yaml...
INFO Executing git pull https://github.com/gb53smith/esphome.git@main
Failed config

external_components: [source /config/esphome/plants.yaml:47]
  
  refusing to merge unrelated histories.
  - [source /config/esphome/plants.yaml:47]
    source: github://gb53smith/esphome@main
    refresh: always
    components: 
      - dht
```
or 
```refusing to merge unrelated histories.``` in the yaml editor frontend.
I am not a GIT expert so I may have messed up my repository.
On the other hand, branches other than "main" and the long form works.

gb53smith, 2021-07-04:
On a separate Ubuntu machine I setup up an esphome development environment.
I used these commands (a little different from the documentation) to setup a node and then to run the compile:
```
esphome plants wizard
esphome dashboard plants &
```
I copied my plants.yaml code from the HomeAssistant Add-On running on another Raspberry Pi to the Ubuntu machine.
Using the frontend at port 6052 **both shorthand versions worked**.  ie with and without @main.

So the issue is with the HomeAssistant Addon. (v1.19.2)  It tried deleting the plants directory but the error reoccurred.
Perhaps something should be deleted in the .esphome directory?

glmnet, 2021-07-04:
Yes. If you delete the folders inside .esphome/external_components you should restart fresh. 

If you figure out a way to reproduce the issue then we will have something to work on. 

By any chance you remember doing something unusual like force pushing or renaming the branch? Those are changes that cannot be updated by ESPHome external components by a simple git pull.

gb53smith, 2021-07-04:
**SOLVED!** Thankyou
I deleted all 13 folders found inside ```.esphome/external_components``` and the short hand notations worked (including the refresh: always option).
I probably did something unexpected in my repository.  Here is a graphic showing all the branch history should you wish to enhance the ESPHome code.
[image]

closed 2021-07-04 by gb53smith