esphome/issues #2209: Custom UART component is not working

zyuan888, 2021-06-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio 2021.6.6
**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP8266 (D1 Mini)

**ESPHome version (latest production, beta, dev branch)**
ESPHome 1.19.4

**Affected component:**
https://esphome.io/custom/uart.html

**Description of problem:**
I am trying to build a TTS device using UART in ESPHome.  My TTS module requires a TX ping from ESP8266, in order to receive text messages.  But I was unable to make the custom UART component working in ESPHome, even with the most basic setting.  I reduced my own codes to bare minimal, adopted sample codes from your website (https://esphome.io/custom/uart.html), but nothing works.  After flashed firmware, the board was having trouble to connect to WiFi.  But I can still update the firmware via OTA.  After I asked some questions in discord, I believe there is an  internal crash in the software, and cased it connected to wifi in safe mode. I tried many different coding, as well as different ESP module, all were having the same issue.  Now, I believe this is a bug in the ESPHome, rather than my own coding or settings.  Please help.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
1) esphome.ymal file
--------------------------------------------------------------------
esphome:
  name: esp-d1-mini-01
  platform: ESP8266
  board: d1_mini
  includes:
    - TTS_Speaker.h

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxxxxxxxxxx"

wifi:
  ssid: "xxxxxxxxxxxxxxxxxx"
  password: "xxxxxxxxxxxxxxxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp-D1-Mini-01 Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxx"

captive_portal:

uart:
  id: uart_bus
  tx_pin: D0
  rx_pin: D1
  baud_rate: 9600

custom_component:
- lambda: |-
    auto my_custom = new MyCustomComponent(id(uart_bus));
    return {my_custom};
```

2) Include file "TTS_Speaker.h"
------------------------------------------------------------------------
#include "esphome.h"

class MyCustomComponent : public Component, public UARTDevice {
 public:
  MyCustomComponent(UARTComponent *parent) : UARTDevice(parent) {}

  void setup() override {
    // nothing to do here
  }
  void loop() override {
    // Use Arduino API to read data, for example
    
    ESP_LOGD("Com", "UART activated");
    delay(15000);
    // etc
  }
};
**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
IINFO Reading configuration /config/esphome/esp-d1-mini-01.yaml...
INFO Starting log output from esp-d1-mini-01.local using esphome API
INFO Connecting to esp-d1-mini-01.local:6053 (192.168.1.192)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 1 seconds
INFO Connecting to esp-d1-mini-01.local:6053 (192.168.1.192)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 1 seconds
INFO Connecting to esp-d1-mini-01.local:6053 (192.168.1.192)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 2 seconds
INFO Connecting to esp-d1-mini-01.local:6053 (192.168.1.192)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 3 seconds
INFO Connecting to esp-d1-mini-01.local:6053 (192.168.1.192)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 5 seconds
INFO Connecting to esp-d1-mini-01.local:6053 (192.168.1.192)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Timeout while waiting for message response!). Re-Trying in 7 seconds
INFO Connecting to esp-d1-mini-01.local:6053 (192.168.1.192)

```

**Additional information and things you've tried:**
I tried many different ESP8266 boards and code variations, but none of them working once I touched UART component.  I also connected my board to laptop's USB port and use the flash program ESPHome-Flasher-1.3.0-Windows-x64.exe to view the logs, and it looks like below:
[23:24:06][I][wifi:235]: WiFi Connecting to 'xxxxxxxxx'...
[23:24:06][D][Com:013]: UART activated
[23:24:08][W][wifi_esp8266:444]: Event: Disconnected ssid='xxxxxxxx' bssid=F0:2F:74:E1:00:40 reason='Auth Expired'
[23:24:21][W][wifi:532]: Error while connecting to network.
[23:24:21][W][wifi:570]: Restarting WiFi adapter...
[23:24:21][I][wifi:235]: WiFi Connecting to 'xxxxxxxxxxx'...
[23:24:21][D][Com:013]: UART activated
[23:24:24][W][wifi_esp8266:444]: Event: Disconnected ssid='xxxxxxxxxxx' bssid=F0:2F:74:E1:00:40 reason='Auth Expired'
[23:24:36][W][wifi:532]: Error while connecting to network.
[23:24:36][W][wifi:570]: Restarting WiFi adapter...
[23:24:36][D][Com:013]: UART activated
[23:24:51][D][wifi:365]: Starting scan...
[23:24:51][I][wifi:137]: Starting fallback AP!
[23:24:51][C][wifi:190]: Setting up AP...
[23:24:51][C][wifi:192]:   AP SSID: 'Esp-D1-Mini-01 Fallback Hotspot'
[23:24:51][C][wifi:193]:   AP Password: 'xxxxxxxx'
[23:24:52][C][wifi:202]:   IP Address: 192.168.4.1
[23:24:52][D][Com:013]: UART activated

*** note: the message "[Com:013]: UART activated" was issued from my own coding via ESP_LOGD to indicate that the program is running.

OttoWinter, 2021-06-30:
> `delay(15000);`

There's the issue. With that long delay you're blocking the ESPHome loop as well, and all other ESPHome code.

zyuan888, 2021-07-01:
Otto,

Thanks you very much.  I removed this line of code and it worked now.  

I overlooked this line when I was doing my test.  I initially thought that it blocked ESPhome loop without any delay, crashed the ESP8266, opposite what you said.   Earlier on, I used exact the same sample codes for testing, it somehow had the same WiFI connection issue, so I moved on to try custom API device, until I realized that I still need a UART bus to send bytes to TX pin. So, I  started to try UART codes again, unfortunately overlooked this 'delay' line.  

Now I have a better understanding how ESPHome works. Thank you.

OttoWinter, 2021-07-10:
Nice!

closed 2021-07-10 by OttoWinter