esphome/issues #2210: Bootloop device after updating

eugene-ki, 2021-06-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Add-on in Hass.io (Supervised)
**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic (ESP8285)
**ESPHome version (latest production, beta, dev branch)**

latest production (1.19.4)
**Affected component:**

WiFi Component

**Description of problem:**
After updating to one of the 1.19.* releases there are bootloop on my Sonoff Basic. It's can't connect to WiFi.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: relay_kitchen
  platform: ESP8266
  board: esp8285
  arduino_version: 2.4.2

wifi:
  ssid: ***
  password: '***'
  fast_connect: true

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "relay_thermo_kitchen_button"
    on_press:
      - switch.toggle: fakebutton

switch:
  - platform: template
    name: "relay_thermo_kitchen_switch"
    optimistic: true
    id: fakebutton
    turn_on_action:
    - switch.turn_on: relay
    - light.turn_on: led
    turn_off_action:
    - switch.turn_off: relay
    - light.turn_off: led
  - platform: gpio
    id: relay
    pin: GPIO12

output:
  - platform: esp8266_pwm
    id: basic_green_led
    pin:
      number: GPIO13
      inverted: True

light:
  - platform: monochromatic
    name: "relay_thermo_kitchen_light"
    output: basic_green_led
    id: led

sensor:
  - platform: wifi_signal
    name: "relay_thermo_kitchen_wifi_signal"
    update_interval: 3600s
  - platform: uptime
    name: "relay_thermo_kitchen_uptime"

```

**Logs (if applicable):**

```
There have been suspected X unsuccessful boot attempts. <-- This message in loop

And my router had this log:
192.168.x.x - MAC - DHCP assigned
192.168.x.x - MAC - DHCP deassigned
192.168.x.x - MAC - DHCP assigned
192.168.x.x - MAC - DHCP deassigned

Also in loop, when Sonoff Basic was turned on.
```

**Additional information and things you've tried:**

I had an old 1.16.2 compiled .bin file and it's working fine after I have flashed it by "ESPHome-Flasher-1.2.0-windows-x64"

OttoWinter, 2021-06-30:
> There have been suspected X unsuccessful boot attempts. <-- This message in loop

Please post the complete log output.

eugene-ki, 2021-06-30:
> > There have been suspected X unsuccessful boot attempts. <-- This message in loop
> 
> Please post the complete log output.

Here is a part of full log. This part goes on in loop.
```
[21:38:27][I][logger:170]: Log initialized
[21:38:27][C][ota:366]: There have been 2 suspected unsuccessful boot attempts.
[21:38:27][I][app:029]: Running through setup()...
[21:38:27][D][binary_sensor:034]: 'relay_thermo_kitchen_button': Sending initial state OFF
[21:38:27][C][switch.gpio:011]: Setting up GPIO Switch 'relay'...
[21:38:27][D][switch:025]: 'relay' Turning OFF.
[21:38:27][D][switch:045]: 'relay': Sending state OFF
[21:38:27][D][switch:025]: 'relay' Turning OFF.
[21:38:27][C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[21:38:27][C][light:097]: Setting up light 'relay_thermo_kitchen_light'...
[21:38:27][D][light:275]: 'relay_thermo_kitchen_light' Setting:
[21:38:27][D][light:288]:   Brightness: 100%
[21:38:27][C][wifi:037]: Setting up WiFi...
[21:38:27][I][wifi:235]: WiFi Connecting to 'Flat-51'...
[21:38:27][D][sensor:117]: 'relay_thermo_kitchen_uptime': Sending state 0.13100 s with 0 decimals of accuracy
[21:38:31][I][wifi:498]: WiFi Connected!
[21:38:31][C][wifi:344]:   SSID: 'Flat-51'
[21:38:31][C][wifi:345]:   IP Address: 192.168.51.40
[21:38:31][C][wifi:347]:   BSSID: 74:4D:28:7D:91:5B
[21:38:31][C][wifi:348]:   Hostname: 'relay_thermo_kitchen'
[21:38:31][C][wifi:352]:   Signal strength: -67 dB ▂▄▆█
[21:38:31][C][wifi:356]:   Channel: 8
[21:38:31][C][wifi:357]:   Subnet: 255.255.255.0
[21:38:31][C][wifi:358]:   Gateway: 192.168.51.1
[21:38:31][C][wifi:359]:   DNS1: 192.168.51.1
[21:38:31][C][wifi:360]:   DNS2: 0.0.0.0
[21:38:31]
[21:38:31]Exception (0):
[21:38:31]epc1=0xa8c0cc8e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[21:38:31]
[21:38:31]ctx: cont 
[21:38:31]sp: 3ffffa00 end: 3fffffd0 offset: 01a0
[21:38:31]
[21:38:31]>>>stack>>>
[21:38:31]3ffffba0:  00000007 0000000f 00302073 00000003  
[21:38:31]3ffffbb0:  3ffea191 00000003 3ffffbe0 00000030  
[21:38:31]3ffffbc0:  00000000 3ffe8308 3ffffc70 3ffea316  
[21:38:31]3ffffbd0:  00000004 3ffe8308 000001f5 4021cbf5  
[21:38:31]3ffffbe0:  3fff1fd5 00000016 00000000 ffff0208  
[21:38:31]3ffffbf0:  3ffffd80 3ffffd70 00000014 00000007  
[21:38:31]3ffffc00:  3fff30b2 0000002e 00000012 3fff1fb4  
[21:38:31]3ffffc10:  00000000 3ffe8308 3ffffcc0 3fff168d  
[21:38:31]3ffffc20:  3fff1658 3fff1658 3fff306c 4022226a  
[21:38:31]3ffffc30:  402225ff 3fff306c 3fff30b2 40222608  
[21:38:31]3ffffc40:  00000008 3ffffc70 3f302075 40100f12  
[21:38:31]3ffffc50:  fffffffa 000001f5 3fff1f74 3fff30c0  
[21:38:31]3ffffc60:  3fff306c 3fff3058 3fff1658 40228855  
[21:38:31]3ffffc70:  005e0001 0000fb00 3fff10bc 40216aa4  
[21:38:31]3ffffc80:  00000018 3fff3058 3fff306c 402292dc  
[21:38:31]3ffffc90:  3ffeaee8 3ffffcc0 00000008 00000000  
[21:38:31]3ffffca0:  3fff165c 00000001 00000000 00000002  
[21:38:31]3ffffcb0:  3fff15b8 00000004 3fff3054 40229944  
[21:38:31]3ffffcc0:  ffffffbd 3fff1fbf 3fff1f74 3fff3054  
[21:38:31]3ffffcd0:  00000000 3fff306c 3fff30d8 4022931c  
[21:38:31]3ffffce0:  3fff1658 3ffffd10 00000004 3fff3054  
[21:38:31]3ffffcf0:  00000016 3fff1658 3fff3054 40228aad  
[21:38:31]3ffffd00:  3fff1658 3ffffd10 00000004 4022265c  
[21:38:31]3ffffd10:  00000494 00000000 3fff17b4 40222698  
[21:38:31]3ffffd20:  3fff3058 3fff1658 3fff18b4 40228b7e  
[21:38:31]3ffffd30:  3fff1894 40217c2c 00000010 00000020  
[21:38:31]3ffffd40:  3ffffdac 3fff1658 3fff3054 40228d57  
[21:38:31]3ffffd50:  3ffffdac 3ffeb3a4 3fff1658 40228de3  
[21:38:31]3ffffd60:  ffffffbd 00000001 3ffffde0 00000001  
[21:38:31]3ffffd70:  3fff1054 3fff0fd0 3fff1054 402152eb  
[... 264 lines cut ...]
[21:38:40]
[21:38:40]wdt reset
[21:38:40]load 0x4010f000, len 1384, room 16 
[21:38:40]tail 8
[21:38:40]chksum 0x2d
[21:38:40]csum 0x2d
[21:38:40]vbb28d4a3
[21:38:40]~ld
[21:38:40][I][logger:170]: Log initialized
[21:38:40][C][ota:366]: There have been 5 suspected unsuccessful boot attempts.
[21:38:40][I][app:029]: Running through setup()...
[21:38:40][D][binary_sensor:034]: 'relay_thermo_kitchen_button': Sending initial state OFF
[21:38:40][C][switch.gpio:011]: Setting up GPIO Switch 'relay'...
[21:38:40][D][switch:025]: 'relay' Turning OFF.
[21:38:40][D][switch:045]: 'relay': Sending state OFF
[21:38:40][D][switch:025]: 'relay' Turning OFF.
[21:38:40][C][esp8266_pwm:017]: Setting up ESP8266 PWM Output...
[21:38:40][C][light:097]: Setting up light 'relay_thermo_kitchen_light'...
[21:38:40][D][light:275]: 'relay_thermo_kitchen_light' Setting:
[21:38:40][D][light:288]:   Brightness: 100%
[21:38:40][C][wifi:037]: Setting up WiFi...
[21:38:40][I][wifi:235]: WiFi Connecting to 'Flat-51'...
[21:38:40][D][sensor:117]: 'relay_thermo_kitchen_uptime': Sending state 0.13000 s with 0 decimals of accuracy

```

adonno, 2021-07-02:
not sure if related, but noticed that my devices try to connect to every available network instead of the one defined in the yaml 
this results in crashing and boot loops until at some point it finally connects to the network and then runs stable

whi-tw, 2021-07-07:
I'm also seeing a similar problem after updating my sonoff minis and sonoff POWR2s.

While the wifi network is available, they constantly bootloop. Now I've turned off the SSID, they're functioning but in failsafe.

Today's job is rolling them all back to 1.8.0.

Warter21, 2021-07-09:
I have a same issue with d1_mini. After rolled back to 1.18 everything is working fine again.

rvdm82, 2021-07-13:
I have a similar issue on an ESP8266 ( Shelly 2.5 ). right after the device connects to the AP it crashes. Tried this on a 2 other  devices, same thing. 
1.18.0 doesn't have this issue. Below is the log output with debug enabled.

`[22:46:52][C][wifi:352]:   Signal strength: -64 dB ▂▄▆█
[22:46:52][C][wifi:356]:   Channel: 6
[22:46:52][C][wifi:357]:   Subnet: 255.255.255.0
[22:46:52][C][wifi:358]:   Gateway: 192.168.255.1
[22:46:52][C][wifi:359]:   DNS1: 192.168.255.1
[22:46:52][C][wifi:360]:   DNS2: 0.0.0.0
[22:46:52]
[22:46:52]Exception (0):
WARNING Exception type: Illegal instruction (Is the flash damaged?)
[22:46:52]epc1=0x4021bb40 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d glazen_kot_luik -t idedata
WARNING Decoded 0x4021bb40: pgm_read_byte_inlined at pgmspace.cpp
[22:46:53]
[22:46:53]ctx: cont 
[22:46:53]sp: 3ffff9c0 end: 3fffffd0 offset: 01a0
[22:46:53]
[22:46:53]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[22:46:53]3ffffb60:  00000007 0000000f 00302073 00000003  
[22:46:53]3ffffb70:  3ffea170 00000003 3ffffba0 00000030  
[22:46:53]3ffffb80:  00000000 3ffe8308 3ffffc30 3ffeab68  
[22:46:53]3ffffb90:  00000004 3ffe8308 000001f5 4021e405  
WARNING Decoded 0x4021e405: _vsnprintf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:73
[22:46:53]3ffffba0:  3fff2305 00000016 00000000 ffff0208  
[22:46:53]3ffffbb0:  3ffffd40 3ffffd30 00000014 00000007  
[22:46:53]3ffffbc0:  3fff1e2a 0000002e 00000001 3fff22e4  
[22:46:53]3ffffbd0:  00000000 3ffe8308 3ffffc80 3fff19c5  
[22:46:53]3ffffbe0:  3fff1990 3fff1990 3fff1de4 402230f6  
WARNING Decoded 0x402230f6: new_linkoutput at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:259
[22:46:53]3ffffbf0:  4022348b 3fff1de4 3fff1e2a 40223494  
WARNING Decoded 0x4022348b: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:305
WARNING Decoded 0x40223494: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:305
[22:46:53]3ffffc00:  00000008 3ffffc30 00302075 40100f12  
WARNING Decoded 0x40100f12: pp_post
[22:46:53]3ffffc10:  00000002 000001f5 3fff22a4 3fff1e38  
[22:46:53]3ffffc20:  3fff1de4 3fff1dd0 3fff1990 402296e5  
WARNING Decoded 0x402296e5: etharp_output_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c:893
[22:46:53]3ffffc30:  005e0001 0000fb00 3fff132c 40218168  
WARNING Decoded 0x40218168: HardwareSerial::write(unsigned char const*, unsigned int)
[22:46:54]3ffffc40:  00000018 3fff1dd0 3fff1de4 4022a16c  
WARNING Decoded 0x4022a16c: ip4_output_if_opt_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:962
[22:46:54]3ffffc50:  3ffeb0b0 3ffffc80 00000008 00000000  
[22:46:54]3ffffc60:  3fff1994 00000001 00000000 00000002  
[22:46:54]3ffffc70:  3fff18f0 00000004 3fff1dcc 4022a7d8  
WARNING Decoded 0x4022a7d8: mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:136
[22:46:54]3ffffc80:  ffffffc0 3fff22ef 3fff22a4 3fff1dcc  
[22:46:54]3ffffc90:  00000000 3fff1de4 3fff1e50 4022a1ac  
WARNING Decoded 0x4022a1ac: ip4_output_if_opt at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:788
[22:46:54]3ffffca0:  3fff1990 3ffffcd0 00000004 3fff1dcc  
[22:46:54]3ffffcb0:  00000016 3fff1990 3fff1dcc 4022993d  
WARNING Decoded 0x4022993d: igmp_send at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
[22:46:54]3ffffcc0:  3fff1990 3ffffcd0 00000004 402234e8  
WARNING Decoded 0x402234e8: do_memp_malloc_pool at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c:231
[22:46:54]3ffffcd0:  00000494 00000000 3fff1aec 40223524  
WARNING Decoded 0x40223524: memp_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c:231
[22:46:54]3ffffce0:  3fff1dd0 3fff1990 3fff1d84 40229a0e  
WARNING Decoded 0x40229a0e: igmp_lookup_group at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
[22:46:54]3ffffcf0:  3fff1d64 402192f0 00000010 00000020  
WARNING Decoded 0x402192f0: operator new(unsigned int)
[22:46:54]3ffffd00:  3ffffd6c 3fff1990 3fff1dcc 40229be7  
WARNING Decoded 0x40229be7: igmp_start_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
 (inlined by) igmp_joingroup_netif at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:521                                                                                                                                                                                                                                      
[22:46:54]3ffffd10:  3ffffd6c 3ffeb560 3fff1990 40229c73  
WARNING Decoded 0x40229c73: igmp_joingroup at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
[... cut ...]
[22:46:54]3fffffa0:  feefeffe feefeffe feefeffe 3fff13ec  
[22:46:54]3fffffb0:  3fffdad0 00000000 3fff13e4 402194e0  
WARNING Decoded 0x402194e0: loop_wrapper() at core_esp8266_main.cpp
[22:46:54]3fffffc0:  feefeffe feefeffe 3ffe8624 40100721  
WARNING Decoded 0x40100721: cont_wrapper
[22:46:54]<<<stack<<<
[22:46:54]
[22:46:54] ets Jan  8 2013,rst cause:1, boot mode:(3,7)
[22:46:54]
[22:46:54]load 0x4010f000, len 1384, room 16 
[22:46:54]tail 8
[22:46:54]chksum 0x2d
[22:46:54]csum 0x2d
[22:46:54]vbb28d4a3
[22:46:54]~ld
`

Habiulty, 2021-07-26:
Same problem with an ESP8266 Sonoff 4CH : reboot just after listing the APs. Rolled back to 1.7, and it works fine

rvdm82, 2021-08-04:
I tires 1.20.3 , and the issue is still present - ESP8266 ( Shelly 2.5 ). r1.18.0 doesn't have this issue. Below is the log output with debug enabled ( stack trace).

```
[15:06:26][C][wifi:375]:   DNS2: 0.0.0.0
[15:06:26]
[15:06:26]Exception (0):
WARNING Exception type: Illegal instruction (Is the flash damaged?)
[15:06:26]epc1=0x46435f69 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d bureau_luik -t idedata
[15:06:27]
[15:06:27]ctx: cont 
[15:06:27]sp: 3ffff9c0 end: 3fffffd0 offset: 01a0
[15:06:27]
[15:06:27]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[15:06:27]3ffffb60:  00000007 0000000f 00302073 00000003  
[15:06:27]3ffffb70:  3ffea23c 00000003 3ffffba0 00000030  
[15:06:27]3ffffb80:  00000000 3ffe8308 3ffffc30 3ffeac6f  
[15:06:27]3ffffb90:  00000004 3ffe8308 000001f5 4021e799  
WARNING Decoded 0x4021e799: _vsnprintf_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/vsnprintf.c:73
[15:06:27]3ffffba0:  3fff2415 00000016 00000000 ffff0208  
[15:06:27]3ffffbb0:  3ffffd40 3ffffd30 00000014 00000007  
[15:06:27]3ffffbc0:  3fff428a 0000002e 00000000 3fff23f4  
[15:06:27]3ffffbd0:  00000000 3ffe8308 3ffffc80 3fff1ad5  
[15:06:27]3ffffbe0:  3fff1aa0 3fff1aa0 3fff4244 402235a6  
WARNING Decoded 0x402235a6: new_linkoutput at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:259
[15:06:27]3ffffbf0:  4022393b 3fff4244 3fff428a 40223944  
WARNING Decoded 0x4022393b: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:305
WARNING Decoded 0x40223944: ethernet_output at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:305
[15:06:27]3ffffc00:  00000008 3ffffc30 00302075 40100f12  
WARNING Decoded 0x40100f12: pp_post
[15:06:27]3ffffc10:  0000002b 000001f5 3fff23b4 3fff4298  
[15:06:27]3ffffc20:  3fff4244 3fff4230 3fff1aa0 40229b95  
WARNING Decoded 0x40229b95: etharp_output_LWIP2 at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c:893
[15:06:27]3ffffc30:  005e0001 0000fb00 3fff143c 402183c8  
WARNING Decoded 0x402183c8: HardwareSerial::write(unsigned char const*, unsigned int)
[15:06:27]3ffffc40:  00000018 3fff4230 3fff4244 4022a61c  
WARNING Decoded 0x4022a61c: ip4_output_if_opt_src at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:962
[15:06:27]3ffffc50:  3ffeb1a8 3ffffc80 00000008 00000000  
[15:06:27]3ffffc60:  3fff1aa4 00000001 00000000 00000002  
[15:06:27]3ffffc70:  3fff1a00 00000004 3fff422c 4022ac88  
WARNING Decoded 0x4022ac88: mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:136
[15:06:27]3ffffc80:  ffffffb4 3fff23ff 3fff23b4 3fff422c  
[15:06:27]3ffffc90:  00000000 3fff4244 3fff42b0 4022a65c  
WARNING Decoded 0x4022a65c: ip4_output_if_opt at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:788
[15:06:27]3ffffca0:  3fff1aa0 3ffffcd0 00000004 3fff422c  
[15:06:27]3ffffcb0:  00000016 3fff1aa0 3fff422c 40229ded  
WARNING Decoded 0x40229ded: igmp_send at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
[15:06:27]3ffffcc0:  3fff1aa0 3ffffcd0 00000004 40223998  
WARNING Decoded 0x40223998: do_memp_malloc_pool at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c:231
[15:06:27]3ffffcd0:  00000494 00000000 3fff1bfc 402239d4  
WARNING Decoded 0x402239d4: memp_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c:231
[15:06:27]3ffffce0:  3fff4230 3fff1aa0 3fff1fec 40229ebe  
WARNING Decoded 0x40229ebe: igmp_lookup_group at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
[15:06:27]3ffffcf0:  3fff41dc 40219550 00000010 00000020  
WARNING Decoded 0x40219550: operator new(unsigned int)
[15:06:27]3ffffd00:  3ffffd6c 3fff1aa0 3fff422c 4022a097  
WARNING Decoded 0x4022a097: igmp_start_timer at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
 (inlined by) igmp_joingroup_netif at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:521                                                                                                                                                                                                                                      
[15:06:27]3ffffd10:  3ffffd6c 3ffeb670 3fff1aa0 4022a123  
WARNING Decoded 0x4022a123: igmp_joingroup at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c:570
[15:06:27]3ffffd20:  ffffffb4 00000001 3ffffda0 00000001  
[15:06:27]3ffffd30:  3fff137c 3fff12f8 3fff137c 4021677b  
WARNING Decoded 0x4021677b: MDNSResponder::_listen()
[15:06:27]3ffffd40:  00000000 3fff12f8 3ffffd90 401006dc  
WARNING Decoded 0x401006dc: free
[15:06:27]3ffffd50:  3fff1640 0000047b 00000000 402195a8  
WARNING Decoded 0x402195a8: operator delete(void*)
[... cut ...]
[15:06:28]3fffffa0:  feefeffe feefeffe feefeffe 3fff14fc  
[15:06:28]3fffffb0:  3fffdad0 00000000 3fff14f4 40219758  
WARNING Decoded 0x40219758: loop_wrapper() at core_esp8266_main.cpp
[15:06:28]3fffffc0:  feefeffe feefeffe 3ffe85f0 40100721  
WARNING Decoded 0x40100721: cont_wrapper
[15:06:28]<<<stack<<<
[15:06:28]
[15:06:28] ets Jan  8 2013,rst cause:4, boot mode:(3,5)
[15:06:28]
[15:06:28]wdt reset
[15:06:28]load 0x4010f000, len 1384, room 16 
[15:06:28]tail 8
[15:06:28]chksum 0x2d
[15:06:28]csum 0x2d
[15:06:28]vbb28d4a3
[15:06:28]~ld

```

oxan, 2021-08-10:
I'll close this in favor of #2309. Please follow the instructions there if you've got this problem.

Especially try removing the `arduino_version` option, as you all seem to be using ancient versions.

closed 2021-08-10 by oxan