esphome/issues #2213: Remote receiver, configured to dump, does not dump received codes to log if action present

cerebrate, 2021-07-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io core-2021.6.6, using ESPHome addon 1.19.4

**ESP (ESP32/ESP8266, Board/Sonoff):**

Adafruit Feather HUZZAH

**ESPHome version (latest production, beta, dev branch)**

1.19.4

**Affected component:**

https://esphome.io/components/remote_receiver.html

**Description of problem:**

When the remote_receiver component is configured to dump received codes to the log (for example, using `dump: lg`), and then the `on_lg` event is configured to take an action on the receipt of an infrared code, received codes are no longer dumped to the log.

(I've submitted this as a bug report on the grounds that, well, this is unuseful behavior from the point of view of debugging which seems likely to be unintentional, as well as unintuitive - my impression was that I had broken something, from the lack of log entries, before I noticed that the unlogged codes received were nonetheless being handled by the `on_lg` event. If it would fit better as a feature request, please let me know and I can move it over there.)

**Problem-relevant YAML-configuration entries:**
```yaml
remote_receiver:
  pin: 
    number: 14
    inverted: True
    mode: INPUT_PULLUP
  dump: lg
  on_lg:
   - homeassistant.event:
      event: esphome.infrared.code_received
      data:
        source: living_room
        code: !lambda |-
          return x.data;
```

**Logs (if applicable):**

```
[14:20:31][I][app:105]: ESPHome version 1.19.4 compiled on Jul  2 2021, 13:55:42
[14:20:31][C][status_led:019]: Status LED:
[14:20:31][C][status_led:020]:   Pin: GPIO0 (Mode: OUTPUT, INVERTED)
[14:20:31][C][wifi:484]: WiFi:
[14:20:31][C][wifi:344]:   SSID: [redacted]
[14:20:31][C][wifi:345]:   IP Address: 172.16.1.6
[14:20:31][C][wifi:347]:   BSSID: [redacted]
[14:20:31][C][wifi:348]:   Hostname: 'env-infrared'
[14:20:31][C][wifi:352]:   Signal strength: -63 dB ▂▄▆█
[14:20:31][C][wifi:356]:   Channel: 4
[14:20:31][C][wifi:357]:   Subnet: 255.255.0.0
[14:20:31][C][wifi:358]:   Gateway: 172.16.0.254
[14:20:31][C][wifi:359]:   DNS1: 172.16.0.128
[14:20:31][C][wifi:360]:   DNS2: (IP unset)
[14:20:31][C][template.switch:058]: Template Switch 'Television'
[14:20:31][C][template.switch:058]:   Assumed State: YES
[14:20:31][C][template.switch:059]:   Restore State: NO
[14:20:31][C][template.switch:060]:   Optimistic: NO
[14:20:31][C][logger:189]: Logger:
[14:20:31][C][logger:190]:   Level: DEBUG
[14:20:31][C][logger:191]:   Log Baud Rate: 115200
[14:20:31][C][logger:192]:   Hardware UART: UART0
[14:20:31][C][status:034]: Status Binary Sensor 'Infrared Transceiver'
[14:20:31][C][status:034]:   Device Class: 'connectivity'
[14:20:31][C][remote_transmitter:018]: Remote Transmitter...
[14:20:31][C][remote_transmitter:019]:   Carrier Duty: 50%
[14:20:31][C][remote_transmitter:020]:   Pin: GPIO13 (Mode: OUTPUT)
[14:20:31][C][remote_receiver.esp8266:059]: Remote Receiver:
[14:20:31][C][remote_receiver.esp8266:060]:   Pin: GPIO14 (Mode: INPUT_PULLUP, INVERTED)
[14:20:31][C][remote_receiver.esp8266:065]:   Buffer Size: 1000
[14:20:31][C][remote_receiver.esp8266:066]:   Tolerance: 25%
[14:20:31][C][remote_receiver.esp8266:067]:   Filter out pulses shorter than: 50 us
[14:20:31][C][remote_receiver.esp8266:068]:   Signal is done after 10000 us of no changes
[14:20:31][C][captive_portal:148]: Captive Portal:
[14:20:31][C][ota:029]: Over-The-Air Updates:
[14:20:31][C][ota:030]:   Address: env-infrared.local:8266
[14:20:31][C][ota:032]:   Using Password.
[14:20:31][C][api:095]: API Server:
[14:20:31][C][api:096]:   Address: env-infrared.local:6053
```

It's probably not all that helpful to include a log to demonstrate _lack_ of log entries, but in the time period covered by the above log several codes were received and passed to HA successfully, and as you can see, were not dumped to the log.

**Additional information and things you've tried:**

glmnet, 2021-07-02:
I guess This is by design. 

Try removing the `on_lg:` action to see if you get the logs.

cerebrate, 2021-07-02:
Oh, I do. But for debugging purposes, it'd be useful to get both a log _and_ the action, belike.

If this is by design, do I need to move it over to feature requests?

glmnet, 2021-07-02:
Yes. 

I believe a log message won't hurt there after a protocol is matched successfully. (This would be logged even if dump is not set) 

But if I were you I'll get away with the log action 😊

OttoWinter, 2021-07-09:
Yeah this is by design - especially with some remotes the code is repeated a lot of times per press. So if we'd log it anyway you'd end up with lots of spam in the console (and make network less stable, etc).

I don't think this needs a feature request, it's quite easy to log the message there yourself with a `logger.log` action

cerebrate, 2021-07-09:
Yeah, now it's been explained, I can see the rationale and it does make sense to me.

What I would still like to request, though, is that this behavior be added to the documentation (at  https://esphome.io/components/remote_receiver.html) which currently doesn't mention it. While the behavior is sensical, it's not _intuitive_, and documenting it would save the next me that comes along hunting down a bugs that doesn't exist. 😄

closed 2021-11-22 by github-actions[bot]