esphome/issues #2215: Cannot use dump with more than one remote_receiver

wesleygas, 2021-07-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io core-2021.6.6, using ESPHome addon 1.19.4

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.19.4

**Affected component:**

https://esphome.io/components/remote_receiver.html

**Description of problem:**

Trying to use `dump` when more than one remote_receiver is configured causes

 ```Too many candidates found for 'receiver_id' type 'remote_base::RemoteReceiverBase'. Some are 'rf_recv', 'ir_recv'.``` 

using, for example, the YAML shown in the issue [#679](https://github.com/esphome/feature-requests/issues/679) from feature requests or the following equivalent:

**Problem-relevant YAML-configuration entries:**
```yaml
remote_receiver:
  - id: "rf_recv"
    pin: 
      number: GPIO32
    dump: all
  - id: "ir_recv"
    pin: 
      number: GPIO34
    dump: all
```

**Additional information and things you've tried:**

If dump is not used, everything works fine

glmnet, 2021-07-03:
This is interesting. Might be just bogus or revealing a issue behind.

jesserockz, 2021-07-03:
The issue is that the dump schema is more complicated than that and attaches a receiver_id to each dump.

wesleygas, 2021-07-03:
Is there a way to define which receiver each dumper should be attached to? 

Like it is done within automations or a binary sensor:
```YAML
on_lg:
     receiver_id: "ir_recv" #Like here
     then:
       ...

binary_sensor:
  - platform: remote_receiver
    receiver_id: "rf_recv" #And here
    id: "rfTab1"
    rc_switch_raw:
      code: ...
```

glmnet, 2021-07-03:
This candidates check already has a special case check for time (as you wouldn't care which time to get the time from as the esp only has one time anyway) maybe dump can be added to the list.

jesserockz, 2021-07-03:
```yaml
...
dump:
  - jvc:
    receiver_id: ...
```

Indentation is correct above

glmnet, 2021-07-03:
Wondering now if dump behaves as expected in the op case. This candidates thingy might be a life saver in the end

jesserockz, 2021-07-04:
We just need to modify the `build_dumpers` function to take the receiver as a param (`var` from caller) instead of using a `receiver_id`. It doesnt make sense to have a dumper that points to another receiver and they are a child of the receiver anyway.

https://github.com/esphome/esphome/blob/d5278351da993ad24f9e7523a0d69f75457ea0dd/esphome/components/remote_base/__init__.py#L227-L234

closed 2021-07-05 by jesserockz