esphome/issues #2217: BME680 different humidity readings (bme680 vs bme680_bsec)

jjansen85, 2021-07-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker (on Synology NAS)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 Wemos D1 mini

**ESPHome version (latest production, beta, dev branch)**

Latest v1.19.4

**Affected component:**

BME680

**Description of problem:**
I have attached and configured the BME680 sensor.
I use the standard BME680 library and the BME680_BSEC library.
The humidity output of both libraries is very different.

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: D4
  scl: D3
  scan: True  

bme680_bsec:
  address: 0x77  

sensor:
  - platform: bme680
    temperature:
      name: "BME680 Temperature"
    humidity:
      name: "BME680 Humidity"
    address: 0x77
    update_interval: 20s    
  - platform: bme680_bsec
    temperature:
      name: "BME680_bsec Temperature"
    humidity:
      name: "BME680_bsec Humidity"

```

**Logs (if applicable):**

```
N/A

```

**Additional information and things you've tried:**

CarlosGS, 2021-07-08:
I'm not sure it is a good idea to use both libraries like this. Bear in mind that the BME680 requires a very consistent timing, as it incorporates a heating element for the gas sensor. Having both libraries running simultaneously could cause "competing" readings (heating before/after reading other parameters) and may give inaccurate values.

Also, note the BSEC library contains proprietary code and could be adjusting/filtering the humidity measurement too.

It would be nice to repeat this test with two BME680 and plot the measurements as an overlapping graph :)

jjansen85, 2021-07-08:
> I'm not sure it is a good idea to use both libraries like this. Bear in mind that the BME680 requires a very consistent timing, as it incorporates a heating element for the gas sensor. Having both [...]

Fair enough. 
I have to experiment with one BME680 (because I have only one). But when I only use the standard library (bme680) it gives me about 64% hum. When I use bme680_bsec library it will drop to about 54%. Temperature and pressure are equal.
My local T+hum measurement gives 61%. 

Complicated. I should get an additional BME680 to do the test you suggested.

closed 2021-11-22 by github-actions[bot]