esphome/issues #2218: [i2c_id] is an invalid option for [sensor.custom].

5tan, 2021-07-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Ubuntu 20.04, esphome installed using pip.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, ESP-WROOM-32E on Espressif board 

**ESPHome version (latest production, beta, dev branch)**

1.18.0

**Affected component:**

sensor

**Description of problem:**
I have two identical (the same address) i2c sensors on two buses.
How to specify which bus should sensor use?

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  - id: bus1
    sda: GPIO32
    scl: GPIO27
  - id: bus2
    sda: GPIO25
    scl: GPIO26

sensor:
  - platform: custom
    # i2c_id: bus1
    # address: 0x10
    lambda: |-
      auto sensor = new Veml7700();
      App.register_component(sensor);
      return {sensor};
    sensors:
      name: ALS1
```

**Logs (if applicable):**

```
  platform: custom
  
  [i2c_id] is an invalid option for [sensor.custom]. Please check the indentation.

```

**Additional information and things you've tried:**

5tan, 2021-07-07:
It looks like it is impossible to use multiple custom i2c sensors on multiple buses.

[This ](https://esphome.io/custom/i2c.html)manual page recommends using Wire. But apparently [Wire is designed for single bus only. ](https://dronebotworkshop.com/multiple-i2c-bus/)
> For all of its great features, it has some shortcomings – Wire does not allow you to have multiple I2C buses, you can only call one instance of it and you can’t redirect your I2C traffic to different pins.

OttoWinter, 2021-07-09:
Correct, `Wire` has some major limitations (hence ESPHome uses its own abstraction for i2c).

I assume your custom component uses an external library? Yeah then you would need to rewrite the external library to use esphome's i2c abstractions instead

5tan, 2021-07-14:
> I assume your custom component uses an external library? 

no

> ESPHome uses its own abstraction for i2c

I have reviewed the code, and it looks like multiple i2c buses are supported by the i2c abstraction. Thus I don't understand why `i2c_id` is invalid config for `sensor.custom`? It should work.

I have managed to workaround the issue by creating custom components.

OttoWinter, 2021-07-26:
> Thus I don't understand why i2c_id is invalid config for sensor.custom?

That's because `sensor.custom` doesn't know about i2c. `sensor.custom` only exposes a component constructor. It doesn't integrate into i2c/other protocols. The developer is expected to set those themself using the `set_i2c_parent` call.

marcoboot, 2021-10-15:
I have the same issue. I have got one working, by just having one I2C bus declared first (veml6075). I now have my second custom I2C device that is on my 2nd I2C bus (arduino based sensor). 
Lets say, my sensor doesn't use a library (it basicly the arduino version of "master reader/slave sender). So only one call of Wire.requestFrom(), and 2 calls of Wire.read(). What do i need to change to be able to set a i2c id?

ssieb, 2022-08-02:
If you make a real component, you can configure the i2c id in yaml.  For now you can pass the i2c component to your constructor and use the proper i2c methods in your component.
This is not an esphome issue and github is not for general support.  Come to the Discord server if you need further help.

closed 2022-08-02 by ssieb