esphome/issues #2220: ble_client sensor keep connecting after disconnect even huge update_interval

myhomeiot, 2021-07-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Ubuntu 20.04, esphome installed as Home Assistant addon.

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32, esp32dev

**ESPHome version (latest production, beta, dev branch)**
1.19.4

**Affected component:**
ble_client sensor https://esphome.io/components/sensor/ble_client.html

**Description of problem:**
ble_client sensor keep connecting after disconnect even huge update_interval, this drain the battery of device, I expect that sensor will connect according update_interval

**Problem-relevant YAML-configuration entries:**
```yaml

esp32_ble_tracker:

ble_client:
  - mac_address: !secret mi_band_mac
    id: ble_client_mi_band

sensor:
  - platform: template
    id: sensor_mi_band
    name: Mi Band Battery
    device_class: battery
    unit_of_measurement: '%'
    state_class: measurement
    accuracy_decimals: 0

  - platform: ble_client
    id: ble_client_mi_band_data
    internal: true
    ble_client_id: ble_client_mi_band
    service_uuid: FEE0
    characteristic_uuid: 00000006-0000-3512-2118-0009AF100700
    notify: false
    update_interval: 1h
    lambda: |-
      id(sensor_mi_band).publish_state(x[1]);
      return 0;
```

**Logs (if applicable):**

```
[19:12:04][D][ble_client:045]: Found device at MAC address [DA:51:XX:XX:XX:XX]
[19:12:04][I][ble_client:083]: Attempting BLE connection to da:51:xx:xx:xx:xx
[19:12:05][I][ble_sensor:033]: [ble_client_mi_band_data] Connected successfully!
...
[19:13:05][W][ble_sensor:039]: [ble_client_mi_band_data] Disconnected!
[19:13:31][D][ble_client:045]: Found device at MAC address [DA:51:XX:XX:XX:XX]
[19:13:31][I][ble_client:083]: Attempting BLE connection to da:51:xx:xx:xx:xx
[19:13:32][I][ble_sensor:033]: [ble_client_mi_band_data] Connected successfully!
```

**Additional information and things you've tried:**
I tried to change notify and update_interval

randysnel, 2021-07-15:
I can confirm the same behaviour as well when connecting to a switchbot curtain as 'server'.

myhomeiot, 2021-07-16:
I'm not sure that this behavior can be fixed soon, meanwhile I made component which will solves this problem. If anybody interested, the component can be found [here](https://github.com/myhomeiot/esphome-components).

randysnel, 2021-11-16:
> I'm not sure that this behavior can be fixed soon, meanwhile I made component which will solves this problem. If anybody interested, the component can be found [here](https://github.com/myhomeiot/esphome-components).

can't you issue a pull request?

myhomeiot, 2021-11-16:
> can't you issue a pull request?

Unfortunately no, my component based on rewrites ble_client and can't work with regular ble_client simultaneously. From ESPHome version 2021.10 my component stops working because lot of changes in ESPHome BLE which I will try to fix when 2021.11 will be released.

robloh, 2022-07-24:
@myhomeiot did you ever get this merged? Thanks!

myhomeiot, 2022-07-25:
@robloh No because it's out of the scope of regular BLE Client, if you want to have this functionality, you can use my [component](https://github.com/myhomeiot/esphome-components), some examples you can find [here](https://github.com/myhomeiot/esphome-components/tree/main/examples/myhomeiot_ble_client).

My component can co-exists with regular BLE Client, just take into account that current platform limit it's **three** BLE Client's and my component (MyHomeIOT BLE Host) take one slot but can work with many MyHomeIOT BLE Client's simultaneously, so if you have for example 5 MiFlora's you can read battery information without any problems.

BTW You can also have this functionality using regular BLE Client if you enable/disable it using [BLE Client Switch](https://esphome.io/components/switch/ble_client.html) but in this case you should remember about platform limitation of three BLE Client's.