esphome/issues #2221: CLI Error -- the following arguments are required: configuration

mikhailbot, 2021-07-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

macOS 11.4
Python 3.8.11

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**ESPHome version (latest production, beta, dev branch)**

1.19.4

**Affected component:**

**Description of problem:**
Cannot run the wizard:

```
❯ esphome wizard
usage: esphome wizard [-h] configuration
esphome wizard: error: the following arguments are required: configuration

❯ esphome version
WARNING Calling ESPHome with the configuration before the command is deprecated and will be removed in the future. 
WARNING Please instead use:
WARNING    esphome version
Version: 1.19.4

❯ python --version
Python 3.8.11
```

**Problem-relevant YAML-configuration entries:**
```yaml
N/A
```

**Logs (if applicable):**

```
N/A

```

**Additional information and things you've tried:**

mikhailbot, 2021-07-07:
I also tried doing `pip install voluptuous==0.11.5` as mentioned in other issues, but to no avail.

glmnet, 2021-07-07:
You must also pass the name of the file you want to create

mikhailbot, 2021-07-07:
Oh man, I totally missed that--thanks. Maybe an improvement to the error message would be helpful too!

closed 2021-07-07 by mikhailbot