esphome/issues #2222: LEDC dimming cycle flickers when using I2C

intermittech, 2021-07-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Using Windows 10 and Home Assistant running in Virtual Machines (Proxmox)

**ESP (ESP32/ESP8266, Board/Sonoff):**

Custom ESP32 dimmer boards, QuinLED-Quad, QuinLED-Deca, Prototype boards

**ESPHome version (latest production, beta, dev branch)**

Newest version, as of writing 1.19.4 in fully updated Home Assistant

**Affected component:**

LEDC dimming between values

**Description of problem:**
Everything works correctly when just using LEDC and dimming between values per channel. After adding I2C Screen or Temperature sensor I noticed flickering at some moments. After lowering temperature sensor read time to 10s and setting dimming transition time to 10s it was pretty clear that each time the temperature sensor was read, any dimming fade/change that was going on gets interrupted causing a visual flicker.

Once fading is done, LEDC PWM output is fully stable, flickers are only visible during transition times.

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: 1
  scl: 3
  frequency: 400kHz
  scan: true

output:
  - platform: ledc
    pin: 2
    frequency: 20000Hz
    id: LED_channel_1

light:
  - platform: monochromatic
    name: "${esphome_name}-1"
    default_transition_length: 5s
    id: LED1
    output: LED_channel_1

sensor:
  - platform: aht10
    temperature:
      name: "test_anpentalan Temperature"
    humidity:
      name: "test_anpentalan Humidity"
    update_interval: 10s
```

**Logs (if applicable):**

```
I can add logs but there is nothing special visible in the logs, all seems to be working normally.

```

**Additional information and things you've tried:**
I've tried varying I2C frequency, 10kHz, 100kHz, 400kHz or 1000kHz makes no difference to the issue. Tried using only temperature sensor (AHT21) or screen, no change to issue, especially with screen updating issue is VERY apparent. Almost like I2C interrupts the (likely) software dimming routine, sending the new values out very quickly to show the smooth fade.

intermittech, 2021-07-08:
Bit of an update on this!

I got some more sensors in and replaced the AHT21 with a AHT10, of course, same results. :)

Then I replaced it with a SHT30 and I was surprised to see the flickering during dimming is completely gone. So where I first feared this was caused by using I2C (since the screen caused the flickering too) it seems to be device/driver dependent, not I2C by itself.

I'm hoping someone more versed in coding can take a look at the 2 sensors components and see if this can indeed be the case. My non-programmer brain assumed that especially using I2C reading sensor type A vs sensor type B would be roughly the same but at least the in the current implementations, that's certainly not the case.

Keeping the issue open until someone has a bit of time to at least verify this. The AHT21 is a decent sensor (values where good) for ~0,5$ and the SHT30 easily costs ~1.5$ so hoping something can be done code wise (or at least verified this is the case and could maybe be improved with software). :)

OttoWinter, 2021-07-09:
Do you mean the flickering is _just_ while dimming?

Looks like `aht10` does blocks in the main loop (but only by ~30ms) [here](https://github.com/esphome/esphome/blob/dev/esphome/components/aht10/aht10.cpp#L24-L25) and [here](https://github.com/esphome/esphome/blob/dev/esphome/components/aht10/aht10.cpp#L64) - which is a bit much but not enough to trigger a warning.

The function would need to be updated to use `set_timeout` etc to wait for the data asynchronously. But probably a bit harder due to the i2c base not supporting that out-of-the box for read_bytes

intermittech, 2021-07-09:
Yes, just while transitioning from value A to value B, once it's on that value the hardware PWM of the ESP32 likely takes over and no more flickering is observed. So I believe it's the "changing values" routine which quickly spams higher or lower brightness values to generate the fade that is getting interrupted.

And yeah, switching to the SHT30 completely solved the issue just sad it's 3x as expensive. In my un-informed ways I figured the reading of one I2C sensor wouldn't be too much vs another but seems the implementations at least are very different!

I appreciate the reply, I'll likely continu developing with the SHT30 going forward, or switch back to a DS18B20 1-wire maybe because of cost, not sure yet. :)

closed 2021-08-04 by intermittech