esphome/issues #2223: SPI screen doesn't work together with Ethernet

intermittech, 2021-07-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Using Windows 10 and Home Assistant running in Virtual Machines (Proxmox)

**ESP (ESP32/ESP8266, Board/Sonoff):**

Custom ESP32 dimmer Prototype board

**ESPHome version (latest production, beta, dev branch)**

Newest version, as of writing 1.19.4 in fully updated Home Assistant

**Affected component:**

SPI screen display

**Description of problem:**
I'm building a ESP32 based dimmer board where you can plug-in a QuinLED-ESP32-AB or AE (Antenna board, Antenna External) or QuinLED-ESP32-ABE (Antenna board + Ethernet) for whatever is your desired way to hook it to the network is.

[Because of problems using I2C](https://github.com/esphome/issues/issues/2222) I switched the screen from using I2C to SPI. When using the normal WiFi boards my pinout for this is working fine and everything from network communication, dimming and updating the SPI OLED screen (SSD1306) are working fine.

When switching the ESP32 board in the prototype carrier board (with the MOSFETs, etc.) and changing the config to add Ethernet everything except the screen works. The screen will still turn on, but whatever pinout I choose the text on screen will be scrolling basically (I will try to add visual evidence later). Everything still works and I can even see my text and graphics getting updated on the screen, but it basically keeps on scrolling. If I put on a single word, that word will jump to the right every 2 seconds until it's looped the screen.

The exact same code and (non-conflicting with Ethernet) pinout works perfectly when you switch back to WiFi.

**Problem-relevant YAML-configuration entries:**
```yaml
ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO17_OUT
  phy_addr: 0
  power_pin: GPIO5

spi:
  clk_pin: GPIO15
  mosi_pin: GPIO16
#  miso_pin: GPIO0

display:
  - platform: ssd1306_spi
    model: "SSD1306 128x64"
    cs_pin: GPIO32
    dc_pin: GPIO0
    reset_pin: GPIO33
    id: my_display
    update_interval : 16ms
```

**Logs (if applicable):**

```
I can add logs but there is nothing special visible in the logs, all seems to be working normally.

```

**Additional information and things you've tried:**
I've tried every pinout I can think of in combination with the pins I have available. These are quite limited though because I also have I2C on the board (disabling this makes no difference) and Ethernet alone takes up 10 GPIO pins of which most are fixed.

It feels like there is a clock issue where a shared clock gets screwed up when Ethernet is activated. As mentioned above, I have tried all combinations that are allowed with the pins I have, it looks like all combinations that give me a working screen (correctly on the WiFi only boards) all seem to cause the scrolling issue.

intermittech, 2021-07-08:
Some new info!

If I set the screen update to 1ms the scrolling is much much faster. If I set it to 1s the Ethernet doesn't come online anymore!

I believe there is definitely something going on with SPI clocks that are having an issue. Hopefully this is fixable somehow, the Ethernet is on dedicated pins over the RMII interface of the ESP32 so I wouldn't think those would not be related to the SPI hardware.

OttoWinter, 2021-07-09:
Might be a hardware limitation. IIRC ESP32 Ethernet uses SPI to communicate with the Ethernet PHY chip.

But if not then it's probably an upstream issue, as ESPHome just forwards SPI/Ethernet to esp-idf (/arduino-esp32)

intermittech, 2021-07-09:
> Might be a hardware limitation. IIRC ESP32 Ethernet uses SPI to communicate with the Ethernet PHY chip.
> 
> But if not then it's probably an upstream issue, as ESPHome just forwards SPI/Ethernet to esp-idf (/arduino-esp32)

Right, I was a bit afraid of that, I appreciate the reply though! Is there another way to interface a screen that is not SPI or I2C? I changed over from I2C because it was causing massive visual glitches while PWM dimming, [in line with my other open issue](https://github.com/esphome/issues/issues/2222) (but then much much worse since the screen constantly updates).

closed 2021-11-22 by github-actions[bot]